nxp-imx/mfgtools #5: Is the license the CPOL or BSD-3c?

fhunleth, 2016-10-20:
The `LICENSE` file and several header files have the BSD license text, but `CPOL.htm` also exists. Was `CPOL.htm` committed by mistake?

nxpfrankli, 2016-10-20:
There are some files which is CPOL.   Most part is BSD.

fhunleth, 2016-10-20:
Would it be possible for you to add a short section to the README to explain what's covered by which license? Fwiw, I couldn't find any files with references to the CPOL.

Also, I noticed that when I switched to the `windows` branch, the `LICENSE` file was removed and replaced by a `COPYING.MIT`. Could you clarify what's covered by the MIT license?

nxpfrankli, 2016-10-20:
I update master (linux) branch. 
Windows branch need check history.

nxpfrankli, 2016-11-08:
COPYING.MIT already removed

closed 2016-12-01 by nxpfrankli