nxp-imx/mfgtools #7: could not open USB Device

joerg-krause, 2016-11-09:
I am using the linux version of this tool. When using `mfgtoolcli -l mmc -p 1`, it cannot open the USB device.

dmesg and lusb shows:

```
# dmesg
[11733.680522] usb 3-11.4.4: new high-speed USB device number 18 using xhci_hcd
[11733.772035] hid-generic 0003:15A2:004F.0007: hiddev0,hidraw4: USB HID v1.10 Device [Freescale,Inc. ROM Recovery] on usb-0000:00:14.0-11.4.4/input0

# lsusb
Bus 003 Device 018: ID 15a2:004f Freescale Semiconductor, Inc. i.MX28 SystemOnChip in RecoveryMode
```

The log shows:

```
ModuleID[2] LevelID[1]: new HidDeviceClass
ModuleID[2] LevelID[10]: Add USB device
ModuleID[2] LevelID[10]: Create Device (nil)
ModuleID[2] LevelID[1]: failed to open no access
ModuleID[2] LevelID[1]: could not open USB Device
```

The `CFG` section in ucl2.xml is:

```
  <CFG>
    <STATE name="BootStrap" dev="MX28" vid="15A2" pid="004F"/>
    <STATE name="Updater" dev="MSC" vid="066F" pid="37FF"/>
  </CFG>
```

joerg-krause, 2016-11-09:
I figured it out that libusb needs root privileges so I created a udev rule. Now I am getting an Segmentation fault:

```
Your Options:
Ready to flash.
Segmentation fault (core dumped)
```

manulicious, 2017-03-20:
I’m getting a segfault, too.

I don’t understand a lot of what’s going on inside ``Device.cpp``, but to me it seems that the constructor for `Device::Device` only treats Windows-specific device paths. Therefore Linux can’t create the device, and the constructor returns `null`.

I suspect this tool simply isn’t meant to support the i.MX28 (yet), since I only find it exclusively distributed for flashing i.MX6/i.MX7 SoC’s. For “mature products” (meaning i.MX53 and older), there is _always_ the Windows-only MfgTool 1.6* referred for download, as you can see here: http://www.nxp.com/products/microcontrollers-and-processors/arm-processors/i.mx-applications-processors/i.mx-software-and-tools:IMXSW_HOME

Sadly, I neither have the experience to make it work myself, nor does the Windows tool work for me.

Logging information:
```
DLL version: 2.3.4
Mon Mar 20 15:16:13 2017
   Start new logging
ModuleID[2] LevelID[10]: address from XML 0x42000000 

ModuleID[2] LevelID[10]: address from XML 0x4772a000 

ModuleID[2] LevelID[10]: address from XML 0x47722000 

ModuleID[2] LevelID[10]: CMyExceptionHandler thread is running
ModuleID[2] LevelID[1]: new HidDeviceClass
ModuleID[2] LevelID[10]: Add USB device

ModuleID[2] LevelID[10]: Create Device (nil)
```

Stack trace of relevant thread:
```
Stack trace of thread 3371:
                #0  0x00007f3bd62235fd libusb_open (libusb-1.0.so.0)
                #1  0x00007f3bd64df666 _ZN11DeviceClass12AddUsbDeviceEPKcP13libusb_device (libMfgToolLib.so)
                #2  0x00007f3bd64e1ba3 _ZN13DeviceManager16OnMsgDeviceEventEtm (libMfgToolLib.so)
                #3  0x00007f3bd64e0914 _Z20DevManagerThreadProcPv (libMfgToolLib.so)
                #4  0x00007f3bd5c7f2e7 start_thread (libpthread.so.0)
                #5  0x00007f3bd57a954f __clone (libc.so.6)
```