nxp-imx/mfgtools #11: mfgtoolcli lost files after successful transfer done

niurenyige, 2017-06-12:
when use mfgtoolcli in debian 8.8 
rootfs.tar.bz2 about 35MB.
the target will lost some files random .
is there a way to guarantee the target files get is same as host .no file lost ,no file broken.

nxpfrankli, 2017-06-12:
Did you run unmount in ucl.xml?     "Tar" always check CRC and magic number,  it will report error if transfer error.

niurenyige, 2017-06-13:
yes ,the last cmd in my ucl.xml pls refer, and if the rootfs.bar.bz2 size is bigger than 36MB ,
it will report tar error.

<CMD state="Updater" type="push" body="pipe tar -jxv -C /mnt/mtd%part_rootfs%" file="files/rootfs.tar.bz2" ifdev="MX6UL MX7D MX6ULL">Sending and writting rootfs</CMD>
	<CMD state="Updater" type="push" body="frf">Finishing rootfs write</CMD>
	<CMD state="Updater" type="push" body="$ umount /mnt/mtd%part_rootfs%">Unmounting rootfs partition</CMD>
	<CMD state="Updater" type="push" body="$ echo Update Complete!">Done</CMD>