nxp-imx/mfgtools #14: "Failed to initialize memory!"

Wesley-Chan, 2017-07-13:
I tried to flash my iMX.6D custom board with mfgtoolscli and this error came up. The command was

    sudo ./mfgtoolcli -c Linux -l eMMC -s board=sabresd -s sxdtb=sdb -s mmc=3

Here is the log.

```
DLL version: 2.3.4
Thu Jul 13 14:30:07 2017
   Start new logging
ModuleID[2] LevelID[10]: address from XML 0x12000000 

ModuleID[2] LevelID[10]: address from XML 0x80800000 

ModuleID[2] LevelID[10]: address from XML 0x12C00000 

ModuleID[2] LevelID[10]: address from XML 0x83800000 

ModuleID[2] LevelID[10]: address from XML 0x18000000 

ModuleID[2] LevelID[10]: address from XML 0x18000000 

ModuleID[2] LevelID[10]: address from XML 0x83000000 

ModuleID[2] LevelID[10]: address from XML 0x83000000 

ModuleID[2] LevelID[10]: address from XML 0x83000000 

ModuleID[2] LevelID[10]: CMyExceptionHandler thread is running
ModuleID[2] LevelID[1]: new MxHidDeviceClass
ModuleID[2] LevelID[1]: new MxHidDeviceClass
ModuleID[2] LevelID[1]: new MxHidDeviceClass
ModuleID[2] LevelID[1]: new MxHidDeviceClass
ModuleID[2] LevelID[1]: new MxHidDeviceClass
ModuleID[2] LevelID[1]: new MxHidDeviceClass
ModuleID[2] LevelID[10]: Add USB device

ModuleID[2] LevelID[10]: Create Device 0x7f5d10003fc0

ModuleID[2] LevelID[10]: new MxHidDevice[0x7f5d10003fc0]
ModuleID[2] LevelID[10]: ExecuteCommand--Boot[WndIndex:0], File is ./Profiles/Linux/OS Firmware/firmware/u-boot-imx6qsabresd_sd.imx
ModuleID[2] LevelID[10]:  Trans Data address is 177ff400 

ModuleID[2] LevelID[10]: ExecuteCommand--Load[WndIndex:0], File is ./Profiles/Linux/OS Firmware/firmware/zImage, address is 0x12000000
ModuleID[2] LevelID[10]:  Trans Data address is 12000000 

ModuleID[2] LevelID[10]:  Trans Data address is 12200000 

ModuleID[2] LevelID[10]:  Trans Data address is 12400000 

ModuleID[2] LevelID[10]: PortMgrDlg(0)--Command Load excute successfully, retry count: 0
ModuleID[2] LevelID[10]: ExecuteCommand--Load[WndIndex:0], File is ./Profiles/Linux/OS Firmware/firmware/fsl-image-mfgtool-initramfs-imx_mfgtools.cpio.gz.u-boot, address is 0x12C00000
ModuleID[2] LevelID[10]:  Trans Data address is 12c00000 

ModuleID[2] LevelID[10]:  Trans Data address is 12e00000 

ModuleID[2] LevelID[10]:  Trans Data address is 13000000 

ModuleID[2] LevelID[10]:  Trans Data address is 13200000 

ModuleID[2] LevelID[10]: PortMgrDlg(0)--Command Load excute successfully, retry count: 0
ModuleID[2] LevelID[10]: ExecuteCommand--Load[WndIndex:0], File is ./Profiles/Linux/OS Firmware/firmware/zImage-imx6q-sabresd.dtb, address is 0x18000000
ModuleID[2] LevelID[10]:  Trans Data address is 18000000 

ModuleID[2] LevelID[10]: PortMgrDlg(0)--Command Load excute successfully, retry count: 0
ModuleID[2] LevelID[10]: ExecuteCommand--Jump[WndIndex:0]
ModuleID[2] LevelID[10]: *********MxHidDevice[0x7f5d10003fc0] Jump to Ramkernel successfully!**********
ModuleID[2] LevelID[10]: ExecuteCommand--Boot[WndIndex:0], File is ./Profiles/Linux/OS Firmware/firmware/u-boot-imx6qsabresd_sd.imx
ModuleID[2] LevelID[1]: Failed to initialize memory!
ModuleID[2] LevelID[1]: PortMgrDlg(0)--MxHidDevice--Command Boot excute failed
[... 62 lines cut ...]

	<CMD state="Updater" type="push" body="$ umount /mnt/mmcblk%mmc%p1">Unmounting vfat partition</CMD>

	<!-- burn rootfs -->
	<CMD state="Updater" type="push" body="$ mkfs.ext3 -E nodiscard /dev/mmcblk%mmc%p2">Formatting rootfs partition</CMD>
	<CMD state="Updater" type="push" body="$ mkdir -p /mnt/mmcblk%mmc%p2"/>
	<CMD state="Updater" type="push" body="$ mount -t ext3 /dev/mmcblk%mmc%p2 /mnt/mmcblk%mmc%p2"/>
	<CMD state="Updater" type="push" body="pipe tar -jxv -C /mnt/mmcblk%mmc%p2" file="files/rootfs.tar.bz2" ifdev="MX6SL MX6D MX6Q MX6SX">Sending and writting rootfs</CMD>

	<CMD state="Updater" type="push" body="frf">Finishing rootfs write</CMD>
	<CMD state="Updater" type="push" body="$ umount /mnt/mmcblk%mmc%p2">Unmounting rootfs partition</CMD>
	<CMD state="Updater" type="push" body="$ echo Update Complete!">Done</CMD>
```

I took a quick glance in the source code and this is the part that failed.

```
if ( !DCDWrite((PUCHAR)(pDCDRegion),TotalDCDDataCnt) )
{
    LogMsg(LOG_MODULE_MFGTOOL_LIB, LOG_LEVEL_FATAL_ERROR, _T("Failed to initialize memory!"));
    goto ERR_HANDLE;
}
```

Can anyone give me a hint about what could possibly go wrong and how to fix this? I'm running this on Gentoo 64-bit and `lsusb` reports `Bus 003 Device 057: ID 15a2:0054 Freescale Semiconductor, Inc. i.MX 6Dual/6Quad SystemOnChip in RecoveryMode`, for what it's worth. Thank you.

nxpfrankli, 2017-07-28:
ModuleID[2] LevelID[10]: PortMgrDlg(0)--Command Load excute successfully, retry count: 0
ModuleID[2] LevelID[10]: ExecuteCommand--Jump[WndIndex:0]
ModuleID[2] LevelID[10]: *********MxHidDevice[0x7f5d10003fc0] Jump to Ramkernel successfully!**********

> Look like your board reboot hear.  Mfgtools start next transfer. 
Did you see any log in uart port>

rbejaran, 2017-10-27:
I see the same problem when trying to flash my i.MX6D with this Linux tool:
```
ModuleID[2] LevelID[10]: *********MxHidDevice[0x7f12c4002790] Jump to Ramkernel successfully!**********
ModuleID[2] LevelID[10]: ExecuteCommand--Boot[WndIndex:0], File is ./Profiles/foo/OS Firmware/factory/u-boot_macan_mfg.imx
ModuleID[2] LevelID[1]: Failed to initialize memory!
ModuleID[2] LevelID[1]: PortMgrDlg(0)--MxHidDevice--Command Boot excute failed
```
The logs above show a problem that really puzzles me. mfgtools tries to execute a 2nd time the very fist command (that has already been executed a few seconds before):
```
<CMD state="BootStrap" type="boot" body="BootStrap" file="factory/u-boot_macan_mfg.imx">Loading factory u-boot</CMD>
```
So why mfgtools is not following the command list order? Doing this command a 2nd time, at this point, may be the cause for the memory initialization problem, right?

For the record, in my case (@Wesley-Chan is this your case too?), after this error mfgtools doesn't block there, it starts all over again from the beginning until it gets blocked when downloading the `u-boot.imx`:
```
<CMD state="Updater" type="push" body="send" file="images/u-boot.imx">Sending u-boot</CMD>
```
Last log line in MfgToolLibLog.txt is this one, which shows mfgtools is stuck during the send command:
```
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is send
```
The UART output shows the send command succeeded:
```
UTP: received command 'send'
UTP: sending Success to kernel for command send.
```
But somehow it seems mfgtool doesn't get this success. I wonder if this just a consequence of the memory initialization problem, or a whole different problem... Does anyone has an idea?

@nxpfrankli , yes the board reboots at that point. It is the expected thing to happen because of the jump command, to boot on the new kernel:
```
<CMD state="BootStrap" type="jump" > Jumping to OS image. </CMD>
```
Thanks.

mateusz-, 2017-12-07:
Besides +1 I don't have that much to contribute. Getting the same result from an Ubuntu virtual machine (16.04) with USB passthrough on MacOS High Sierra.

@rbejaran Were you able to resolve this issue?

Anyways, flashing on Windows seems to work.

mateusz-, 2017-12-08:
Here's what my own issue was.

In my case, I was using a virtual machine to pass through a USB device, but something similar may happen when trying to flash on a host, especially when dealing with a DIY distro such as Gentoo.

Basically, after my board rebooted I had to re-attach the new USB device, and the flashing process continued. Simple as that.

I am still getting the "Failed to initialize memory!" message in the log, but the process seems to finish successfully.

```bash
mateusz@mateusz-VirtualBox:~/Projects/mfgtools$ more MfgToolLibLog.txt 

DLL version: 2.3.4
Fri Dec  8 10:43:55 2017
   Start new logging
ModuleID[2] LevelID[10]: address from XML 0x80800000 

ModuleID[2] LevelID[10]: address from XML 0x83800000 

ModuleID[2] LevelID[10]: address from XML 0x83000000 

ModuleID[2] LevelID[10]: CMyExceptionHandler thread is running
ModuleID[2] LevelID[1]: new MxHidDeviceClass
ModuleID[2] LevelID[10]: Add USB device

ModuleID[2] LevelID[10]: Create Device 0x7f422c002790

ModuleID[2] LevelID[10]: new MxHidDevice[0x7f422c002790]
ModuleID[2] LevelID[10]: ExecuteCommand--Boot[WndIndex:0], File is ./Profiles/Linux/OS Firmware/firmware/u-boot-vesta-mfg-300.imx
ModuleID[2] LevelID[10]:  Trans Data address is 877ff7d0 

ModuleID[2] LevelID[10]: ExecuteCommand--Load[WndIndex:0], File is ./Profiles/Linux/OS Firmware/firmware/zImage, address is 0x80800000
ModuleID[2] LevelID[10]:  Trans Data address is 80800000 

ModuleID[2] LevelID[10]:  Trans Data address is 80a00000 

ModuleID[2] LevelID[10]:  Trans Data address is 80c00000 

ModuleID[2] LevelID[10]: PortMgrDlg(0)--Command Load excute successfully, retry count: 0
ModuleID[2] LevelID[10]: ExecuteCommand--Load[WndIndex:0], File is ./Profiles/Linux/OS Firmware/firmware/fsl-image-mfgtool-initramfs-imx_mfgtools.cpio.gz.u-boot, address is 0x838
00000
ModuleID[2] LevelID[10]:  Trans Data address is 83800000 

ModuleID[2] LevelID[10]:  Trans Data address is 83a00000 

ModuleID[2] LevelID[10]:  Trans Data address is 83c00000 

ModuleID[2] LevelID[10]:  Trans Data address is 83e00000 

ModuleID[2] LevelID[10]: PortMgrDlg(0)--Command Load excute successfully, retry count: 0
ModuleID[2] LevelID[10]: ExecuteCommand--Load[WndIndex:0], File is ./Profiles/Linux/OS Firmware/firmware/vesta-300-mfg.dtb, address is 0x83000000
ModuleID[2] LevelID[10]:  Trans Data address is 83000000 

ModuleID[2] LevelID[10]: PortMgrDlg(0)--Command Load excute successfully, retry count: 0
ModuleID[2] LevelID[10]: ExecuteCommand--Jump[WndIndex:0]
ModuleID[2] LevelID[10]: *********MxHidDevice[0x7f422c002790] Jump to Ramkernel successfully!**********
ModuleID[2] LevelID[10]: ExecuteCommand--Boot[WndIndex:0], File is ./Profiles/Linux/OS Firmware/firmware/u-boot-vesta-mfg-300.imx
ModuleID[2] LevelID[1]: Failed to initialize memory!
ModuleID[2] LevelID[1]: PortMgrDlg(0)--MxHidDevice--Command Boot excute failed
ModuleID[2] LevelID[10]: Add USB device

ModuleID[2] LevelID[10]: DeviceTypeDisk--_devices size: 0
ModuleID[2] LevelID[10]: Create Device 0x7f422c003300

ModuleID[2] LevelID[10]: ExecuteCommand--Boot[WndIndex:0], File is ./Profiles/Linux/OS Firmware/firmware/u-boot-vesta-mfg-300.imx
ModuleID[2] LevelID[10]: ExecuteCommand--Load[WndIndex:0], File is ./Profiles/Linux/OS Firmware/firmware/zImage, address is 0x80800000
ModuleID[2] LevelID[10]: ExecuteCommand--Load[WndIndex:0], File is ./Profiles/Linux/OS Firmware/firmware/fsl-image-mfgtool-initramfs-imx_mfgtools.cpio.gz.u-boot, address is 0x838
00000
ModuleID[2] LevelID[10]: ExecuteCommand--Load[WndIndex:0], File is ./Profiles/Linux/OS Firmware/firmware/vesta-300-mfg.dtb, address is 0x83000000
ModuleID[2] LevelID[10]: ExecuteCommand--Jump[WndIndex:0]
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is send
[... 11 lines cut ...]
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is $ mount -t vfat /dev/mmcblk1p2 /mnt/mmcblk1p2
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is $ mkfs.vfat /dev/mmcblk1p3
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is send
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is $ cp $FILE /mnt/mmcblk1p2/zImage
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is send
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is $ cp $FILE /mnt/mmcblk1p2/zImage-vesta-300.dtb
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is $ umount /mnt/mmcblk1p2
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is $ mkfs.ext3 -E nodiscard /dev/mmcblk1p5
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is $ mkdir -p /mnt/mmcblk1p5
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is $ mount -t ext3 /dev/mmcblk1p5 /mnt/mmcblk1p5
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is pipe tar -jxv -C /mnt/mmcblk1p5
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is frf
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is $ umount /mnt/mmcblk1p5
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is $ mkfs.ext3 -E nodiscard /dev/mmcblk1p6
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is $ mkfs.ext3 -E nodiscard /dev/mmcblk1p7
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is send
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is $ tar xf $FILE 
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is send
[... cut ...] -header.sh $FILE
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is $ busybox hexdump -R qspi-tmp > qspi-header
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is $ dd if=qspi-header of=/dev/mtd0 bs=1k seek=1
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is $ rm qspi-tmp
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is $ rm qspi-header
ModuleID[2] LevelID[10]: ExecuteCommand--Push[WndIndex:0], Body is $ echo Update Complete!
```

chelliwell, 2018-01-24:
I'm having a similar problem. Well, the same error message, though not even getting as far as the posters above.
I'm running on a (old, slow) 32-bit Ubuntu 14.04.4.
For further background - to avoid repeating here - please see https://community.nxp.com/thread/468646
On this machine, it fails on the first stage. Log file has:

```
DLL version: 2.3.4
Wed Jan 24 14:51:35 2018
   Start new logging
ModuleID[2] LevelID[10]: address from XML 0x10500000

ModuleID[2] LevelID[10]: address from XML 0x10800000

ModuleID[2] LevelID[10]: address from XML 0x11180000

ModuleID[2] LevelID[10]: address from XML 0x11200000

ModuleID[2] LevelID[10]: address from XML 0x10600000

ModuleID[2] LevelID[10]: address from XML 0x10540000

ModuleID[2] LevelID[10]: address from XML 0x11100000

ModuleID[2] LevelID[10]: CMyExceptionHandler thread is running
ModuleID[2] LevelID[1]: new MxHidDeviceClass
ModuleID[2] LevelID[1]: new MxHidDeviceClass
ModuleID[2] LevelID[1]: new MxHidDeviceClass
ModuleID[2] LevelID[10]: Add USB device

ModuleID[2] LevelID[10]: Create Device 0xb4e019a8

ModuleID[2] LevelID[10]: new MxHidDevice[0xb4e019a8]
ModuleID[2] LevelID[10]: ExecuteCommand--Boot[WndIndex:0], File is ./Profiles/myboard/OS Firmware/files/u-boot/mfg/u-boot-tx6s-8035_mfg.bin
ModuleID[2] LevelID[1]: MxHidDevice::Read() Error reading from device -7 with size 1025.
ModuleID[2] LevelID[1]: Failed to read HAB type from ROM!!!
ModuleID[2] LevelID[1]: Failed to initialize memory!
ModuleID[2] LevelID[1]: PortMgrDlg(0)--MxHidDevice--Command Boot excute failed
ModuleID[2] LevelID[10]: ExecuteCommand--Boot[WndIndex:0], File is ./Profiles/myboard/OS Firmware/files/u-boot/mfg/u-boot-tx6s-8035_mfg.bin
ModuleID[2] LevelID[1]: Write failed
ModuleID[2] LevelID[1]: Failed to initialize memory!
ModuleID[2] LevelID[1]: PortMgrDlg(0)--MxHidDevice--Command Boot excute failed
...

```

I was trying on a faster 64-bit machine too, but this one is back at the office so I can't be sure it's staying in the reset state. Whereas locally I can press the reset button.
Neither system is getting further than this.
Any thoughts please? Or suggestions to debug further? (I can strace and/or wireshark it)

rbejaran, 2018-01-25:
Hi Gents,
For the record, I couldn't make it work (though I didn't tried the unplug-replug USB as mentioned @mateusz- , but anyways that wouldn't have made the trick for me since I wanted to use mfgtools on a continuous integration environment, so manual interaction is forbidden).
What I did instead is that I gave up this mfgtools project, and used lower-level utilites utp_com and usb_imx_loader as described in this tutorial: https://community.nxp.com/thread/441563
@chelliwell , I'm not sure this is a problem but I find strange that your log says your u-boot file is a .bin. Mine was .imx, as the others' too. Are you sure this is the right file? At this point mfgtools will copy the "temporary" u-boot (along with the "temporary" kernel) that will run on RAM. It's not the image you want to enventually right on the device FLASH.
Cheers.

chelliwell, 2018-01-25:
I'll double-check but yes, I'm pretty sure my u-boot's are correct - all the same images work with the Windows version. (It's built in a Yocto environment, so the extension is set there). And I don't think I'm even getting as far as anything trying to execute yet!
It does seem odd: presumably it does work somewhere, for some people. But a lot of people have problems getting it to work.
What's the 'provenance' of this project? NXP don't provide 'their' Linux source, but here's something very similar (from someone at NXP?) - how close are the versions, and how close is this project to the Windows version? As mentioned in my other thread - for me the official NXP Linux tool works even _less_..!!
(I'll take a look at imx_usb_loader/utp_com)