nxp-imx/mfgtools #21: mfgtoolcli never exit

5frank, 2018-01-25:
mfgtoolcli never exit. It seems to flash the device successfully, then it hangs forever.
The `--noexit` flag is not set in this case. 
gdb backtrace after mfgtoolcli is killed by keyboard interrupt signal:
```
#0  0x00007ffff6e5ef2d in nanosleep () at ../sysdeps/unix/syscall-template.S:81
#1  0x00007ffff6e5edc4 in __sleep (seconds=0) at ../sysdeps/unix/sysv/linux/sleep.c:137
#2  0x000000000040959a in main (argc=16, argv=0x7fffffffe598) at mfgtoolCLI.cpp:422
```
Relevant lines last in main() function in mfgtoolCLI.cpp:
...
420     while(true) 
421     { 
422         sleep(3); 
423     } 
424  
425 } 
```

I am guessing the main thread should wait on working thread(s) to finish and then exit, but I do not find where, so perhaps I am mistaken?

nefethael, 2018-05-15:
mfgtoolcli is spamming "no brakes Could not stop operation" in my case ...

It seems to be unmaintained, maybe i'll switch to this [method](https://community.nxp.com/thread/441563)

5frank, 2018-05-15:
@nefethael I started on a python script that converts ucl2.xml to shell scirpts that uses the method you linked. I have'nt had the time to test it but perhaps you find it useful. Here it is https://gist.github.com/5frank/d5835a580d9fbe66b63b09b5a360ae40

5frank, 2018-05-29:
@nxpfrankli could you please explain why this issue is closed and labeled "wontfix"? is there a workaround? are you unable to reproduce the problem?

nefethael, 2018-05-29:
@5frank i assume old mfgtool is depecrated now and won't be maintained anymore :(

nxpfrankli, 2018-05-29:
Yes, we will use uuu to replace old mfgtools.
 
uuu is really the same source code between linux and windows.
uuu is more simple implement. 

For legacy i.MX6/7 platform,  uuu archive the same function