nxp-imx/mfgtools #28: mfgtool doesn't wait for command to finish before starting next command

takooakes, 2018-04-17:
For our build we format the eMMC and then push our build to it. If I run mfgtoocli normally it gets to the point where it is pushing over our files and hangs. If I run it in a debugger and break at COpCmd_Push::ExecuteCommand and wait for the eMMC reformat command to report success (UTP: sending Success to kernel for command) then it works. It looks like the library isn't waiting for the device to report back before moving on.

Koltak, 2018-04-20:
Can confirm, we are running into the same or at least a similar problem.
The problem seems to lie with the `pipe` command, which we used like this:

```xml
<CMD state="Updater" type="push" body="pipe gunzip > /dev/mmcblk3" file="files/foo.bin.gz"> foo </CMD>
```

yinonlevy, 2018-05-10:
Hi,
Somebody know what cause this problem and if there a way to workaround this bug .
I need to load 1GB to the emmc .
Thanks Yinon