nxp-imx/mfgtools #31: Build error undefined reference to symbol 'inflateEnd'

falstaff84, 2018-07-03:
Build error when building on a Arch Linux machine:

```
[ 93%] Building CXX object uuu/CMakeFiles/uuu.dir/uuu.cpp.o
[ 96%] Linking CXX executable uuu
/usr/bin/ld: ../libuuu/libuuc_s.a(zip.cpp.o): undefined reference to symbol 'inflateEnd'
/usr/lib/libz.so.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
```

nxpfrankli, 2018-07-05:
What's you zlib version?

falstaff84, 2018-07-05:
$ pacman -Q zlib
    zlib 1:1.2.11-2

nxpfrankli, 2018-07-05:
It is same as me. 

objdump -TC /lib/x86_64-linux-gnu/libz.so.1 | grep End
000000000000d300 g    DF .text  0000000000000076  Base        inflateEnd
0000000000009830 g    DF .text  0000000000000036  ZLIB_1.2.0  inflateBackEnd
0000000000007c00 g    DF .text  000000000000011e  Base        deflateEnd

falstaff84, 2018-07-05:
Looks similar:

    $ objdump -TC /lib/libz.so.1 | grep End
    0000000000005f00 g    DF .text  00000000000000cf  Base        deflateEnd
    000000000000a050 g    DF .text  0000000000000067  Base        inflateEnd
    0000000000007800 g    DF .text  0000000000000036  ZLIB_1.2.0  inflateBackEnd

falstaff84, 2018-07-05:
Hm, actually, CMake tries to link libzip, not libz:

    ....
    CMakeFiles/uuu.dir/link.txt --verbose=1              
    /usr/bin/c++  -std=c++11 -pthread -static-libstdc++ -static-libgcc  -rdynamic CMakeFiles/uuu.dir/uuu.cpp.o  -o uuu  -L/home/ags/projects/toradex/imx/mfgtools/uuu/libuuu -Wl,- rpath,/home/ags/projects/toradex/imx/mfgtools/uuu/libuuu ../libuuu/libuuc_s.a -lusb-1.0 -lzip            
    /usr/bin/ld: ../libuuu/libuuc_s.a(zip.cpp.o): undefined reference to symbol 'inflateEnd'                                                   
    /usr/lib/libz.so.1: error adding symbols: DSO missing from command line                                                                    
    collect2: error: ld returned 1 exit status

Which actually is missing that symbol:

    $ objdump -TC /lib/libzip.so | grep End
    0000000000000000      DF *UND*  0000000000000000              inflateEnd
    0000000000000000      DF *UND*  0000000000000000              deflateEnd
    0000000000000000      DF *UND*  0000000000000000              BZ2_bzDecompressEnd
    0000000000000000      DF *UND*  0000000000000000              BZ2_bzCompressEnd

Using libz fixes the issue for me, but not sure if that is the proper fix:

    pkg_check_modules(LIBZIP REQUIRED libz)

nxpfrankli, 2018-07-05:
--- a/libuuu/CMakeLists.txt
+++ b/libuuu/CMakeLists.txt
@@ -2,7 +2,7 @@ cmake_minimum_required (VERSION 2.6)

 find_package(PkgConfig REQUIRED)
 pkg_check_modules(LIBUSB REQUIRED libusb-1.0>=1.0.16)
-pkg_check_modules(LIBZIP REQUIRED libzip)
+pkg_check_modules(LIBZIP REQUIRED zlib)
 find_package(Threads)

nxpfrankli, 2018-07-09:
Did my change fix your problem?

falstaff84, 2018-07-10:
No, still the same:

$ make                                                                                  [2]
    -- Checking for module 'zlib'
    --   Found zlib, version 1.2.11
    -- Checking for module 'libzip'
    --   Found libzip, version 1.5.1
    -- Configuring done
    -- Generating done
    ...
    [ 96%] Linking CXX executable uuu
    /usr/bin/ld: ../libuuu/libuuc_s.a(zip.cpp.o): undefined reference to symbol 'inflateEnd'
    /usr/lib/libz.so.1: error adding symbols: DSO missing from command line
    collect2: error: ld returned 1 exit status

nxpfrankli, 2018-11-05:
Someone send pull request to fix built error in ARCH. 
Can you help check if this problem still exist at latest commit?

closed 2018-11-07 by agners