nxp-imx/mfgtools #35: Missing USB/UTP support in uuu?

sergioprado, 2018-07-10:
Hello,

Mfgtools (mfgtoolcli) has the support to run commands on the target using the USB/UTP protocol. I didn't find this feature in uuu.

Is it possible to run commands on the target with uuu? If so, how?

nxpfrankli, 2018-07-10:
UTP will not be supportted.  We will use FBK protocol.  

https://github.com/codeauroraforum/imx-uuc/blob/master/ufb.c

sergioprado, 2018-07-10:
Understood, thanks.

Is there any documentation about this protocol? Do I have to enable any specific feature in the Linux kernel?

nxpfrankli, 2018-07-11:
low level protocol 
https://android.googlesource.com/platform/system/core/+/2ec418a4c98f6e8f95395456e1ad4c2956cac007/fastboot/fastboot_protocol.txt

high level protocol 
https://github.com/codeauroraforum/mfgtools/wiki

closed 2018-07-11 by nxpfrankli