nxp-imx/mfgtools #36: New jump syntax in uuu [linux]

Grinimod, 2018-07-10:
Hi,
I have the same boot structure as @Koltak, and I want to port his  script:
```
<CMD state="BootStrap" type="boot" body="BootStrap"
	file="firmware/u-boot.imx">Loading U-boot</CMD>
<CMD state="BootStrap" type="load" file="firmware/zImage"
	address="0x12000000" loadSection="OTH" setSection="OTH"
	HasFlashHeader="FALSE">Loading Kernel.</CMD>
<CMD state="BootStrap" type="load"
	file="firmware/zImage.dtb" address="0x18000000"
	loadSection="OTH" setSection="OTH" HasFlashHeader="FALSE">Loading device
	tree.</CMD>
<CMD state="BootStrap" type="load"
	file="firmware/image.cpio.gz.u-boot"
	address="0x12C00000" loadSection="OTH" setSection="OTH"
	HasFlashHeader="FALSE">Loading Initramfs.</CMD>
<CMD state="BootStrap" type="jump" >Jumping to OS image.</CMD>
```

Using the provided examples I came up with (shell mode):
```
SDP: boot -f u-boot-signed.imx -nojump
SDP: write -f zImage -addr 0x12000000
SDP: write -f zImage.dtb -addr 0x18000000
SDP: write -f image.cpio.gz.u-boot -addr 0x12C00000
```

This seems to work, but I have no clue how to use the jump afterwards properly. I tried it with the -addr and -f options and all feasible arguments, but I didnt get any result other than segfaults.

Letting out the -nojump flag will result in the startup of the device after line 1, but there is no kernel etc.

Thanks for your help!

nxpfrankli, 2018-07-10:
sdp: jump -f u-boot.imx -ivt 0

nxpfrankli, 2018-07-10:
added  "sdp: jump -f u-boot.imx -ivt 0" after lines

nxpfrankli, 2018-07-10:
Actually, we prefer enable uboot's fastboot. Most task can be complete by uboot's fastboot protocol.