nxp-imx/mfgtools #37: Poor file transfer performance on windows mfgtools v2.8?

thkaw, 2018-07-13:
Hi,

I found when I using mfgtools 2.6.2, the transfer speed can reach 18MB/s

But I compile latest windows mfgtools(v2.8). The speed drop to only around 2.8MB/s

Is any change make this result?

Thanks

nxpfrankli, 2018-07-13:
Not sure,  Can you try our latest uuu tools?  
Which platform do you use?

thkaw, 2018-07-16:
My compile platform is Windows.

But I also try 2.3.2 source code to compile with Visual Studio 2017 Enterprise+Windows 10 latest version.
The speed still low.

I don't find 2.6.2 source code that I can't verify the problem is.

Is there need adjust any special compiler parameter to compile?

I will try to compile uuu now.

thkaw, 2018-07-19:
@nxpfrankli We got the problem. We release and testing in wrong build profile.

Should using RELEASE build, but accident using DEBUG build when compile project.

closed 2018-07-19 by thkaw