nxp-imx/mfgtools #42: Communication via FBK in uuu/ufb

Grinimod, 2018-07-20:
Hi,
currently I am not able to receive/process FBK (ucmd) messages on my device. The first sent data package is 
```
0000   80 74 0b fb 1a 95 ff ff 53 03 01 0a 01 00 2d 00
0010   8f dc 51 5b 00 00 00 00 b9 c0 02 00 8d ff ff ff
0020   0f 00 00 00 0f 00 00 00 00 00 00 00 00 00 00 00
0030   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0040   55 43 6d 64 3a 65 63 68 6f 20 48 65 6c 6c 6f
```
(USB URB_BULK captured by wireshark)
of which the last line translates to the command
`UCmd:echo Hello`
Running the ufb client on the imx  will result in reading from /dev/utp*
 forever in the main-loop.

I expect my custom linuxrc to be insufficient, so I would kindly ask for a minimal example utilizing only uuu  (single UDC)

FYI: I am using yocto krogoth but an update to rocko is pending

nxpfrankli, 2018-07-20:
What's type storage do you want to program. 
Most storage program can be finished in uboot. 

        mkdir /sys/kernel/config/usb_gadget/g1
	cd /sys/kernel/config/usb_gadget/g1
	echo 0x066F > idVendor
	echo 0x9BFF > idProduct
	mkdir strings/0x409

	if [ -e /sys/devices/soc0/soc_uid ]; then
		cat /sys/devices/soc0/soc_uid > strings/0x409/serialnumber
	else
		echo 0000000000000000 > strings/0x409/serialnumber
	fi

	echo "FSL i.MX Board" > strings/0x409/product
	mkdir configs/c.1
	echo 5 > configs/c.1/MaxPower

	echo 1 > os_desc/use
	echo "MSFT100" > os_desc/qw_sign
	echo 0x40 > os_desc/b_vendor_code

	mkdir functions/ffs.utp
	mkdir /dev/usb-utp
	mount -t functionfs utp /dev/usb-utp
	ln -s functions/ffs.utp configs/c.1/
	ln -s configs/c.1 os_desc

# you can check here if there are /dev/usb-utp/ep0
# if no, kernel have not built in ffs gadget support. 

        ufb /dev/usb-utp/ep0 &

	echo run utp at /dev/usb-utp$2/ep0;
	while [ ! -e /dev/usb-utp/ep1 ]
	do
		echo "."
		sleep 1;
	done

	echo  "you udc gadget controller"  > UDC

Grinimod, 2018-07-23:
I tried 
```
#!/bin/sh
export PATH=/sbin:/bin:/usr/sbin:/usr/bin

mount -t sysfs none /sys
mount -t proc none /proc
mount -t devtmpfs none /dev
mount -t configfs none /sys/kernel/config

modprobe sysflags

# disable turn off display
echo -e "\033[9;0]" > /dev/tty0

#mkdir -p /myconf/config/usb_gadget/g1
#cd /myconf/config/usb_gadget/g1
mkdir -p /sys/kernel/config/usb_gadget/g1
cd /sys/kernel/config/usb_gadget/g1

echo 0x066F > idVendor
echo 0x9BFF > idProduct
mkdir -p strings/0x409
echo 0000000000000000 > strings/0x409/serialnumber
echo "FSL i.MX Board" > strings/0x409/product
mkdir -p configs/c.1
echo 5 > configs/c.1/MaxPower

mkdir os_desc
echo 1 > os_desc/use
echo "MSFT100" > os_desc/qw_sign
echo 0x40 > os_desc/b_vendor_code

mkdir -p functions/ffs.utp
mkdir  /dev/usb-utp
mount -t functionfs utp /dev/usb-utp
ln -s functions/ffs.utp configs/c.1/
ln -s configs/c.1 os_desc
```
resulting in
```
mkdir: can't create directory 'functions/ffs.utp': No such file or directory
mount: mounting utp on /dev/usb-utp failed: No such device
ln: configs/c.1/ffs.utp: No such file or directory
```
Switching the config location will only leave the mount failing. 

To answer your questions above, there is eMMC to flash: fuses to burn and partitions to fill and tests to run, which can be done atm by ucl2 scripts in the old MFGTool on windows.

Many thanks!

nxpfrankli, 2018-07-23:
Check your kernel config if CONFIG_USB_F_FS=y

Grinimod, 2018-07-24:
It was initially not set, setting it to yes seems to not have any impact.

The full initial USB kernel config was
```
$ grep -Rh "CONFIG_USB" .|grep -v "#"|sort|uniq
CONFIG_USB_ARCH_HAS_HCD=y
CONFIG_USB_CHIPIDEA_HOST=y
CONFIG_USB_CHIPIDEA_OF=y
CONFIG_USB_CHIPIDEA_PCI=y
CONFIG_USB_CHIPIDEA_UDC=y
CONFIG_USB_CHIPIDEA=y
CONFIG_USB_COMMON=y
CONFIG_USB_DEFAULT_PERSIST=y
CONFIG_USB_EHCI_HCD=y
CONFIG_USB_EHCI_MXC=y
CONFIG_USB_EHCI_PCI=y
CONFIG_USB_EHCI_ROOT_HUB_TT=y
CONFIG_USB_EHCI_TT_NEWSCHED=y
CONFIG_USB_F_MASS_STORAGE=y
CONFIG_USB_GADGET_STORAGE_NUM_BUFFERS=2
CONFIG_USB_GADGET_VBUS_DRAW=2
CONFIG_USB_GADGET=y
CONFIG_USB_HID=y
CONFIG_USB_LIBCOMPOSITE=y
CONFIG_USB_MASS_STORAGE=y
CONFIG_USB_MXS_PHY=y
CONFIG_USB_OHCI_LITTLE_ENDIAN=y
CONFIG_USB_PHY=y
CONFIG_USB_SUPPORT=y
CONFIG_USB=y
```

nxpfrankli, 2018-07-24:
Missed CONFIG_USB_CONFIGFS_F_FS

CONFIG_USB_ACM=m
CONFIG_USB_ARCH_HAS_HCD=y
CONFIG_USB_ARMLINUX=y
CONFIG_USB_AUDIO=m
CONFIG_USB_BELKIN=y
CONFIG_USB_CDC_COMPOSITE=m
CONFIG_USB_CDNS3_GADGET=y
CONFIG_USB_CDNS3_HOST=y
CONFIG_USB_CDNS3=y
CONFIG_USB_CHIPIDEA_HOST=y
CONFIG_USB_CHIPIDEA_OF=y
CONFIG_USB_CHIPIDEA_PCI=y
CONFIG_USB_CHIPIDEA_UDC=y
CONFIG_USB_CHIPIDEA=y
CONFIG_USB_COMMON=y
CONFIG_USB_CONFIGFS_ACM=y
CONFIG_USB_CONFIGFS_ECM_SUBSET=y
CONFIG_USB_CONFIGFS_ECM=y
CONFIG_USB_CONFIGFS_EEM=y
**CONFIG_USB_CONFIGFS_F_FS=y**
CONFIG_USB_CONFIGFS_F_HID=y
CONFIG_USB_CONFIGFS_F_LB_SS=y
CONFIG_USB_CONFIGFS_F_MIDI=y
CONFIG_USB_CONFIGFS_F_UAC1=y
CONFIG_USB_CONFIGFS_F_UAC2=y
**CONFIG_USB_CONFIGFS_MASS_STORAGE=y**
CONFIG_USB_CONFIGFS_NCM=y
CONFIG_USB_CONFIGFS_OBEX=y
CONFIG_USB_CONFIGFS_RNDIS=y
CONFIG_USB_CONFIGFS_SERIAL=y
CONFIG_USB_CONFIGFS=y
CONFIG_USB_DEFAULT_PERSIST=y
CONFIG_USB_DWC2_DUAL_ROLE=y
CONFIG_USB_DWC2=y
CONFIG_USB_DWC3_DUAL_ROLE=y
CONFIG_USB_DWC3_EXYNOS=y
CONFIG_USB_DWC3_OF_SIMPLE=y
CONFIG_USB_DWC3_PCI=y
CONFIG_USB_DWC3=y
CONFIG_USB_EHCI_EXYNOS=y
CONFIG_USB_EHCI_HCD_PLATFORM=y
CONFIG_USB_EHCI_HCD=y
CONFIG_USB_EHCI_MSM=y
CONFIG_USB_EHCI_PCI=y
CONFIG_USB_EHCI_ROOT_HUB_TT=y
CONFIG_USB_EHCI_TT_NEWSCHED=y
CONFIG_USB_EHSET_TEST_FIXTURE=y
CONFIG_USB_ETH_EEM=y
CONFIG_USB_ETH=m
CONFIG_USB_ETH_RNDIS=y
CONFIG_USB_F_ACM=y
CONFIG_USB_F_ECM=y
CONFIG_USB_F_EEM=y
**CONFIG_USB_F_FS=y**
CONFIG_USB_F_HID=y
**CONFIG_USB_F_MASS_STORAGE=y**
CONFIG_USB_F_MIDI=y
CONFIG_USB_F_NCM=y
CONFIG_USB_F_OBEX=y
CONFIG_USB_F_RNDIS=y
CONFIG_USB_F_SERIAL=y
CONFIG_USB_F_SS_LB=y
CONFIG_USB_F_SUBSET=y
CONFIG_USB_F_UAC1=y
CONFIG_USB_F_UAC2=y
CONFIG_USB_GADGET_STORAGE_NUM_BUFFERS=2
CONFIG_USB_GADGET_VBUS_DRAW=2
CONFIG_USB_GADGET=y
CONFIG_USB_G_NCM=m
CONFIG_USB_GPIO_VBUS=y
CONFIG_USB_G_SERIAL=m
CONFIG_USB_GSPCA=m
CONFIG_USB_HCD_TEST_MODE=y
CONFIG_USB_HID=y
CONFIG_USB_HSIC_USB3503=y
CONFIG_USB_ISP1760_HCD=y
CONFIG_USB_ISP1760_HOST_ROLE=y
CONFIG_USB_ISP1760=y
CONFIG_USB_LIBCOMPOSITE=y
CONFIG_USB_MASS_STORAGE=m
CONFIG_USB_MXS_PHY=y
CONFIG_USB_NET_AX88179_178A=m
CONFIG_USB_NET_AX8817X=m
CONFIG_USB_NET_CDCETHER=m
CONFIG_USB_NET_CDC_NCM=m
CONFIG_USB_NET_CDC_SUBSET_ENABLE=m
CONFIG_USB_NET_CDC_SUBSET=m
CONFIG_USB_NET_DM9601=m
CONFIG_USB_NET_DRIVERS=y
CONFIG_USB_NET_MCS7830=m
CONFIG_USB_NET_NET1080=m
CONFIG_USB_NET_PLUSB=m
CONFIG_USB_NET_SMSC75XX=m
CONFIG_USB_NET_SMSC95XX=m
CONFIG_USB_NET_SR9800=m
CONFIG_USB_NET_ZAURUS=m
CONFIG_USB_OHCI_EXYNOS=y
CONFIG_USB_OHCI_HCD_PCI=y
CONFIG_USB_OHCI_HCD_PLATFORM=y
CONFIG_USB_OHCI_HCD=y
CONFIG_USB_OHCI_LITTLE_ENDIAN=y
CONFIG_USB_OTG_WHITELIST=y
CONFIG_USB_OTG=y
CONFIG_USB_PEGASUS=m
CONFIG_USB_PHY=y
CONFIG_USB_RENESAS_USBHS=m
CONFIG_USB_RENESAS_USBHS_UDC=m
CONFIG_USB_RTL8150=m
CONFIG_USB_RTL8152=m
CONFIG_USB_STORAGE=y
CONFIG_USB_SUPPORT=y
CONFIG_USB_TEST=m
CONFIG_USB_U_ETHER=y
CONFIG_USB_ULPI_VIEWPORT=y
CONFIG_USB_ULPI=y
CONFIG_USB_USBNET=m
CONFIG_USB_U_SERIAL=y
CONFIG_USB_VIDEO_CLASS_INPUT_EVDEV=y
CONFIG_USB_VIDEO_CLASS=m
CONFIG_USB_XHCI_HCD=y
CONFIG_USB_XHCI_PCI=y
CONFIG_USB_XHCI_PLATFORM=y
CONFIG_USB_XHCI_RCAR=y
CONFIG_USB_XHCI_TEGRA=y
CONFIG_USB=y
CONFIG_USB_ZERO=m

Grinimod, 2018-07-25:
Problem lies in kernel config, my changes never made it into the build/.config, because
```
Value of CONFIG_USB_CONFIGFS is redefined by fragment ***/usb-otg.cfg.sanitized:
Previous value: # CONFIG_USB_CONFIGFS is not set
New value: CONFIG_USB_CONFIGFS=y
```
but somehow the changes got overridden (merge_log.txt & mismatch.txt)
```
Value requested for CONFIG_USB_CONFIGFS not in final .config
Requested value:  CONFIG_USB_CONFIGFS=y
Actual value:     # CONFIG_USB_CONFIGFS is not set
```
As CONFIG_USB_CONFIGFS is not set, all following _* are not even mentioned, which is consistent.
Tried to manipulate the .config and start the kernel package at `-C compile`, but that left me with numerous inconsistent choice states. 

I tried `-c menuconfig`, but was not able to locate the parameters that way yet.

A last trial could be to inject a whole working .config  (+apply needed customization patches for own kernel modules) to continue the flow and fix the kernel config build afterwards.

Grinimod, 2018-07-26:
Got it corrected via:
1. Path in menuconfig
Device Drivers
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;USB support
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;USB gadget support *(very last entry)*
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;USB Gadget Drivers (...)
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;USB functions configurable through configfs
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;**Mass storage**
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;**Function filesystem (functionFS)**

2. Diffed the result and added that to usb-otg.cfg

3. build/.config now contains

CONFIG_USB_HID=y
CONFIG_USB_OHCI_LITTLE_ENDIAN=y
CONFIG_USB_SUPPORT=y
CONFIG_USB_COMMON=y
CONFIG_USB_ARCH_HAS_HCD=y
CONFIG_USB=y
CONFIG_USB_DEFAULT_PERSIST=y
CONFIG_USB_EHCI_HCD=y
CONFIG_USB_EHCI_ROOT_HUB_TT=y
CONFIG_USB_EHCI_TT_NEWSCHED=y
CONFIG_USB_EHCI_PCI=y
CONFIG_USB_EHCI_MXC=y
CONFIG_USB_CHIPIDEA=y
CONFIG_USB_CHIPIDEA_OF=y
CONFIG_USB_CHIPIDEA_PCI=y
CONFIG_USB_CHIPIDEA_UDC=y
CONFIG_USB_CHIPIDEA_HOST=y
CONFIG_USB_PHY=y
CONFIG_USB_MXS_PHY=y
CONFIG_USB_GADGET=y
CONFIG_USB_GADGET_VBUS_DRAW=2
CONFIG_USB_GADGET_STORAGE_NUM_BUFFERS=2
CONFIG_USB_LIBCOMPOSITE=y
**CONFIG_USB_F_MASS_STORAGE=y**
**CONFIG_USB_F_FS=y**
CONFIG_USB_CONFIGFS=y
**CONFIG_USB_CONFIGFS_MASS_STORAGE=y**
**CONFIG_USB_CONFIGFS_F_FS=y**

In the end it did not solve the mount problem:
```
Starting kernel ...

/proc/version
Linux version 4.1.15 (Grinimod@pc) (gcc version 5.3.0 (GCC) ) #1 SMP PREEMPT Thu Jul 26 10:17:33 CEST 2018
mount: mounting utp on /dev/usb-utp failed: No such device
/dev/usb-utp
drwxr-xr-x    2 root     root            40 Jan  1 00:00 .
drwxr-xr-x    5 root     root          2560 Jan  1 00:00 ..
```

nxpfrankli, 2018-07-26:
Good So I can close this issue

closed 2018-07-26 by nxpfrankli