nxp-imx/mfgtools #46: Bad Linux ARM zImage magic!

lrtxpra, 2018-08-29:
I have a problem about migrate xml file to uuu script.
according to the wiki "migration from ucl2.xml to uuu script"
I converted my xml file to uuu script, but when I run " ./uuu my_uuu_script "
the minicom display "Bad Linux ARM zImage magic!".

when I use mfgtool on windows, it runs ok, I don't know if there are some env parameters related to mfgtool that not be set. or just the problem about kernel.

the uuu scriipt is
```
uuu_version 1.0.1
SDP: boot -f firmware/uboot-myimx6ek200-6q.imx
FB: ucmd echo boot from uboot

FB: ucmd setenv fastboot_buffer ${loadaddr}
FB: ucmd download -f firmware/mfg-zImage-myimx6

FB: ucmd setenv fastboot_buffer ${initrd_addr}
FB: ucmd download -f firmware/initramfs-imx_mfgtools.cpio.gz.u-boot

FB: ucmd setenv fastboot_buffer ${fdt_addr}
FB: ucmd download -f firmware/myimx6ek200-6q.dtb

FB: acmd booti ${loadaddr} ${initrd_addr} ${fdt_addr}

FBK: ucmd flash_erase /dev/mtd0 0 0

FBK: ucp files/uboot-myimx6ek200-6q.imx t:/tmp
FBK: ucmd dd if=/tmp/uboot-myimx6ek200-6q.imx of=/dev/mtd0 bs=512 seek=2

FBK: ucp firmware/mksdcard-31452.sh.tar t:/tmp
FBK: ucmd tar xf /tmp/uboot-myimx6ek200-6q.imx 
FBK: ucmd sh mksdcard.sh /dev/mmcblk3

FBK: ucmd mkfs.vfat /dev/mmcblk3p1
FBK: ucmd mkdir -p /mnt/mmcblkxp1
FBK: ucmd mount -t vfat /dev/mmcblk3p1 /mnt/mmcblkxp1

FBK: ucp files/zImage-myimx6 t:/tmp
FBK: ucmd cp /tmp/zImage-myimx6 /mnt/mmcblkxp1/zImage-myimx6
FBK: ucp myimx6ek200-6q.dtb t:/tmp
FBK: ucmd cp /tmp/myimx6ek200-6q.dtb /mnt/mmcblkxp1/myimx6ek200-6q.dtb
FBK: ucmd umount /mnt/mmcblkxp1

FBK: ucmd mkfs.ext3 -j /dev/mmcblk3p2
FBK: ucmd mkdir -p /mnt/mmcblkxp2
FBK: ucmd mount -t ext3 /dev/mmcblk3p2 /mnt/mmcblkxp2

FBK: acmd pipe tar -jxv -C /mnt/mmcblkxp2
FBK: ucp files/armhf-rootfs-ubuntu-xenial.tar.bz2 t:-
FBK: sync
FBK: acmd pipe tar -jxv -C /mnt/mmcblkxp2
FBK: ucp files/kernel-modules-myimx6.tar.bz2 t:-
FBK: sync
FBK: acmd pipe tar -jxv -C /mnt
FBK: ucp files/myzr_rtfs_cfg.tar.bz2 t:-
FBK: sync
FBK: ucmd sh /mnt/myzr_rtfs_cfg.sh /mnt/mmcblkxp2 myimx6ek200 armhf-rootfs-ubuntu-xenial.tar.bz2 normal
FBK: sync
FBK: acmd pipe tar -jxv -C /mnt/mmcblkxp2
FBK: ucp files/l31452_rtfs_update.tar.bz2 t:-
FBK: sync
FBK: acmd pipe tar -jxv -C /mnt/mmcblkxp2/home/root
FBK: ucp files/my-demo.tar.bz2 t:-
FBK: sync
FBK: ucmd umount /mnt/mmcblkxp2
FBK: ucmd echo Update Complete!
```

and the boot output 
```
U-Boot 2015.04-svn86 (Aug 09 2017 - 10:21:01)                                   
                                                                                
CPU:   Freescale i.MX6Q rev1.5 at 792 MHz                                       
CPU:   Temperature 44 C                                                         
Reset cause: POR                                                                
Board: MYZR i.MX6 Evaluation Kit                                                
Model: MY-IMX6-EK200-6Q-1G                                                      
I2C:   ready                                                                    
DRAM:  1 GiB                                                                    
MMC:   FSL_SDHC: 0, FSL_SDHC: 1                                                 
SF: Detected SST25VF016B with page size 256 Bytes, erase size 4 KiB, total 2 MiB
*** Warning - bad CRC, using default environment                                
                                                                                
No panel detected: default to Hannstar-XGA                                      
Display: Hannstar-XGA (1024x600)                                                
In:    serial                                                                   
Out:   serial                                                                   
Err:   serial                                                                   
Net:   FEC [PRIME]                                                              
Boot from USB for mfgtools                                                      
Use default environment for                              mfgtools               
Run bootcmd_mfg: run mfgtool_args; bootz ${loadaddr} ${initrd_addr} ${fdt_addr};
Hit any key to stop autoboot:  0                                                
Bad Linux ARM zImage magic! 
```

nxpfrankli, 2018-08-29:
you need enable fastboot in uboot.  Old uboot don't support these.  
But you can try this ways

SDP:  dcd -f uboot-myimx6ek200-6q.imx
SDP:  write  -ivt 0 -f uboot-myimx6ek200-6q.imx
SDP:  write -addr <your kernel load address> -f zImage
SDP:  write -addr <your initramfs address> -f initramfs
SDP:  write -addr <your dtb address> -f dtbfile
SDP:  jump -ivt 0 -f uboot-myimx6ek200-6q.imx

Follow all FBK: commands.

lrtxpra, 2018-08-29:
OK，thanks first
it runs ok,
but there is a new problem,
it stucks at 
I think it because uuu uses the FBK. and files are compiled for mfgtools to use.
```
Starting UTP                                                                    
uuc 0.5 [built Dec  3 2015 08:26:46]                                            
UTP: Waiting for device to appear                                               
random: nonblocking pool is initialized
```

nxpfrankli, 2018-08-29:
Yes, you need update initramfs include imx-uuc latest change.

closed 2018-09-07 by nxpfrankli