nxp-imx/mfgtools #62: imx8mmevk libusb error timeout

austinf, 2018-10-25:
When trying to load code onto sd card on the imx8mmevk board, with the following command:

`sudo ./uuu/uuu -b sd_all ~/imx-boot-imx8mmevk-sd.bin ~/core-image-minimal-imx8mmevk.sdcard
`
I get:
```
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.1.87-3-g9d88a8c

Success 0    Failure 1

1:1    1/ 1   [HID(W):LIBUSB_ERROR_TIMEOUT ] SDP: boot -f /home/austinf/imx-boot-imx8mmevk-sd.bin
```

Not sure how to debug what's going wrong further.

lznuaa, 2018-10-25:
The early imx8m chip need apply a ROM patch.   windows system have not such problem.

austinf, 2018-10-25:
My only windows system is 32 bit, can 32 bit windows binaries be provided? The ones on the release page appear to be 64 bit only.

Building from source on windows requires quite a bit of environment setup it looks like.

nxpfrankli, 2018-10-25:
We don't prefer use 32bit system, because it will be problem when download big files.

nxpfrankli, 2018-10-25:
did you boot from other boot source? 
check if rom patch locked 
u-boot=> fuse read 0 0

austinf, 2018-10-25:
I can boot off an sdcard.

u-boot=> fuse read 0 0
Reading bank 0:

Word 0x00000000: b0010103

How do I interpret that result?

nxpfrankli, 2018-11-05:
this is early chip problem.  Someone report there are not such problem at  ubuntu 18.04

closed 2018-11-05 by nxpfrankli