nxp-imx/mfgtools #66: convert an old mfgtools flash script : miss usage documentation

JLMorange, 2018-11-27:
Hi,
I have an old mfgtools script to move to latest 3.0 format, but the documentation is really ligth.... so I don't know what are the commands, I can use

the script is as follow : 

`<UCL><CFG><STATE name="BootStrap" dev="MX7D" vid="15A2" pid="0076"/><STATE name="Updater" dev="MSC" vid="066F" pid="37FF"/></CFG><LIST name="eMMC" desc="Choose eMMC as media"><CMD state="BootStrap" type="boot" body="BootStrap" file="firmware/u-boot-imx7d%7duboot%_sd.imx" ifdev="MX7D">Loading U-boot</CMD><CMD state="BootStrap" type="load" file="firmware/zImage" address="0x80800000" loadSection="OTH" setSection="OTH" HasFlashHeader="FALSE" ifdev="MX6SL MX6SX MX7D MX6UL">Loading Kernel.</CMD><CMD state="BootStrap" type="load" file="firmware/%initramfs%" address="0x83800000" loadSection="OTH" setSection="OTH" HasFlashHeader="FALSE" ifdev="MX6SL MX6SX MX7D MX6UL">Loading Initramfs.</CMD><CMD state="BootStrap" type="load" file="firmware/zImage-imx7d-%7ddtb%.dtb" address="0x83000000" loadSection="OTH" setSection="OTH" HasFlashHeader="FALSE" ifdev="MX7D">Loading device tree.</CMD><CMD state="BootStrap" type="jump"> Jumping to OS image. </CMD><CMD state="Updater" type="push" body="send" file="mksdcard.sh">Sending partition shell</CMD><CMD state="Updater" type="push" body="$ sh $FILE /dev/mmcblk%mmc%"> Partitioning...</CMD><CMD state="Updater" type="push" body="$ dd if=/dev/zero of=/dev/mmcblk%mmc% bs=1k seek=384 conv=fsync count=192">Clear u-boot environment</CMD><CMD state="Updater" type="push" body="$ dd if=/dev/zero of=/dev/mmcblk%mmc% bs=1048576 seek=10 count=1 conv=fsync">Clear bootloader bootflags</CMD><CMD state="Updater" type="push" body="$ printf A       | dd of=/dev/mmcblk%mmc% seek=10485760 bs=1 count=1 conv=fsync">Set active slot to A</CMD><CMD state="Updater" type="push" body="$ printf '\\x03' | dd of=/dev/mmcblk%mmc% seek=10486272 bs=1 count=1 conv=fsync">Set slot A boot count to 3</CMD><CMD state="Updater" type="push" body="$ printf '\\x00' | dd of=/dev/mmcblk%mmc% seek=10486784 bs=1 count=1 conv=fsync">Set slot B boot count to 0</CMD><CMD state="Updater" type="push" body="$ printf Y       | dd of=/dev/mmcblk%mmc% seek=10487296 bs=1 count=1 conv=fsync">Set bootable flag of A to Y</CMD><CMD state="Updater" type="push" body="$ printf N       | dd of=/dev/mmcblk%mmc% seek=10487808 bs=1 count=1 conv=fsync">Set bootable flag of B to N</CMD><CMD state="Updater" type="push" body="$ printf Y       | dd of=/dev/mmcblk%mmc% seek=10488320 bs=1 count=1 conv=fsync">Set successful flag of A to Y</CMD><CMD state="Updater" type="push" body="$ printf N       | dd of=/dev/mmcblk%mmc% seek=10488832 bs=1 count=1 conv=fsync">Set successful flag of B to N</CMD><CMD state="Updater" type="push" body="$ echo 0 > /sys/block/mmcblk%mmc%boot0/force_ro">access boot partition 1</CMD><CMD state="Updater" type="push" body="send" file="../../../../u-boot-imx7dsabresd.imx" ifdev="MX7D">Sending u-boot.bin</CMD><CMD state="Updater" type="push" body="$ dd if=$FILE of=/dev/mmcblk%mmc%boot0 bs=512 seek=2">write U-Boot to sd card</CMD><CMD state="Updater" type="push" body="$ echo 1 > /sys/block/mmcblk%mmc%boot0/force_ro"> re-enable read-only access </CMD><CMD state="Updater" type="push" body="$ mmc bootpart enable 1 1 /dev/mmcblk%mmc%">enable boot partion 1 to boot</CMD><CMD state="Updater" type="push" body="$ dd conv=fsync if=/dev/zero of=/dev/mmcblk%mmc%p11 bs=1048576 count=10">Cleaning /config partition</CMD><CMD state="Updater" type="push" body="$ dd conv=fsync if=/dev/zero of=/dev/mmcblk%mmc%p12 bs=1048576 count=10">Cleaning /var/log partition</CMD><CMD state="Updater" type="push" body="$ dd if=/dev/zero of=/dev/mmcblk2p14 bs=1 count=64 seek=4112">Clear config partition key blob</CMD><CMD state="Updater" type="push" body="$ dd if=/dev/zero of=/dev/mmcblk%mmc%p7 conv=fsync bs=1048576 count=10">Trashing partition / in slot B</CMD><CMD state="Updater" type="push" body="$ dd if=/dev/zero of=/dev/mmcblk%mmc%p8 conv=fsync bs=1048576 count=10">Trashing partition /opt in slot B</CMD><CMD state="Updater" type="push" body="$ dd if=/dev/zero of=/dev/mmcblk%mmc%p9 conv=fsync bs=1048576 count=10">Trashing partition /usr/local in slot B</CMD><CMD state="Updater" type="push" body="$ dd if=/dev/zero of=/dev/mmcblk%mmc%p10 conv=fsync bs=1048576 count=10">Trashing partition /boot in slot B</CMD><CMD state="Updater" type="push" body="pipe dd of=/dev/mmcblk%mmc%p2 bs=1M conv=fsync" file="../../../../sercomm-image.rootfs.ext4" ifdev="MX6UL MX7D">
		Sending and writing rootfs image</CMD><CMD state="Updater" type="push" body="pipe dd of=/dev/mmcblk%mmc%p3 bs=1M conv=fsync" file="../../../../sercomm-image.opt.ext4" ifdev="MX6UL MX7D">
		Sending and writing image for partition /opt</CMD><CMD state="Updater" type="push" body="pipe dd of=/dev/mmcblk%mmc%p5 bs=1M conv=fsync" file="../../../../sercomm-image.usr-local.ext4" ifdev="MX6UL MX7D">
		Sending and writing image for partition /usr/local</CMD><CMD state="Updater" type="push" body="pipe dd of=/dev/mmcblk%mmc%p6 bs=1M conv=fsync" file="../../../../sercomm-image.boot.ext2" ifdev="MX6UL MX7D">
		Sending and writing image for partition /boot</CMD><CMD state="Updater" type="push" body="frf">Finishing rootfs write</CMD><CMD state="Updater" type="push" body="$ echo Update Complete!">Finished (token XowTBQSc8J21tILGKL2pKptzRuoKnyqo)</CMD></LIST></UCL>`

nxpfrankli, 2018-11-27:
Most case you needn't  use complex scripts.  Some built-in script can burn image into emmc. 

Anyway,  if you have to use linux kernel to do something. Please refer:

https://github.com/NXPmicro/mfgtools/wiki/migration-from-ucl2.xml-to-uuu-script

JLMorange, 2018-12-03:
thanks for the link, it's helpful, I started to convert the script
however the document is a bit strange...  for example for body="BootStrap" commands, there is several entries with several commands... is there a document that explain all the SDPS SDP SDPU FB FBK and variants? what is the difference between a ucmd and a acmd?
thanks and regards
JLM

nxpfrankli, 2018-12-03:
> thanks for the link, it's helpful, I started to convert the script
> however the document is a bit strange... for example for body="BootStrap" commands, there is several entries with several [...]

https://github.com/NXPmicro/mfgtools/wiki/UUU-default-support-protocol-list

nxpfrankli, 2018-12-03:
> according to https://github.com/NXPmicro/mfgtools/wiki/UUU-default-support-protocol-list
> my device (MX7D) use SDP
> so I try to translate
> 
> ```
> 	<CMD state="BootStrap" type="boot" body="BootStrap" file ="firmware/u-boot-imx7d%7duboot%_sd.imx" ifdev="MX7D">Loading U-boot</CMD>
> ```
> seems that the line is the thrid line of array at https://github.com/NXPmicro/mfgtools/wiki/migration-from-ucl2.xml-to-uuu-script
> 
> but
> 
> ```
> SDP: boot -f flash.bin SDP: boot -f flash.bin
> 
> SDPU: write -f flash.bin -offset 0x57c00
> 
> SDPU: jump
> ```
> seems obscure... where does the 0x57c00 comes from?
> why is there 2 SDP: boot -f flash.bin at first line?
> why is there a write command?

Some device(like imx8mm, imx8mq) use SPL,   first SDP: boot, just boot SPL
Second SDPU: write download uboot fit image,  0x57c00 is offset of FIT image in flash.bin

JLMorange, 2018-12-04:
> Some device(like imx8mm, imx8mq) use SPL, first SDP: boot, just boot SPL
ok but why "SDP: boot -f flash.bin SDP: boot -f flash.bin" for me it's the same command copied two times... no SPL in it....

> Second SDPU: write download uboot fit image, 0x57c00 is offset of FIT image in flash.bin
ok, It would be nice to explain it in https://github.com/NXPmicro/mfgtools/wiki/migration-from-ucl2.xml-to-uuu-script because it don't appear at "<CMD state="BootStrap" type="boot" body="BootStrap" file ="firmware/flash.bin" ifdev="MX8MM">Loading U-boot</CMD>" 

seems that I managed to translate the boot script, however I would like to have a log of the output of FBK: ucmd because I do FBK: ucmd md5sum /dev/mmcblk0 at the end of script in order to compare with the other ucl2.xml script... is there a kind of log? (by the way, same for the old mfgtoolcli that use ucl2.xml script, is there a way to get the output of <CMD> ?)

thanks and regards

JLMorange, 2018-12-04:
by the way, it seems that "flash once mode" don't work, it seems to run always in deamon mode... waiting forever....

nxpfrankli, 2018-12-04:
> by the way, it seems that "flash once mode" don't work, it seems to run always in deamon mode... waiting forever....

Did you add "FBK: done" as last command?

nxpfrankli, 2018-12-04:
> ok but why "SDP: boot -f flash.bin SDP: boot -f flash.bin" for me it's the same command copied two times... no SPL in it....
>
>
> ok, It would be nice to explain it in [...]

uuu -v  xxx 
will show output of linux cmd output. 

> 
> thanks and regards

JLMorange, 2018-12-04:
> > by the way, it seems that "flash once mode" don't work, it seems to run always in deamon mode... waiting forever....
> 
> Did you add "FBK: done" as last command?

yes...
but for some reason my script now hang on first line : 
```SDP: boot -f mfgtool/Profiles/Linux/OS_Firmware/firmware/u-boot-imx7dsabresd_sd.imx``` the next line ```FB: ucmd setenv loadaddr 0x80800000``` hangs saying it wait for a compatible device....

nxpfrankli, 2018-12-04:
you need use 4.14 release's uboot.  Old uboot doesn't support fastboot protocol. 

May old uboot below commands work

sdp: dcd -f u-boot.imx7d.imx
sdp: write -f image -addr <you get from ucl.xml>
sdp: write -f dtb  -addr    <you get from ucl.xml>
sdp: write -f initramfs -addr <you get from ucl.xml>
sdp: jump -f u-boot.imx -ivt 0

4.14 release's uboot use FB protocol download image\dtb\initramfs. 

As my previous comments, most case needn't download kernel. 

use uboot/fastboot can do burn image. like

uuu -b emmc bootloader
will write bootloader into emmc partition.

nxpfrankli, 2018-12-05:
Try download,  https://www.nxp.com/pages/alpha-beta-bsps-for-microprocessors:IMXPRERELEASES

nxpfrankli, 2018-12-12:
No further action. If need, please reopen it

closed 2018-12-12 by nxpfrankli

JLMorange, 2018-12-17:
hi! coming back to this task....

I downloaded the beta bsp for sabresd 
 in my script I have 
´´´
SDP: boot -f beta/u-boot-imx7dsabresd_nand.imx
FB: ucmd setenv fastboot_buffer ${loadaddr}
FB: ucmd download -f mfgtool/Profiles/Linux/OS_Firmware/firmware/zImage
FB: ucmd setenv fastboot_buffer  ${initrd_addr}
FB: ucmd download -f mfgtool/Profiles/Linux/OS_Firmware/firmware/fsl-image-mfgtool-initramfs-imx_mfgtools.cpio.gz.u-boot
FB: ucmd setenv fastboot_buffer ${fdt_addr}
FB: ucmd download -f mfgtool/Profiles/Linux/OS_Firmware/firmware/zImage-imx7d-sdb.dtb
FB: acmd booti ${loadaddr} ${initrd_addr} ${fdt_addr}
´´´

the uuu tool output 
´´´
$ ./uuu uuu.auto-MX7D -V
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.0-36-g8b35656

Enjoy auto [tab] command complete by put below script into /etc/bash_completion.d/uuu
  _uuu_autocomplete()
  {
       COMPREPLY=($(/home/orange/src/build/deploy/svh-client-firmware-orange-develop-imx7dsabresd-0.97.0/uuu $1 $2 $3))
  }
  complete -o nospace -F _uuu_autocomplete  uuu

Success 0    Failure 1                                                                                                                                                                                                            
                                                                                                                                                                                                                                   
1:2    1/ 1   [============100%============] SDP: boot -f beta/u-boot-imx7dsabresd_nand.imx 
´´´
I see shortly a red line appearing (something like "failure to open...." not sure, it's overwritten by the prompt right after)

dmesg show 
´´´
[19742.441981] usb 1-2: new high-speed USB device number 48 using xhci_hcd
[19742.750130] usb 1-2: New USB device found, idVendor=15a2, idProduct=0076
[19742.750132] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[19742.750134] usb 1-2: Product: SE Blank ULT1
[19742.750135] usb 1-2: Manufacturer: Freescale SemiConductor Inc 
[19742.753020] hid-generic 0003:15A2:0076.001D: hiddev0,hidraw1: USB HID v1.10 Device [Freescale SemiConductor Inc  SE Blank ULT1] on usb-0000:00:0c.0-2/input0
[19743.273741] usb 1-2: usbfs: process 8360 (uuu) did not claim interface 0 before use
[19744.905499] usb 1-4: new high-speed USB device number 49 using xhci_hcd
[19745.208489] usb 1-4: New USB device found, idVendor=0525, idProduct=a4a5
[19745.208492] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[19745.208493] usb 1-4: Product: USB download gadget
[19745.208495] usb 1-4: Manufacturer: FSL
[19745.208496] usb 1-4: SerialNumber: 00000196b4439b02
[19745.211791] usb-storage 1-4:1.0: USB Mass Storage device detected
[19745.211846] usb-storage 1-4:1.0: Quirks match for vid 0525 pid a4a5: 10000
[19745.229410] usb 1-2: USB disconnect, device number 48
´´´

and on serial line I get 
´´´
-Boot 2018.03-imx_v2018.03_4.14.62_1.0.0_beta+g7fac6b9 (Oct 16 2018 - 19:16:50 +0000)

CPU:   Freescale i.MX7D rev1.2 1000 MHz (running at 792 MHz)
CPU:   Commercial temperature grade (0C to 95C) at 41C
Reset cause: POR
Model: Freescale i.MX7 SabreSD Board
Board: i.MX7D SABRESD RevC in secure mode
DRAM:  1 GiB
PMIC: PFUZE3000 DEV_ID=0x30 REV_ID=0x11
NAND:  0 MiB
MMC:   FSL_SDHC: 0
Loading Environment from NAND... *** Warning - readenv() failed, using default environment

Failed (-5)
Display: TFT43AB (480x272)
Video: 480x272x24
In:    serial
Out:   serial
Err:   serial
Detect USB boot. Will enter fastboot mode!
Net:   
Error: ethernet@30be0000 address not set.
No ethernet found.
Fastboot: Normal
Boot from USB for mfgtools
Use default environment for 				 mfgtools
Run bootcmd_mfg: run mfgtool_args;if iminfo ${initrd_addr}; then if test ${tee} = yes; then bootm ${tee_addr} ${initrd_addr} ${fdt_addr}; else bootz ${loadaddr} ${initrd_addr} ${fdt_addr}; fi; else echo "Run fastboot ..."; fastboot 0; fi;
Hit any key to stop autoboot:  0 

## Checking Image at 83800000 ...
Unknown image format!
Run fastboot ...

´´´

so for me I have now a fastboot compatible uboot, but for some reason FB:  lines aren't performed... 
any idea?
thanks and regards
JLM

nxpfrankli, 2018-12-17:
try sudo -v ./uuu uuu.auto-MX7D

nxpfrankli, 2018-12-17:
Not sure why mass storage appear in 

[19745.211791] usb-storage 1-4:1.0: USB Mass Storage device detected
[19745.211846] usb-storage 1-4:1.0: Quirks match for vid 0525 pid a4a5: 10000

JLMorange, 2018-12-18:
```
$ sudo ./uuu -v uuu.auto-MX7D
[sudo] password for orange: 
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.0-36-g8b35656

Enjoy auto [tab] command complete by put below script into /etc/bash_completion.d/uuu
  _uuu_autocomplete()
  {
       COMPREPLY=($(/home/orange/src/build/deploy/svh-client-firmware-orange-develop-imx7dsabresd-0.97.0/uuu $1 $2 $3))
  }
  complete -o nospace -F _uuu_autocomplete  uuu

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 0x0002
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 0x0002
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02
Wait for Known USB Device Appear...
New USB Device Attached at 1:3
1:3>Start Cmd:SDP: boot -f beta/u-boot-imx7dsabresd_nand.imx
6400%1:3>Okay
New USB Device Attached at 1:4
1:4>Start Cmd:FB: ucmd setenv fastboot_buffer ${loadaddr}
1:4>Okay
1:4>Start Cmd:FB: ucmd download -f mfgtool/Profiles/Linux/OS_Firmware/firmware/zImage
1:4>Fail 

```
so a setenv worked... but the download fail really fast.... 
on the serial line I get 
```
Unknown command 'download' - try 'help'
```
and if I add a FB: ucmd help to the script before the download I get 
```
?       - alias for 'help'
base    - print or set address offset
bdinfo  - print Board Info structure
bmode   - sd1|emmc|qspi|ecspi1:0|ecspi1:1|ecspi1:2|ecspi1:3|weim|qspi1|usdhc1|usdhc2|usdhc3|mmc1|mmc2|mmc3 [noreset]
bmp     - manipulate BMP image data
boot    - boot default, i.e., run 'bootcmd'
bootaux - Start auxiliary core
bootd   - boot default, i.e., run 'bootcmd'
bootefi - Boots an EFI payload from memory
bootelf - Boot from an ELF image in memory
bootm   - boot application image from memory
bootp   - boot image via network using BOOTP/TFTP protocol
bootvx  - Boot vxWorks from an ELF image
bootz   - boot Linux zImage image from memory
clocks  - display clocks
clrlogo - fill the boot logo area with black
cmp     - memory compare
coninfo - print console devices and information
cp      - memory copy
crc32   - checksum calculation
dcache  - enable or disable data cache
dfu     - Device Firmware Upgrade
dhcp    - boot image via network using DHCP/TFTP protocol
dm      - Driver model low level access
echo    - echo args to console
editenv - edit environment variable
env     - environment handling commands
erase   - erase FLASH memory
exit    - exit script
ext2load- load binary file from a Ext2 filesystem
ext2ls  - list files in a directory (default /)
ext4load- load binary file from a Ext4 filesystem
ext4ls  - list files in a directory (default /)
ext4size- determine a file's size
ext4write- create a file in the root directory
false   - do nothing, unsuccessfully
fastboot- use USB Fastboot protocol
fatinfo - print information about filesystem
fatload - load binary file from a dos filesystem
fatls   - list files in a directory (default /)
fatsize - determine a file's size
fdt     - flattened device tree utility commands
flinfo  - print FLASH memory information
fuse    - Fuse sub-system
go      - start application at address 'addr'
gpio    - query and control gpio pins
help    - print command description/usage
i2c     - I2C sub-system
icache  - enable or disable instruction cache
iminfo  - print header information for application image
itest   - return true/false on integer compare
loadb   - load binary file over serial line (kermit mode)
loads   - load S-Record file over serial line
loadx   - load binary file over serial line (xmodem mode)
loady   - load binary file over serial line (ymodem mode)
loop    - infinite loop on address range
md      - memory display
mdio    - MDIO utility commands
mii     - MII utility commands
mm      - memory modify (auto-incrementing address)
mmc     - MMC sub system
mmcinfo - display MMC info
mtest   - simple RAM read/write test
mw      - memory write (fill)
nand    - NAND sub-system
nboot   - boot from NAND device
nfs     - boot image via network using NFS protocol
nm      - memory modify (constant address)
ping    - send ICMP ECHO_REQUEST to network host
pmic    - PMIC sub-system
printenv- print environment variables
protect - enable or disable FLASH write protection
read    - Load binary data from a partition
regulator- uclass operations
reset   - Perform RESET of the CPU
run     - run commands in an environment variable
saveenv - save environment variables to persistent storage
setenv  - set environment variables
setexpr - set environment variable as the result of eval expression
sf      - SPI flash sub-system
showvar - print local hushshell variables
sleep   - delay execution for some time
source  - run script from memory
test    - minimal test like /bin/sh
tftpboot- boot image via network using TFTP protocol
true    - do nothing, successfully
ums     - Use the UMS [USB Mass Storage]
usb     - USB sub-system
usbboot - boot from USB device
version - print monitor, compiler and linker version
```

> Not sure why mass storage appear in
> 
> [19745.211791] usb-storage 1-4:1.0: USB Mass Storage device detected
> [19745.211846] usb-storage 1-4:1.0: Quirks match for vid 0525 pid a4a5: 10000

maybe because of ums command?

JLMorange, 2018-12-18:
I tried another way to boot, so I used 
```
sdp: dcd -f beta/u-boot-imx7dsabresd_nand.imx
sdp: write -f mfgtool/Profiles/Linux/OS_Firmware/firmware/zImage -addr 0x80800000
sdp: write -f mfgtool/Profiles/Linux/OS_Firmware/firmware/zImage-imx7d-sdb.dtb -addr  0x83000000
sdp: write -f mfgtool/Profiles/Linux/OS_Firmware/firmware/fsl-image-mfgtool-initramfs-imx_mfgtools.cpio.gz.u-boot -addr 0x83800000
SDP: boot -f beta/u-boot-imx7dsabresd_nand.imx 
```

this worked launching linux however the following 
```
FBK: ucp mfgtool/Profiles/Linux/OS_Firmware/mksdcard.sh t:/tmp
FBK: ucmd sh /tmp/mksdcard.sh /dev/mmcblk2
```
aren't executed....

on the serial, I see the kernel boot, and stop at 
```
Freeing unused kernel memory: 388K (8091f000 - 80980000)
Starting UTP
uuc 0.5 [built Mar 12 2016 19:04:20]
UTP: Waiting for device to appear
mmc2: new HS400 MMC card at address 0001
mmcblk2: mmc2:0001 008G30 7.28 GiB 
mmcblk2boot0: mmc2:0001 008G30 partition 1 4.00 MiB
mmcblk2boot1: mmc2:0001 008G30 partition 2 4.00 MiB
mmcblk2rpmb: mmc2:0001 008G30 partition 3 4.00 MiB
 mmcblk2: p1 p2 p3 p4 < p5 p6 p7 p8 p9 p10 p11 p12 p13 p14 >

```

nxpfrankli, 2018-12-18:
> ```
> $ sudo ./uuu -v uuu.auto-MX7D
> [sudo] password for xxxxx: 
> uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.0-36-g8b35656
> 
> Enjoy auto [tab] command complete by put below script into /etc/bash_completion.d/uuu
>   _uuu_autocomplete()
>   {
>        COMPREPLY=($(/home/orange/src/build/deploy/svh-client-firmware-orange-develop-imx7dsabresd-0.97.0/uuu $1 $2 $3))
>   }
>   complete -o nospace -F _uuu_autocomplete  uuu
> 
> Build in config:
> 	Pctl	 Chip		 Vid	 Pid	 BcdVersion
> 	==================================================
> 	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 0x0002
> 	SDPS:	 MX8QM		 0x1fc9	 0x0129	 0x0002
> 	SDP:	 MX7D		 0x15a2	 0x0076
> 	SDP:	 MX6Q		 0x15a2	 0x0054
> 	SDP:	 MX6D		 0x15a2	 0x0061
> 	SDP:	 MX6SL		 0x15a2	 0x0063
> 	SDP:	 MX6SX		 0x15a2	 0x0071
> 	SDP:	 MX6UL		 0x15a2	 0x007d
> 	SDP:	 MX6ULL		 0x15a2	 0x0080
> 	SDP:	 MX6SLL		 0x1fc9	 0x0128
> 	SDP:	 MX7ULP		 0x1fc9	 0x0126
> 	SDP:	 MXRT106X	 0x1fc9	 0x0135
> 	SDP:	 MX8MM		 0x1fc9	 0x0134
> 	SDP:	 MX8MQ		 0x1fc9	 0x012b
> 	SDPU:	 SPL		 0x0525	 0xb4a4
> 	FBK:	 		 0x066f	 0x9afe
> 	FBK:	 		 0x066f	 0x9bff
> 	FB:	 		 0x0525	 0xa4a5
> 	FB:	 		 0x18d1	 0x0d02
> Wait for Known USB Device Appear...
> New USB Device Attached at 1:3
> 1:3>Start Cmd:SDP: boot -f beta/u-boot-imx7dsabresd_nand.imx
> 6400%1:3>Okay
> New USB Device Attached at 1:4
> 1:4>Start Cmd:FB: ucmd setenv fastboot_buffer ${loadaddr}
> 1:4>Okay
> 1:4>Start Cmd:FB: ucmd download -f mfgtool/Profiles/Linux/OS_Firmware/firmware/zImage
> 1:4>Fail 

It should be 

FB: download -f mfgtool/Profiles/Linux/OS_Firmware/firmware/zImage

download is one of FB command. 

> ```
> so a setenv worked... but the download fail really fast....
> on the serial line I get
>
> ```
> Unknown command 'download' - try 'help'
> ```
> and if I add a FB: ucmd help to the script before the [...]

JLMorange, 2018-12-18:
are UTP and FBK the same? I tried to split the script in two : one is starting the kernel and I see the ```Linux 4.1.15-1.0.0-mfgtool+g3924425 with 30b10000.usb``` appears (listed as ```066f:37ff SigmaTel, Inc```)
so now I launch the 2 script, hoping for uuu to wait for the 066f:37ff and start fbk commands... but nothing happens.... the device connect, with old mfgtool I see lot of UTP commands sent.... but with uuu just nothing....

nxpfrankli, 2018-12-18:
> I tried another way to boot, so I used
> 
> ```
> sdp: dcd -f beta/u-boot-imx7dsabresd_nand.imx
> sdp: write -f mfgtool/Profiles/Linux/OS_Firmware/firmware/zImage -addr 0x80800000
> sdp: write -f mfgtool/Profiles/Linux/OS_Firmware/firmware/zImage-imx7d-sdb.dtb -addr  0x83000000
> sdp: write -f mfgtool/Profiles/Linux/OS_Firmware/firmware/fsl-image-mfgtool-initramfs-imx_mfgtools.cpio.gz.u-boot -addr 0x83800000
> SDP: boot -f beta/u-boot-imx7dsabresd_nand.imx 
> ```
> this worked launching linux however the following
> 
> ```
> FBK: ucp mfgtool/Profiles/Linux/OS_Firmware/mksdcard.sh t:/tmp
> FBK: ucmd sh /tmp/mksdcard.sh /dev/mmcblk2
> ```
> aren't executed....
> 
> on the serial, I see the kernel boot, and stop at
> 
> ```
> Freeing unused kernel memory: 388K (8091f000 - 80980000)
> Starting UTP
> uuc 0.5 [built Mar 12 2016 19:04:20]
> UTP: Waiting for device to appear
> mmc2: new HS400 MMC card at address 0001
> mmcblk2: mmc2:0001 008G30 7.28 GiB 
> mmcblk2boot0: mmc2:0001 008G30 partition 1 4.00 MiB
> mmcblk2boot1: mmc2:0001 008G30 partition 2 4.00 MiB
> mmcblk2rpmb: mmc2:0001 008G30 partition 3 4.00 MiB
>  mmcblk2: p1 p2 p3 p4 < p5 p6 p7 p8 p9 p10 p11 p12 p13 p14 >
> ```
> on usb side I see again a `Bus 001 Device 023: ID 15a2:0076 Freescale Semiconductor, Inc. `

Seem your board is rebooted. 
Did you use updated initramfs?  check if ufb running?

JLMorange, 2018-12-19:
ok, I fixed the thinkgs.... the trick was to use the kernel and dtd and initramfs of the beta package..... now the update of the target works :)