nxp-imx/mfgtools #72: issue to flashing  the  i.MX 8M Quad EVK Board

dreampet, 2018-12-25:
here is the log:
E:\IMX8\android_p9.0.0_1.0.0-beta_image_8qmek>uuu.exe uuu-android-mx8qm-mek-emmc.lst
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.39-4-g219e333

Success 0    Failure 1

       1/ 0   [                            ]
1:6    1/21   [Bulk(W):LIBUSB_ERROR_NOT_FOU] FB: ucmd setenv fastboot_dev mmc

dreampet, 2019-01-01:
someone report the same problem on NXP community
[fail to update android9.0 by uuu tool](https://community.nxp.com/message/1096219)

nxpfrankli, 2019-01-07:
Do you use win7?  look like winusb driver was not installed.  Please refer https://github.com/NXPmicro/mfgtools/wiki/WIN7-User-Guide

gowricpms, 2019-01-08:
I had an issue with the non-Android version as well and was asked to post my question here. Here's the original post:
https://community.nxp.com/message/1097467
According to this document that Igor pointed me to:
https://community.nxp.com/servlet/JiveServlet/downloadBody/341877-102-2-287322/AMF-AUT-T3367.pdf
uuu does not support imx8mqevk yet. Is this true?
If it is, when will support be added? We are also trying to get this working on Ubuntu and not Windows. Any help will be greatly appreciated!

-Gowri

nxpfrankli, 2019-01-08:
> I had an issue with the non-Android version as well and was asked to post my question here. Here's the original post:
> https://community.nxp.com/message/1097467
> According to this document that Igor [...]

i.MX8 MQ uboot need be updated. After L4.14-beta release. 

Please upload your uart output. 

> 
> -Gowri

dreampet, 2019-01-08:
> Do you use win7? look like winusb driver was not installed. Please refer https://github.com/NXPmicro/mfgtools/wiki/WIN7-User-Guide

@nxpfrankli  Hi, I use the Win10 64Bit system.

nxpfrankli, 2019-01-08:
Previous uboot WCID implement have bugs,  can you double check if winusb driver install by device manage or https://zadig.akeo.ie/?

raveslave, 2019-01-08:
Hi, I have the same issue, tested on ubuntu 18.04 (using the new L4.14.78_1.0.0_MX8MQ zip)

gowricpms, 2019-01-08:
> i.MX8 MQ uboot need be updated. After L4.14-beta release.
>
> Please upload your uart output.

I will send you the uart output shortly because it looks like we lost the original setup. I am now using an imx8mm EVK. Also, when can we expect the L4.14-beta release to become available?

nxpfrankli, 2019-01-08:
imx8mm EVK 4.9 GA already support uuu

gowricpms, 2019-01-08:
> imx8mm EVK 4.9 GA already support uuu

I'm going to try to get a 4.14* (sumo) image onto an imx8mm EVK. Is this also supported? Has anyone tried it yet to see if something is broken? I understand that you said 4.9.* (rocko) supports it.

We still intend to use i.MX8 MQ (waiting to get more of those EVKs) so we'd be interested to know when the L4.14-beta release will become available.

nxpfrankli, 2019-01-08:
4.14-beta already pushed. 

https://www.nxp.com/support/developer-resources/run-time-software/linux-software-and-development-tools/embedded-linux-for-i.mx-applications-processors:IMXLINUX?&code=IMXLINUX&nodeId=012FC8ZrDRB24A4117&fpsp=1&tab=Design_Tools_Tab

All in 4.14-beta support uuu. 
 
some early i.MX8MM chip missed a ROM patch, which impact linux support.

dreampet, 2019-01-09:
[image]
@nxpfrankli the Zadig info, I try to click the "Replace Driver", but it take a long time and failure finally.

dreampet, 2019-01-09:
[image]
@nxpfrankli plug the another and seems the driver changed to WinUSB , but uuu still failure.

nxpfrankli, 2019-01-09:
"SE Blank 8QM" should use HID.  
"User serial download" (PID VID is 0x0525  0xa4a5 ) should use winusb.

nxpfrankli, 2019-01-09:
Did you found device PID VID 0x525, 0xa4a5 device?  if you use 8QM,  are you use type C port or baseboard USB Port?

dreampet, 2019-01-09:
@nxpfrankli 
1. can't found PID VID 0x525, 0xa4a5 in device manager.
2. I use IMX8QM EVK Board and use the USB OTG port on the baseboard. 
3. I also try to  use the TYPE C port but found nothing on the device manager

dreampet, 2019-01-09:
@nxpfrankli
By the way, I found a confuse setting on the uuu-android-mx8qm-mek-emmc.lst ：
uuu_version 1.1.29
#uuu scripts for imx8qm revB board Android imx_pi9.0 eMMC
CFG: FB: -vid 0x1fc9 -pid 0x0129
SDPS: boot -f uuu-u-boot-imx8qm.imx

It seems that the 8QM use PID VID 1FC9,0129 but not PID VID 0x525, 0xa4a5 ?

nxpfrankli, 2019-01-09:
you can check what usb device appear now?

dreampet, 2019-01-09:
[image]
 if I comment this line, uuu wait forever

nxpfrankli, 2019-01-09:
This should be problem.  Report to our android team.  Windows look like always treat 1fc9\0129 as HID device.

raveslave, 2019-01-09:
> imx8mm EVK 4.9 GA already support uuu

yes, but the qualcomm wifi doesn't seem supported on that release.

raveslave, 2019-01-09:
Please see dmesg log on what happens when trying to flash the imx8m evk
_via: Ubuntu 18.04 (Linux ubuntu 4.15.0-20-generic)_

```
[ 3533.065603] usb 1-1: new high-speed USB device number 29 using ehci-pci
[ 3533.222274] usb 1-1: New USB device found, idVendor=1fc9, idProduct=012b
[ 3533.222276] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 3533.222279] usb 1-1: Product: SE Blank M850 
[ 3533.222279] usb 1-1: Manufacturer: NXP       SemiConductor Inc 
[ 3533.233839] hid-generic 0003:1FC9:012B.001B: hiddev0,hidraw0: USB HID v1.00 Device [NXP       SemiConductor Inc  SE Blank M850 ] on usb-0000:00:1d.7-1/input0
[ 3533.562099] usb 1-1: usbfs: process 12998 (uuu) did not claim interface 0 before use
[ 3534.186370] usb 1-1: USB disconnect, device number 29
[ 3535.105510] usb 1-1: new high-speed USB device number 30 using ehci-pci
[ 3535.262145] usb 1-1: New USB device found, idVendor=0525, idProduct=b4a4
[ 3535.262147] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 3535.262148] usb 1-1: Product: USB download gadget
[ 3535.262148] usb 1-1: Manufacturer: FSL
[ 3535.276558] hid-generic 0003:0525:B4A4.001C: hiddev0,hidraw0: USB HID v1.10 Device [FSL USB download gadget] on usb-0000:00:1d.7-1/input0

```

and the output from uuu

```
user@ubuntu:/home/user/L4.14.78_1.0.0_ga_images_MX8MQEVK$ uuu uuu.auto
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.39-11-ga4bbea2

Success 0    Failure 0                                                                               
                                                                                                      
1:1    1/ 0   [                            ]                                                          

```

raveslave, 2019-01-09:
FYI, running with the .zip directly seems broken.

```
uuu L4.14.78_1.0.0_ga_images_MX8MQEVK.zip
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.39-11-ga4bbea2

terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
```

nxpfrankli, 2019-01-09:
> Please see dmesg log on what happens when trying to flash the imx8m evk
> _via: Ubuntu 18.04 (Linux ubuntu 4.15.0-20-generic)_
>
> ```
> [ 3533.065603] usb 1-1: new high-speed USB device number 29 using [...]

Can you run "uuu -v uuu.auto"?  I need more information.

raveslave, 2019-01-09:
```
$ uuu -v uuu.auto
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.39-11-ga4bbea2

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0xffff]
	SDPV:	 SPL1		 0x0525	 0xb4a5
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02
Wait for Known USB Device Appear...
New USB Device Attached at 1:1
1:1>Start Cmd:SDP: boot -f imx-boot-imx8mqevk-sd.bin-flash_evk
6400%1:1>Okay
New USB Device Attached at 1:1

```
after this nothing happens

raveslave, 2019-01-09:
ps. the same release works in windows :-(    
would be great to see working linux/mac support though! :-)

nxpfrankli, 2019-01-09:
can you try to revert version uuu_1.2.0? which formal stable version.

nxpfrankli, 2019-01-09:
In our linux environment, everything work well.  I will download a clean 18.04 and try again. 
Mac have another problem when claim usb interface.

nxpfrankli, 2019-01-09:
I use live ubuntu 18.04 image.  Everything seem work well. 
not sure why uuu can't detect hid-generic 0003:0525:B4A4.001C in your machine.

Can you try "ctrl-C" uuu and run uuu again? What's debug UART port log? 

```
sudo ./uuu -v -b spl Downloads/imx-boot-imx8mmevk-sd.bin-flash_evk 
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.53-0-gea5f009

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0xffff]
	SDPV:	 SPL1		 0x0525	 0xb4a5
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02

Run built-in script:
 
uuu_version 1.2.39

# This command will be run when i.MX6/7 i.MX8MM, i.MX8MQ
SDP: boot -f Downloads/imx-boot-imx8mmevk-sd.bin-flash_evk

# This command will be run when ROM support stream mode
# i.MX8QXP, i.MX8QM
SDPS: boot -f Downloads/imx-boot-imx8mmevk-sd.bin-flash_evk

# These commands will be run when use SPL and will be skipped if no spl
# SDPU will be deprecated. please use SDPV instead of SDPU
# {
SDPU: delay 1000
SDPU: write -f Downloads/imx-boot-imx8mmevk-sd.bin-flash_evk -offset 0x57c00
SDPU: jump
SDPU: done
# }

# These commands will be run when use SPL and will be skipped if no spl
# if (SPL support SDPV)
# {
SDPV: delay 1000
SDPV: write -f Downloads/imx-boot-imx8mmevk-sd.bin-flash_evk -skipspl
SDPV: jump
SDPV: done
# }

Wait for Known USB Device Appear...
New USB Device Attached at 3:11
3:11>Start Cmd:SDP: boot -f Downloads/imx-boot-imx8mmevk-sd.bin-flash_evk
6400%3:11>Okay
New USB Device Attached at 3:11
3:11>Start Cmd:SDPU: delay 1000
3:11>Okay
3:11>Start Cmd:SDPU: write -f Downloads/imx-boot-imx8mmevk-sd.bin-flash_evk -offset 0x57c00
100%3:11>Okay
3:11>Start Cmd:SDPU: jump
6400%3:11>Okay
3:11>Start
```

gowricpms, 2019-01-10:
Here's my output from using L4.14.78_1.0.0_ga_images_MX8MMEVK.zip

gowri@gowri-ubuntu:~/yocto_drive/mfgtools/uuu$ sudo ./uuu -v ~/Desktop/L4.14.78_1.0.0_ga_images_MX8MMEVK/uuu.auto 
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.39-14-gea5f009

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0xffff]
	SDPV:	 SPL1		 0x0525	 0xb4a5
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02
Wait for Known USB Device Appear...

I probably did not word my question correctly the last time but we are building our own image so with those binaries and my uuu.auto script which is based on your script above, this is what I see:
gowri@gowri-ubuntu:~/yocto_drive/mfgtools/uuu$ sudo ./uuu -v ../uuu.auto 
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.39-14-gea5f009

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0xffff]
	SDPV:	 SPL1		 0x0525	 0xb4a5
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02
Wait for Known USB Device Appear...
>Start Cmd:CFG: FB: -vid 0x1fc9 -pid 0x0134
>Okay

My uuu.auto script is as follows:
uuu_version 1.2.39
CFG: FB: -vid 0x1fc9 -pid 0x0134
# This command will be run when i.MX6/7 i.MX8MM, i.MX8MQ
#SDP: boot -f files/imx-boot-imx8mmevk-sd.bin-flash_evk
# This command will be run when ROM support stream mode
# i.MX8QXP, i.MX8QM
SDPS: boot -f files/imx-boot-imx8mmevk-sd.bin-flash_evk
# These commands will be run when use SPL and will be skipped if no spl
# SDPU will be deprecated. please use SDPV instead of SDPU
SDPU: delay 1000
SDPU: write -f files/imx-boot-imx8mmevk-sd.bin-flash_evk -offset 0x57c00
SDPU: jump
SDPU:done

dmesg contents are attached.
[dmesg.txt](https://github.com/NXPmicro/mfgtools/files/2743385/dmesg.txt)

gowricpms, 2019-01-10:
Here's my output from using L4.14.78_1.0.0_ga_images_MX8MMEVK.zip

gowri@gowri-ubuntu:~/yocto_drive/mfgtools/uuu$ sudo ./uuu -v ~/Desktop/L4.14.78_1.0.0_ga_images_MX8MMEVK/uuu.auto
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.39-14-gea5f009

Build in config:
Pctl Chip Vid Pid BcdVersion
==================================================
SDPS: MX8QXP 0x1fc9 0x012f [0x0002..0xffff]
SDPS: MX8QM 0x1fc9 0x0129 [0x0002..0xffff]
SDP: MX7D 0x15a2 0x0076
SDP: MX6Q 0x15a2 0x0054
SDP: MX6D 0x15a2 0x0061
SDP: MX6SL 0x15a2 0x0063
SDP: MX6SX 0x15a2 0x0071
SDP: MX6UL 0x15a2 0x007d
SDP: MX6ULL 0x15a2 0x0080
SDP: MX6SLL 0x1fc9 0x0128
SDP: MX7ULP 0x1fc9 0x0126
SDP: MXRT106X 0x1fc9 0x0135
SDP: MX8MM 0x1fc9 0x0134
SDP: MX8MQ 0x1fc9 0x012b
SDPU: SPL 0x0525 0xb4a4 [0x0000..0x04ff]
SDPV: SPL1 0x0525 0xb4a4 [0x0500..0xffff]
SDPV: SPL1 0x0525 0xb4a5
FBK: 0x066f 0x9afe
FBK: 0x066f 0x9bff
FB: 0x0525 0xa4a5
FB: 0x18d1 0x0d02
Wait for Known USB Device Appear...

I probably did not word my question correctly the last time but we are building our own image so with those binaries and my uuu.auto script which is based on your script above, this is what I see:
gowri@gowri-ubuntu:~/yocto_drive/mfgtools/uuu$ sudo ./uuu -v ../uuu.auto
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.39-14-gea5f009

Build in config:
Pctl Chip Vid Pid BcdVersion
==================================================
SDPS: MX8QXP 0x1fc9 0x012f [0x0002..0xffff]
SDPS: MX8QM 0x1fc9 0x0129 [0x0002..0xffff]
SDP: MX7D 0x15a2 0x0076
SDP: MX6Q 0x15a2 0x0054
SDP: MX6D 0x15a2 0x0061
SDP: MX6SL 0x15a2 0x0063
SDP: MX6SX 0x15a2 0x0071
SDP: MX6UL 0x15a2 0x007d
SDP: MX6ULL 0x15a2 0x0080
SDP: MX6SLL 0x1fc9 0x0128
SDP: MX7ULP 0x1fc9 0x0126
SDP: MXRT106X 0x1fc9 0x0135
SDP: MX8MM 0x1fc9 0x0134
SDP: MX8MQ 0x1fc9 0x012b
SDPU: SPL 0x0525 0xb4a4 [0x0000..0x04ff]
SDPV: SPL1 0x0525 0xb4a4 [0x0500..0xffff]
SDPV: SPL1 0x0525 0xb4a5
FBK: 0x066f 0x9afe
FBK: 0x066f 0x9bff
FB: 0x0525 0xa4a5
FB: 0x18d1 0x0d02
Wait for Known USB Device Appear...

    Start Cmd:CFG: FB: -vid 0x1fc9 -pid 0x0134
    Okay

My uuu.auto script is as follows:
uuu_version 1.2.39
CFG: FB: -vid 0x1fc9 -pid 0x0134
SDPS: boot -f files/imx-boot-imx8mmevk-sd.bin-flash_evk
SDPU: delay 1000
SDPU: write -f files/imx-boot-imx8mmevk-sd.bin-flash_evk -offset 0x57c00
SDPU: jump
SDPU:done

dmesg contents are attached.
[dmesg.txt](https://github.com/NXPmicro/mfgtools/files/2743388/dmesg.txt)

gowricpms, 2019-01-10:
Here's my output from using L4.14.78_1.0.0_ga_images_MX8MMEVK.zip

gowri@gowri-ubuntu:~/yocto_drive/mfgtools/uuu$ sudo ./uuu -v ~/Desktop/L4.14.78_1.0.0_ga_images_MX8MMEVK/uuu.auto
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.39-14-gea5f009
Build in config:
Pctl Chip Vid Pid BcdVersion

SDPS: MX8QXP 0x1fc9 0x012f [0x0002..0xffff]
SDPS: MX8QM 0x1fc9 0x0129 [0x0002..0xffff]
SDP: MX7D 0x15a2 0x0076
SDP: MX6Q 0x15a2 0x0054
SDP: MX6D 0x15a2 0x0061
SDP: MX6SL 0x15a2 0x0063
SDP: MX6SX 0x15a2 0x0071
SDP: MX6UL 0x15a2 0x007d
SDP: MX6ULL 0x15a2 0x0080
SDP: MX6SLL 0x1fc9 0x0128
SDP: MX7ULP 0x1fc9 0x0126
SDP: MXRT106X 0x1fc9 0x0135
SDP: MX8MM 0x1fc9 0x0134
SDP: MX8MQ 0x1fc9 0x012b
SDPU: SPL 0x0525 0xb4a4 [0x0000..0x04ff]
SDPV: SPL1 0x0525 0xb4a4 [0x0500..0xffff]
SDPV: SPL1 0x0525 0xb4a5
FBK: 0x066f 0x9afe
FBK: 0x066f 0x9bff
FB: 0x0525 0xa4a5
FB: 0x18d1 0x0d02
Wait for Known USB Device Appear...

I probably did not word my question correctly the last time but we are building our own image so with those binaries and my uuu.auto script which is based on your script above, this is what I see:
gowri@gowri-ubuntu:~/yocto_drive/mfgtools/uuu$ sudo ./uuu -v ../uuu.auto
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.39-14-gea5f009
Build in config:
Pctl Chip Vid Pid BcdVersion

SDPS: MX8QXP 0x1fc9 0x012f [0x0002..0xffff]
SDPS: MX8QM 0x1fc9 0x0129 [0x0002..0xffff]
SDP: MX7D 0x15a2 0x0076
SDP: MX6Q 0x15a2 0x0054
SDP: MX6D 0x15a2 0x0061
SDP: MX6SL 0x15a2 0x0063
SDP: MX6SX 0x15a2 0x0071
SDP: MX6UL 0x15a2 0x007d
SDP: MX6ULL 0x15a2 0x0080
SDP: MX6SLL 0x1fc9 0x0128
SDP: MX7ULP 0x1fc9 0x0126
SDP: MXRT106X 0x1fc9 0x0135
SDP: MX8MM 0x1fc9 0x0134
SDP: MX8MQ 0x1fc9 0x012b
SDPU: SPL 0x0525 0xb4a4 [0x0000..0x04ff]
SDPV: SPL1 0x0525 0xb4a4 [0x0500..0xffff]
SDPV: SPL1 0x0525 0xb4a5
FBK: 0x066f 0x9afe
FBK: 0x066f 0x9bff
FB: 0x0525 0xa4a5
FB: 0x18d1 0x0d02
Wait for Known USB Device Appear...

Start Cmd:CFG: FB: -vid 0x1fc9 -pid 0x0134
Okay

My uuu.auto script is as follows:
uuu_version 1.2.39
CFG: FB: -vid 0x1fc9 -pid 0x0134
SDPS: boot -f files/imx-boot-imx8mmevk-sd.bin-flash_evk
SDPU: delay 1000
SDPU: write -f files/imx-boot-imx8mmevk-sd.bin-flash_evk -offset 0x57c00
SDPU: jump
SDPU:done

dmesg contents are attached.
[dmesg.txt](https://github.com/NXPmicro/mfgtools/files/2743396/dmesg.txt)

nxpfrankli, 2019-01-10:
what is output debug UART port?

nxpfrankli, 2019-01-10:
CFG: FB: -vid 0x1fc9 -pid 0x0134 is wrong.

nxpfrankli, 2019-01-10:
No HID usb device attached? Are you set board to serial download mode?

gowricpms, 2019-01-10:
Please note that I'm trying this on imx8M Mini so I believe:
CFG: FB: -vid 0x1fc9 -pid 0x0134
is correct?

My bad. I didn't have serial download mode set. With my script above (with the VID PID for 8M Mini), this is what I now see:

gowri@gowri-ubuntu:~/yocto_drive/mfgtools/uuu$ sudo ./uuu -v ../uuu.auto 
[sudo] password for gowri: 
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.39-14-gea5f009

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0xffff]
	SDPV:	 SPL1		 0x0525	 0xb4a5
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02
Wait for Known USB Device Appear...
>Start Cmd:CFG: FB: -vid 0x1fc9 -pid 0x0134
>Okay
New USB Device Attached at 2:14
2:14>Start Cmd:FB: ucmd setenv fastboot_buffer ${loadaddr}
2:14>Fail Bulk(W):LIBUSB_ERROR_IO

Nothing on the UART.
Dmesg is attached
[dmesg.txt](https://github.com/NXPmicro/mfgtools/files/2747278/dmesg.txt)

nxpfrankli, 2019-01-10:
CFG: FB: -vid 0x1fc9 -pid 0x0134
it wrong.

FB always is the difference PID\VID with ROM.

Linux Default uboot is
FB: 0x0525 0xa4a5

Uboot console should show at uart.
Lsusb check what device connect now.

Best regards
Frank Li

dreampet, 2019-01-12:
@nxpfrankli
Hi, I have found out what's wrong of this issue, this is the MX8QM info of my EVK board:
HID\VID_1FC9&PID_0129&**REV_0001**
HID\VID_1FC9&PID_0129
HID\VID_1FC9&UP:FF00_U:0001
HID_DEVICE_UP:FF00_U:0001
HID_DEVICE_UPR:FF00-FFFF
HID_DEVICE

but the uuu only support the at least REV_0002

https://github.com/NXPmicro/mfgtools/blob/5e0c52d36a94c72df099d8531b57b86a32d35875/libuuu/config.cpp#L45

Should uuu support this version of MX8QM?

dreampet, 2019-01-12:
I try to modify the version to 0x0001, uuu can found out the board but libusb write failure.

E:\IMX8\android_p9.0.0_1.0.0-beta_image_8qmek>uuu -v uuu-android-mx8qm-mek-emmc.lst
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.56-2-g5e0c52d

Build in config:
        Pctl     Chip            Vid     Pid     BcdVersion
        ==================================================
        SDPS:    MX8QXP          0x1fc9  0x012f  [0x0001..0xffff]
        SDPS:    MX8QM           0x1fc9  0x0129  [**0x0001..0xffff**]
        SDP:     MX7D            0x15a2  0x0076
        SDP:     MX6Q            0x15a2  0x0054
        SDP:     MX6D            0x15a2  0x0061
        SDP:     MX6SL           0x15a2  0x0063
        SDP:     MX6SX           0x15a2  0x0071
        SDP:     MX6UL           0x15a2  0x007d
        SDP:     MX6ULL          0x15a2  0x0080
        SDP:     MX6SLL          0x1fc9  0x0128
        SDP:     MX7ULP          0x1fc9  0x0126
        SDP:     MXRT106X        0x1fc9  0x0135
        SDP:     MX8MM           0x1fc9  0x0134
        SDP:     MX8MQ           0x1fc9  0x012b
        SDPU:    SPL             0x0525  0xb4a4  [0x0000..0x04ff]
        SDPV:    SPL1            0x0525  0xb4a4  [0x0500..0xffff]
        FBK:                     0x066f  0x9afe
        FBK:                     0x066f  0x9bff
        FB:                      0x0525  0xa4a5
        FB:                      0x18d1  0x0d02
Wait for Known USB Device Appear...
New USB Device Attached at 1:64
1:64>Start Cmd:SDPS: boot -f uuu-u-boot-imx8qm.imx
1:64>Fail HID(R):**LIBUSB_ERROR_TIMEOUT**

nxpfrankli, 2019-01-12:
QM A0 and QXP A0 is not supported by UUU. That's reason why filter it out. 
QM B0 and QXP B0 ROM change greatly.  We don't plan support no production chips.

dreampet, 2019-01-13:
OK, I build the latest version of mfgtool, and it works with MX8QM A0 now.

closed 2019-01-13 by dreampet