nxp-imx/mfgtools #75: Aborts with `munmap_chunk(): invalid pointer`

agx, 2018-12-29:
```
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.31-5-g33fa776                                                                                                                                       

Enjoy auto [tab] command complete by put below script into /etc/bash_completion.d/uuu
  _uuu_autocomplete()
  {
       COMPREPLY=($(/usr/bin/uuu $1 $2 $3))
  }
  complete -o nospace -F _uuu_autocomplete  uuu

Success 1    Failure 0                                                                                                                                                                                            
                                                                                                                                                                                                                   
1:1    4/ 4   [Done                        ] FB: Done                                                                                                                                                              
munmap_chunk(): invalid pointer
```

The backtrace is
```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
        set = {__val = {0, 139716289058560, 140732962586768, 139716289143978, 94210239630512, 94210239630720, 94210239630720, 0, 139712668940032, 4095, 4096, 139716289143978, 7, 0, 1, 281470681751424}}
        pid = <optimized out>
        tid = <optimized out>
        ret = <optimized out>
#1  0x00007f123b27f535 in __GI_abort () at abort.c:79
        save_stage = 1
        act = {__sigaction_handler = {sa_handler = 0x1, sa_sigaction = 0x1}, sa_mask = {__val = {139716279555529, 4222451713, 140732962586944, 140732962586944, 140732962586944, 140732962586944, 
              140732962586945, 140732962586945, 140732962586944, 140732962586945, 0, 0, 0, 0, 140732962586416, 140732962586672}}, sa_flags = -230801616, sa_restorer = 0x1000}
        sigs = {__val = {32, 0 <repeats 15 times>}}
#2  0x00007f123b2d6718 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7f123b3e129a "%s\n") at ../sysdeps/posix/libc_fatal.c:181
        ap = {{gp_offset = 24, fp_offset = 0, overflow_arg_area = 0x7ffef23e4040, reg_save_area = 0x7ffef23e3fd0}}
        fd = 2
        list = <optimized out>
        nlist = <optimized out>
        cp = <optimized out>
        written = <optimized out>
#3  0x00007f123b2dce3a in malloc_printerr (str=str@entry=0x7f123b3e2c00 "munmap_chunk(): invalid pointer") at malloc.c:5382
No locals.
#4  0x00007f123b2dd3a4 in munmap_chunk (p=<optimized out>) at malloc.c:2830
        size = <optimized out>
        __PRETTY_FUNCTION__ = "munmap_chunk"
        block = <optimized out>
        total_size = <optimized out>
#5  0x000055af05fedf4b in __gnu_cxx::new_allocator<char>::deallocate (this=0x7ffef23e4090, __p=<optimized out>) at /usr/include/c++/8/ext/new_allocator.h:116
No locals.
#6  std::allocator_traits<std::allocator<char> >::deallocate (__a=..., __n=<optimized out>, __p=<optimized out>) at /usr/include/c++/8/bits/alloc_traits.h:462
No locals.
#7  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_destroy (__size=<optimized out>, this=0x7ffef23e4090) at /usr/include/c++/8/bits/basic_string.h:226
No locals.
#8  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_dispose (this=0x7ffef23e4090) at /usr/include/c++/8/bits/basic_string.h:221
No locals.
#9  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string (this=0x7ffef23e4090, __in_chrg=<optimized out>) at /usr/include/c++/8/bits/basic_string.h:657
No locals.
#10 runshell (shell=0) at ./uuu/uuu.cpp:622
        uboot_cmd = 0
        prompt = {static npos = 18446744073709551615, _M_dataplus = {<std::allocator<char>> = {<__gnu_cxx::new_allocator<char>> = {<No data fields>}, <No data fields>}, _M_p = 0x7ffef23e40a0 "U>"}, 
          _M_string_length = 2, {_M_local_buf = "U>\000\362\376\177\000\000\377\207B;\022\177\000", _M_allocated_capacity = 140732958522965}}
#11 0x000055af05feb4a8 in main (argc=<optimized out>, argv=<optimized out>) at ./uuu/uuu.cpp:906
        a = <optimized out>
        deamon = <optimized out>
        shell = <optimized out>
        filename = {static npos = 18446744073709551615, _M_dataplus = {<std::allocator<char>> = {<__gnu_cxx::new_allocator<char>> = {<No data fields>}, <No data fields>}, 
            _M_p = 0x55af07db6360 "download/flash_devkit.lst"}, _M_string_length = 25, {_M_local_buf = "\036\000\000\000\000\000\000\000p\004\000\000\000\000\000", _M_allocated_capacity = 30}}
        cmd = {static npos = 18446744073709551615, _M_dataplus = {<std::allocator<char>> = {<__gnu_cxx::new_allocator<char>> = {<No data fields>}, <No data fields>}, _M_p = 0x7ffef23e4400 ""}, 
          _M_string_length = 0, {_M_local_buf = "\000\000\000\000\000\000\000\000\351\000\000\000y\000\000", _M_allocated_capacity = 0}}
        ret = 0
        cmd_script = {static npos = 18446744073709551615, _M_dataplus = {<std::allocator<char>> = {<__gnu_cxx::new_allocator<char>> = {<No data fields>}, <No data fields>}, _M_p = 0x7ffef23e4420 ""}, 
          _M_string_length = 0, {_M_local_buf = "\000D>\362\376\177\000\000\377\033\001\000\000\000\000", _M_allocated_capacity = 140732962587648}}
        nt_session = {_M_t = {
--Type <RET> for more, q to quit, c to continue without paging--
            _M_impl = {<std::allocator<std::_Rb_tree_node<std::pair<unsigned long const, ShowNotify> > >> = {<__gnu_cxx::new_allocator<std::_Rb_tree_node<std::pair<unsigned long const, ShowNotify> > >> = {<No data fields>}, <No data fields>}, <std::_Rb_tree_key_compare<std::less<unsigned long> >> = {
                _M_key_compare = {<std::binary_function<unsigned long, unsigned long, bool>> = {<No data fields>}, <No data fields>}}, <std::_Rb_tree_header> = {_M_header = {_M_color = std::_S_red, 
                  _M_parent = 0x0, _M_left = 0x7ffef23e43a8, _M_right = 0x7ffef23e43a8}, _M_node_count = 0}, <No data fields>}}}

```

agx, 2018-12-31:
It happens at the very end of this script when flashing on a imx8mq:

```
uuu_version 1.0.1

SDP: boot -f u-boot-devkit-recovery.imx
# This command will be run when use SPL
SDPU: delay 1000
SDPU: write -f u-boot-devkit-recovery.imx -offset 0x57c00
SDPU: jump
# This command will be run when ROM support stream mode
SDPS: boot -f u-boot-devkit-recovery.imx
SDPU: delay 1000
#FB: ucmd setenv fastboot_buffer 0x43000000
FB: ucmd setenv fastboot_dev mmc
FB: ucmd setenv mmcdev 0
FB: flash -raw2sparse all devkit.img
FB: Done
```

The files can be downloaded from here:

https://arm01.puri.sm/job/manual/job/agx-image-build/
https://arm01.puri.sm/job/u-boot_builds/job/devkit-recovery_uboot_build/

nxpfrankli, 2019-04-09:
I can't reproduce at latest version.  Can you try latest one?

agx, 2019-04-09:
Are you talking about master or uuu_dev branch?

agx, 2019-04-09:
With the latest release my local build with -O2 succeeds but our CI build (which builds a ton of other C/C++ packages) fails like

```
$ uuu flash_devkit.lst 
uuu (Universal Update Utility) for nxp imx chips -- lib

terminate called after throwing an instance of 'std::invalid_argument'
  what():  stoll
Aborted (core dumped)
```

i have to figure out the exact but maybe you already have an idea. this 6b465581bed5a93188781189bc9c73b56e65a1bc
```

nxpfrankli, 2019-04-09:
You can try prebuild one https://ci.appveyor.com/project/nxpfrankli/mfgtools-kvqcg/builds/23718938/job/a0rxxv7cb20icwpx/artifacts

agx, 2019-04-09:
Thanks but I'm not interested in prebuilt binaries. I can built working ones my own but the softwareis so brittle that e.g. dpkg-buildpackage passing in additioal cxxflags make the whole thing explode (as with -O2 before) and now with some other flag I'll have to dig out.

agx, 2019-04-10:
The last reported segfault is due to a broken gitversion.h so opening another issue.

closed 2019-04-10 by agx