nxp-imx/mfgtools #77: libuuu/gen/gitversion.h is missing in all recent packaged versions

JLMorange, 2019-01-10:
I tried to build using a pre-packaged version available on https://github.com/NXPmicro/mfgtools/releases
from 1.2.0 till 1.2.39
all fail because libuuu/gen/gitversion.h  is not included
I fixed the build script by using a git version, regenerating the file and re-generating it in my build script
best regards

nxpfrankli, 2019-01-10:
tar ball can't correct identify source version.  github don't provide customer script to generate it.  which is known issue

joerg-krause, 2019-04-30:
Why not define a version string in CMakeLists.txt, e.g. using the `project()` command, and use this version string if no git clone is used? This why, both the source tarball and the git clone will compile.

FloweyTheFlower420, 2019-04-30:
CMakeList.txt can't sync with git tag and commit hash automatically. 
github don't support external script to generate tarball.

joerg-krause, 2019-04-30:
Maybe I was not clear enough. I've pushed an example to my fork which does apply a version string even when building the source tarball: https://github.com/joerg-krause/mfgtools/tree/gitversion.

nxpfrankli, 2019-04-30:
If version upgrade 1.2.128,  I have to change CMakeList.txt (project(uuu VERSION 1.2.91)).

joerg-krause, 2019-04-30:
Yes. Is it a no-go?

joerg-krause, 2019-04-30:
Another solution would be to add a travis rule to automatically publish a release tarball when tagging a commit. Something like this (totally untested!):

```
jobs:
  include:
    - stage: deploy
      if: tag IS present
      script: skip
      deploy:
        provider: releases
        api_key: $GITHUB_OAUTH_TOKEN
        file: mfgtools-$TRAVIS_TAG.tar.gz
        overwrite: true
        skip_cleanup: true
        on:
          tags: true
```

nxpfrankli, 2019-04-30:
but in release page https://github.com/NXPmicro/mfgtools/releases,
source tar ball is generate by github, I am not sure how to replace it with generated by travis-ci

joerg-krause, 2019-04-30:
You're right, you cannot overwrite the github source tarball, but you can create your own one and upload it. I've tested it on my fork: https://github.com/joerg-krause/mfgtools. In [releases](https://github.com/joerg-krause/mfgtools/releases) you can see the created tarball `uuu_1.2.92.tar.gz`.

Note, I am not an expert in Travis. I've just re-used the code from some other project.
Note, this is just a test and I will remove the tarball later.

nxpfrankli, 2019-04-30:
but Source code (zip) and Source code (tar.gz), which will cause confuse.

joerg-krause, 2019-04-30:
Maybe. However, the automatically generated source tarballs from github have some disadvantages:

1. They can change at any point. The tarballs are generated and cached, once the cash is invalidated, a new tarball is generated by github. This new tarball will have a different checksum. This makes it difficult for package management systems as they rely on a valid checksum. An uploaded release tarball will not change (until you force it).
2. They do not contain submodules.

FloweyTheFlower420, 2019-05-01:
but we can't prevent github to create it.  So I don't prefer use tarball at all.

joerg-krause, 2019-05-01:
Note that the automatically generated source tarball does not even compile. In my opinion, that is confusing.

nxpfrankli, 2019-05-02:
That's problem,  we claim tarball is not supported before work out and good solution.

PR #171 (2020-03-13): libuuu: allow to build without git
Read .tarball-version if pressent, instead of git description.

----
Hi,

this is an idea to fix  #77 .
A tarball source release will have to add the .tarball-version file.
For example with a script like:
```bash
version=$(git describe --tags)
mkdir "$version"
echo "$version" > "$version/.tarball-version"
git archive --prefix="$version/" HEAD > "$version.tar"
tar --append -f "$version.tar" "$version/.tarball-version"
gzip "$version.tar"
```
closed 2020-03-13 by nxpfrankli