nxp-imx/mfgtools #78: uuu Linux script question for determining emmc device

gowricpms, 2019-01-14:
I am opening a new issue since this is different from the one we were previously discussing. Wanted to repeat that I am using i.MX8M Mini. My uuu script on Linux is able to RAM load u-boot, the kernel and rootfs. I am trying to understand how to determine what my emmc device number is. I saw that the sample script example_kernel_emmc.uuu has the following lines:

# get mmc dev number from kernel command line
# Wait for emmc
FBK: ucmd while [ ! -e /dev/mmcblk*boot0 ]; do sleep 1; echo "wait for /dev/mmcblk*boot* appear"; done;

# serach emmc device number, if your platform have more than two emmc chip, please echo dev number >/tmp/mmcdev 
FBK: ucmd dev=`ls /dev/mmcblk*boot*`; dev=($dev); dev=${dev[0]}; dev=${dev#/dev/mmcblk}; dev=${dev%boot*}; echo $dev > /tmp/mmcdev;

Could you please explain how this works? It looks like /tmp/mmcdev in my case is 2. However when I write the various binaries to the correct partitions on /sys/block/mmcblk2, it is unable to load the kernel and stops at u-boot. Here are my u-boot env parameters and their values:
baudrate=115200
boot_fdt=try
bootargs=console=ttymxc1,115200 earlycon=ec_imx6q,0x30890000,115200 root=/dev/nfs ip=dhcp nfsroot=:,v3,tcp
bootcmd=mmc dev ${mmcdev}; if mmc rescan; then if run loadbootscript; then run bootscript; else if run loadimage; then run mmcboot; else run netboot; fi; fi; else booti ${loadaddr} - ${fdt_addr}; fi
bootcmd_mfg=run mfgtool_args;booti ${loadaddr} ${initrd_addr} ${fdt_addr};
bootdelay=2
bootscript=echo Running bootscript from mmc ...; source
console=ttymxc1,115200 earlycon=ec_imx6q,0x30890000,115200
ethact=ethernet@30be0000
ethaddr=00:04:9f:05:9b:b5
ethprime=FEC
fdt_addr=0x43000000
fdt_file=fsl-imx8mm-evk.dtb
fdt_high=0xffffffffffffffff
fdtcontroladdr=bc92e3e8
get_cmd=dhcp
image=Image
initrd_addr=0x43800000
initrd_high=0xffffffffffffffff
loadaddr=0x40480000
loadbootscript=fatload mmc ${mmcdev}:${mmcpart} ${loadaddr} ${script};
loadfdt=fatload mmc ${mmcdev}:${mmcpart} ${fdt_addr} ${fdt_file}
loadimage=fatload mmc ${mmcdev}:${mmcpart} ${loadaddr} ${image}
mfgtool_args=setenv bootargs console=${console},${baudrate} rdinit=/linuxrc g_mass_storage.stall=0 g_mass_storage.removable=1 g_mass_storage.idVendor=0x066F g_mass_storage.idProduct=0x37FF g_mass_storage.iSerialNumber="" clk_ignore_unused
mmcargs=setenv bootargs console=${console} root=${mmcroot}
mmcautodetect=yes
mmcboot=echo Booting from mmc ...; run mmcargs; if test ${boot_fdt} = yes || test ${boot_fdt} = try; then if run loadfdt; then booti ${loadaddr} - ${fdt_addr}; else echo WARN: Cannot load the DT; fi; else echo wait for boot; fi;
mmcdev=0
mmcpart=1
mmcroot=/dev/mmcblk1p2 rootwait rw
netargs=setenv bootargs console=${console} root=/dev/nfs ip=dhcp nfsroot=${serverip}:${nfsroot},v3,tcp
netboot=echo Booting from net ...; run netargs;  if test ${ip_dyn} = yes; then setenv get_cmd dhcp; else setenv get_cmd tftp; fi; ${get_cmd} ${loadaddr} ${image}; if test ${boot_fdt} = yes || test ${boot_fdt} = try; then if ${get_cmd} ${fdt_addr} ${fdt_file}; then booti ${loadaddr} - ${fdt_addr}; else echo WARN: Cannot load the DT; fi; else booti; fi;
script=boot.scr
stderr=serial
stdin=serial
stdout=serial

This seems to suggest that it is trying to load the kernel out of /dev/mmcblk1p2 instead of /dev/mmcblk2p2. How can I fix this?

nxpfrankli, 2019-01-15:
FBK: ucmd while [ ! -e /dev/mmcblk*boot0 ]; do sleep 1; echo "wait for /dev/mmcblk*boot* appear"; done;

FBK: ucmd dev=`ls /dev/mmcblk*boot*`; dev=($dev); dev=${dev[0]}; dev=${dev#/dev/mmcblk}; dev=${dev%boot*}; echo $dev > /tmp/mmcdev;

FBK: ucmd mmc=`cat /tmp/mmcdev`; PARTSTR=$'10M,500M,0c\n600M,,83\n'; echo "$PARTSTR" | sfdisk --force /dev/mmcblk${mmc}

nxpfrankli, 2019-01-15:
uboot use logical index of usdhc.  kernel use physical index of usdhc.  it is confused. 

you can use fatls mmc 1:1 to check if correct file located fat partition. 

FBK: ucmd dev=ls /dev/mmcblk*boot*; dev=($dev); dev=${dev[0]}; dev=${dev#/dev/mmcblk}; dev=${dev%boot*}; echo $dev > /tmp/mmcdev;

emmc have boot partition, above script search mmcblk*bloot* to determinate mmcdev number in kernel without dependence with platform.

nxpfrankli, 2019-01-16:
It should be for RevB boards. 
SW1101 [1‐8]: 01110010
SW1102 [1‐8]: 00101010

Can you try uuu -b emmc files/imx-boot-imx8mmevk-sd.bin-flash_evk

then boot from emmc, check if uboot  boot?

gowricpms, 2019-01-16:
According to Freescale documentation, uboot should be written at an offset of 33K for iMX8. The xml for Windows MfgTool uses the correct offset too. Uboot boots up now but I now have a kernel panic:

[    4.024314] ---[ end Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/admin-guide/init.rst for guidance.

I'll continue looking into this but if you have any idea about what could be wrong, it would really help!

nxpfrankli, 2019-01-16:
Latest tar will omit some symblink, 
please use below command

FBK: acmd export EXTRACT_UNSAFE_SYMLINKS=1; tar -jx -C /mnt/mtd

gowricpms, 2019-01-16:
That worked! Thanks so much :)

closed 2019-01-16 by gowricpms

pingwuu, 2020-05-11:
@gowricpms & @nxpfrankli ,
According to Freescale documentation, uboot should be written at an offset of 33K for iMX8
===
which document provide this information for "offset of 33K for iMX8"? Thanks.