nxp-imx/mfgtools #83: IMX8M Mini UUU tool issue

sekumare84, 2019-03-04:
Hi ,
I have downloaded pre-built uuu tool binary version 1.2.0 & 1.2.91 from NXPmicro/mfgtools repo. 
  - I have installed Ubuntu 18.04 in my windows as a virtual machine and forwarded the USB OTG (imx8mm mini eval board ) as a USB device to the virtual machine. 
  - The core-minimal-image has been built.
  - The core-minim-image-xx-sdcard.bz2 and imx8xxx-sd.bin-flash_evk files have been copied into the virtual machine. 
 -  The uuu.auto file is present in the same folder where the images are available. 
 -  Make sure that NXP semiconductor device is available in the virtual machine using lsusb command. 
 -   The following command executed in linux machine
         - sudo ./uuu Images/
             - The "images" is a directory where the core-minimal-image and imx8xx-sd.bin-flashe_evk files are available. 
   - After executing the above command, the first command alone executed and the uuu tool is not proceeding further. 
 -  When I have verified the lsusb, the NXP semiconductor USB device is not visisble. 
 - Not sure why this is happening only in Linux and the same version of UUU tool works well in Windows machine.

**Logs for reference** 
-----------------------
myVirtualBox:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 010: ID 1fc9:0134 NXP Semiconductors 
Bus 001 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

**UUU logs :** 
-----------------
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.91-0-g3799f4d

SDP: boot -f images/imx-boot-imx8xx-sd.bin-flash_evk
# This command will be run when use SPL
SDPU: delay 1000
SDPU: write -f images/imx-boot-imx8xx-sd.bin-flash_evk -offset 0x57c00
SDPU: jump
# This command will be run when ROM support stream mode
SDPS: boot -f images/imx-boot-imx8xxx-sd.bin-flash_evk

Wait for Known USB Device Appear
New USB Device Attached at 1:2
1:2>Start Cmd:SDP: boot -f images/imx-boot-imx8xx-sd.bin-flash_evk
6400%1:2>Okay

nxpfrankli, 2019-03-04:
virtual machine is quite complex, 8mm need switch 3 usb work mode.

ROM,  SPL and fastboot.  

You need switch 3 type device into virtual machine.  Their VID and PID is difference.

sekumare84, 2019-03-04:
Hi Frank,
- You do we need to configure imx8mm to switch 3 usb work mode? How to do that? 
- How to do switch type 3 device in virutal machine. What is the VID and PID?

Thanks.

sekumare84, 2019-03-04:
Additional info, I have configured USB 3.0 (xHCI) controller in the virtual machine.

nxpfrankli, 2019-03-04:
Whole flow as

ROM serial download, 
1. uuu use SDP download spl
2. SPL running,  USB switch to HID mode (the same protocol, but difference PID\VID) to download uboot
3. uboot running, switch to FB mode, 
4. uuu  continue download other stuff by FB.  

SDP:     MX8MM           0x1fc9  0x0134,   ROM use as HID device.
SDPV:    SPL1            0x0525  0xb4a4       SPL use it as HID to download uboot
FB:                      0x0525  0xa4a5              uboot use as generic bulk io to download other stuff.

I don't how to change virtual machine automatically.

sekumare84, 2019-03-04:
Ok.

sekumare84, 2019-03-04:
Hi Frank,

Added all the USB devices in the virtual machine USB filter and used uuu tool. Able to download/program the images in to EMMC. 

Thanks.

closed 2019-03-04 by nxpfrankli