nxp-imx/mfgtools #84: Is it supported to combine running scripts with single commands?

manszigher, 2019-03-08:
Hi,

I have an use-case where I want to run a script fastboot.lst that will make sure I get my device into FB. After entering FB I need to run some FB: flash commands flashing some files. After flashing all my files I would like to run a script again finalize.lst which will run some FB: ucmd. All of this I can successfully do with a single script but when trying to separate into the described setup I always get stuck when trying to run switch from running the FB: flash to run the finalize.lst script. Is this a scenario that should be supported?

BR
Måns Zigher

manszigher, 2019-03-08:
Trying to set it to a question since it is not really a issue.

manszigher, 2019-03-08:
Here is my script that I am trying to split up
<pre>
uuu_version 1.0.1

SDP: boot -f uboot.boot
# This command will be run when use SPL
SDPU: delay 1000
SDPU: write -f uboot.boot -offset 0x57c00
SDPU: jump
# This command will be run when ROM support stream mode
SDPS: boot -f uboot.boot

FB: ucmd setenv fastboot_dev mmc
FB: ucmd setenv mmcdev ${emmc_dev}
FB: ucmd mmc dev ${emmc_dev}

# The "gpt" partition is special in that it causes FastBoot to reload its cached
# partition table from disk, which means it must be the first image we flash for
# the rest of the partition names to work.
FB: flash gpt partition.gpt
FB: flash u-boot uboot.boot
FB: flash kernel-a linux.kernel
FB: flash kernel-b linux.kernel
FB[-t 30000]: flash rootfs-a rootfsa.rootfs
FB[-t 30000]: flash rootfs-b rootfsb.rootfs
FB[-t 30000]: flash -raw2sparse device-data data.device-data
FB: ucmd mmc partconf ${emmc_dev} 0 1 0

FB: ucmd setenv fastboot_buffer ${loadaddr}
FB: download -f data.eeprom
FB: ucmd bno write ${fastboot_buffer} ${fastboot_bytes}

FB: ucmd env default -a
FB: ucmd saveenv

FB: done
</pre>

manszigher, 2019-03-08:
I am trying to split this up into a fastboot.lst

<pre>
uuu_version 1.0.1

SDP: boot -f uboot.boot
# This command will be run when use SPL
SDPU: delay 1000
SDPU: write -f uboot.boot -offset 0x57c00
SDPU: jump
# This command will be run when ROM support stream mode
SDPS: boot -f uboot.boot

FB: ucmd setenv fastboot_dev mmc
FB: ucmd setenv mmcdev ${emmc_dev}
FB: ucmd mmc dev ${emmc_dev}
FB: done
</pre>

and then run the following commands as single commands

<pre>
FB: flash gpt partition.gpt
FB: flash u-boot uboot.boot
FB: flash kernel-a linux.kernel
FB: flash kernel-b linux.kernel
FB[-t 30000]: flash rootfs-a rootfsa.rootfs
FB[-t 30000]: flash rootfs-b rootfsb.rootfs
FB[-t 30000]: flash -raw2sparse device-data data.device-data
</pre>

And at the end run the finalize.lst script

<pre>
FB: ucmd mmc partconf ${emmc_dev} 0 1 0

FB: ucmd setenv fastboot_buffer ${loadaddr}
FB: download -f data.eeprom
FB: ucmd bno write ${fastboot_buffer} ${fastboot_bytes}

FB: ucmd env default -a
FB: ucmd saveenv

FB: done
</pre>

manszigher, 2019-03-08:
Splitting the finalize.lst script into single uuu commands and running them one by one seems to work. Any ideas why it will not work to go from running single uuu commands to running a uuu script?

nxpfrankli, 2019-03-08:
did you add "uuu_version 1.0.1" in first line at your split script?

manszigher, 2019-03-11:
No your are right I had it commented out for some reason. Thanks!

closed 2019-03-11 by manszigher