nxp-imx/mfgtools #85: [QUESTION] Is the i.MX 28 supported?

karora, 2019-03-13:


nxpfrankli, 2019-03-13:
Not yet.  it is simple. but I have not environment to test.

karora, 2019-03-13:
I have i.MX28 boards, and would *love* to lose the pain of the Windows 7 VM I need to have around just to run mfgtool...  Is there anything I could do to help?

nxpfrankli, 2019-03-13:
prebuild windows version.
https://ci.appveyor.com/project/nxpfrankli/mfgtools-kvqcg/builds/23049006/job/m5c0s90jrl9dmprb/artifacts

prebuild linux verison

https://ci.appveyor.com/project/nxpfrankli/mfgtools-kvqcg/builds/23049006/job/ilt1onrwonkgr6g1/artifacts

try 

uuu loader.sb

karora, 2019-03-14:
Using that build on Debian Linux, I am able to at least load a u-boot image:

```
$ ./uuu save/u-boot+2019.01.sb
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.96-0-g993996a

Success 1    Failure 0                                                                                                                               
                                                                                                                                                      
1:22   2/ 2   [Done                        ] SDPS: done              
```

I'll have to read the manual some more to see if I can write a full image to the Nand this way, which would be awesome, but meanwhile this is certainly useful.

Thanks!

nxpfrankli, 2019-03-14:
It needs new kernel, which support usb configfs. and integrate new imx-uuc in initramfs

karora, 2019-03-14:
We're using Linux 4.14 on our I.MX283 boards, so that should be doable for me. I'll work on it...

karora, 2019-03-17:
I'm now trying to build a u-boot to match, following the instructions at:
  https://github.com/NXPmicro/mfgtools/wiki/uboot-config-requirement

I added the following to my config:
```
CONFIG_CMD_FASTBOOT=y
CONFIG_USB_FUNCTION_FASTBOOT=y
CONFIG_USB_GADGET=y
CONFIG_USB_GADGET_DOWNLOAD=y
CONFIG_USB_GADGET_MANUFACTURER="FSL"
CONFIG_USB_GADGET_VENDOR_NUM=0x0525
CONFIG_USB_GADGET_PRODUCT_NUM=0xa4a5
CONFIG_CI_UDC=y      # UDC need change according system, some system use CONFIG_USB_DWC3, some use CONFIG_USB_CDNS3
CONFIG_FSL_FASTBOOT=y
CONFIG_FASTBOOT=y
CONFIG_FASTBOOT_BUF_ADDR=0x42000000
CONFIG_FASTBOOT_BUF_SIZE=0x40000000
CONFIG_FASTBOOT_FLASH=y
CONFIG_FASTBOOT_FLASH_MMC_DEV=1
CONFIG_EFI_PARTITION=y
CONFIG_ANDROID_BOOT_IMAGE=y

CONFIG_SPL_USB_HOST_SUPPORT=y
CONFIG_SPL_USB_GADGET_SUPPORT=y
CONFIG_SPL_USB_SDP_SUPPORT=y
CONFIG_SDP_LOADADDR=0x40400000

```

I assume the gadget vendor / product need to stay as written, but the comment there about "CONFIG_CI_UDC=y"  that "UDC need change according system, some system use CONFIG_USB_DWC3, some use CONFIG_USB_CDNS3" sadly means nothing to me.  Do I need to change it?  How do I know what to set it to?

Similarly, I've just used `CONFIG_SDP_LOADADDR=0x40400000` as-is, but there's a comment about this needing to point to free space.  How do I find a reasonable value for the i.MX283?

Thanks for all your help!

nxpfrankli, 2019-03-18:
Yes, mx28 use CONFIG_CI_UDC

Generally, mx28 doesn't need SPL. 

if you want to use SPL,  you can use LOAD_ADDR, which is in config header file.

karora, 2019-03-20:
Thanks for all your help - I think we can close this now.

closed 2019-03-20 by karora