nxp-imx/mfgtools #89: [Question] IMX8QM load bootloader and exit

embexus, 2019-03-15:
Hello,
I didn't figure out how just to load bootloader over SDP and exit when it has started ?
I don't want uuu to still waiting.

zaki@xps:~/Desktop/B2-B0$ sudo uuu -v uuu.auto 
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.56-3-g0ffa8ea

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0xffff]
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02
Wait for Known USB Device Appear...
New USB Device Attached at 3:26
New USB Device Attached at 3:26
3:26>Start Cmd:SDPS: boot -f flash.bin
100%3:26>Okay
New USB Device Attached at 3:26

nxpfrankli, 2019-03-15:
If don't use spl',  
uuu flash.bin

if use spl, 
uuu -b spl flash.bin

embexus, 2019-03-15:
Thank you ! that was easy, although didn't figure it out of documentation.

nxpfrankli, 2019-03-15:
https://github.com/NXPmicro/mfgtools/wiki

````
Download uboot
uuu bootloader
````

Do you think above description is not clear?

closed 2019-03-25 by nxpfrankli