nxp-imx/mfgtools #91: issue with flashing Universal update utility for iMX8MQuad and i.MX8MQLite

mathanrajmurugan, 2019-03-26:
When I tried to flashing the Yocto build Image on the board using UUU tool of the-14-78.

But bootloader is fine, But when I tried to write rootfs.sdcard it giving error log message,
Please clarify the same, I saw some threads, still don't have support for  L4.14.78 release
Is sorted out and still in beta version.

sudo ./uuu/uuu -b emmc_all imx-boot-imx8mq-sd.bin-flash_evk core-image-minimal-imx8mq-20190325111905.rootfs.sdcard

uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.91-5-g8019217

Success 0 Failure 1

3:3 1/ 8 [Bulk(R):LIBUSB_ERROR_TIMEOUT ] FB: ucmd setenv fastboot_dev mmc

nxpfrankli, 2019-03-26:
Any output from uart. 
recently, we found there are problem at some PC host USB3.0 port. 

Try apply below uboot patch to disable U1 and U2. 

Or try to connect to USB2.0 port

````
commit ae0cdc741f7313398539f7b576ced73871f6cd71
Author: Li Jun <jun.li@nxp.com>
Date:   Wed Mar 6 16:15:01 2019 +0800

    MLK-21055 usb: dwc3: gadget: disable U1 and U2

    Currenlty U1 and U2 low power modes are allowed in device mode.
    Allowing U1 and U2 low power modes during data transfers in
    device mode is causing U1 exit failure on some USB3 host: which
    will transite to SS.inactive instead of U0, then host will send
    warm reset and ultimately result in reenumeration. This is observed
    on UUU tool with some PC host. Hence disable U1 and U2 low power
    modes for now.

    USB3 spec 7.5.10.4.2 Exit from Recovery.Configuration
    The port shall transition to eSS.Inactive when the following
    conditions are met:
    1. Either the Ux_EXIT_TIMER or the 6-ms timer
       (tRecoveryConfigurationTimeout) times out.
    2. For a downstream port, the transition to Recovery is not to
       attempt a Hot Reset.

    Signed-off-by: Li Jun <jun.li@nxp.com>

diff --git a/drivers/usb/dwc3/ep0.c b/drivers/usb/dwc3/ep0.c
index e61d980..84548b8 100644
--- a/drivers/usb/dwc3/ep0.c
+++ b/drivers/usb/dwc3/ep0.c
@@ -391,7 +391,6 @@ static int dwc3_ep0_handle_feature(struct dwc3 *dwc,
        u32                     recip;
        u32                     wValue;
        u32                     wIndex;
-       u32                     reg;
        int                     ret;
        enum usb_device_state   state;

@@ -415,27 +414,12 @@ static int dwc3_ep0_handle_feature(struct dwc3 *dwc,
                                return -EINVAL;
                        if (dwc->speed != DWC3_DSTS_SUPERSPEED)
                                return -EINVAL;
-
-                       reg = dwc3_readl(dwc->regs, DWC3_DCTL);
-                       if (set)
-                               reg |= DWC3_DCTL_INITU1ENA;
-                       else
-                               reg &= ~DWC3_DCTL_INITU1ENA;
-                       dwc3_writel(dwc->regs, DWC3_DCTL, reg);
                        break;
-
                case USB_DEVICE_U2_ENABLE:
                        if (state != USB_STATE_CONFIGURED)
                                return -EINVAL;
                        if (dwc->speed != DWC3_DSTS_SUPERSPEED)
                                return -EINVAL;
-
-                       reg = dwc3_readl(dwc->regs, DWC3_DCTL);
-                       if (set)
-                               reg |= DWC3_DCTL_INITU2ENA;
-                       else
-                               reg &= ~DWC3_DCTL_INITU2ENA;
-                       dwc3_writel(dwc->regs, DWC3_DCTL, reg);
                        break;

                case USB_DEVICE_LTM_ENABLE:
@@ -539,7 +523,6 @@ static int dwc3_ep0_set_config(struct dwc3 *dwc, struct usb_ctrlrequest *ctrl)
        enum usb_device_state state = dwc->gadget.state;
        u32 cfg;
        int ret;
-       u32 reg;

        dwc->start_config_issued = false;
        cfg = le16_to_cpu(ctrl->wValue);
@@ -563,14 +546,6 @@ static int dwc3_ep0_set_config(struct dwc3 *dwc, struct usb_ctrlrequest *ctrl)
                                usb_gadget_set_state(&dwc->gadget,
                                                USB_STATE_CONFIGURED);

-                       /*
-                        * Enable transition to U1/U2 state when
-                        * nothing is pending from application.
-                        */
-                       reg = dwc3_readl(dwc->regs, DWC3_DCTL);
-                       reg |= (DWC3_DCTL_ACCEPTU1ENA | DWC3_DCTL_ACCEPTU2ENA);
-                       dwc3_writel(dwc->regs, DWC3_DCTL, reg);
-
                        dwc->resize_fifos = true;
                        dev_dbg(dwc->dev, "resize FIFOs flag SET");
                }
diff --git a/drivers/usb/dwc3/gadget.c b/drivers/usb/dwc3/gadget.c
index 2727126..28c40d2 100644
--- a/drivers/usb/dwc3/gadget.c
+++ b/drivers/usb/dwc3/gadget.c
@@ -2563,6 +2563,7 @@ static irqreturn_t dwc3_interrupt(int irq, void *_dwc)
 int dwc3_gadget_init(struct dwc3 *dwc)
 {
        int                                     ret;
+       u32 reg;

        dwc->ctrl_req = dma_alloc_coherent(sizeof(*dwc->ctrl_req),
                                        (unsigned long *)&dwc->ctrl_req_addr);
@@ -2621,6 +2622,10 @@ int dwc3_gadget_init(struct dwc3 *dwc)
                goto err4;
        }

+       reg = dwc3_readl(dwc->regs, DWC3_DCTL);
+       reg &= ~(DWC3_DCTL_INITU1ENA | DWC3_DCTL_INITU2ENA);
+       dwc3_writel(dwc->regs, DWC3_DCTL, reg);
+
        return 0;

 err4:
````

mathanrajmurugan, 2019-03-26:
Okay , Thanks for the patch.
let me try to apply patch in the uboot source and update you!
And for the answer : Yes I can see the output log of Bootloader, it successfully flashed when it's trying to load the full yocto image of rrotfs.sdcard it gives an error in the libusb error timout

mathanrajmurugan, 2019-03-27:
Still same error , Any suggestion ??

nxpfrankli, 2019-03-27:
Can you give me UART log?

mathanrajmurugan, 2019-03-27:
Console log:
root@Mathan-SECO:/home/seco/Music# picocom -b 115200 -r -l /dev/ttyUSB0
picocom v1.7

port is        : /dev/ttyUSB0
flowcontrol    : none
baudrate is    : 115200
parity is      : none
databits are   : 8
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : yes
nolock is      : yes
send_cmd is    : sz -vv
receive_cmd is : rz -vv
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,

Terminal ready

U-Boot SPL 2018.03+fslc (Mar 27 2019 - 04:49:52 +0000)
SPL PMIC:  PFUZE100 ID=0x10
C12 Micro fw: 0xf01
C12 RAM 1GB
DDRINFO: start lpddr4 ddr init
DRAM PHY training for 3200MTS
check ddr4_pmu_train_imem code
check ddr4_pmu_train_imem code pass
check ddr4_pmu_train_dmem code
check ddr4_pmu_train_dmem code pass
Training PASS
DRAM PHY training for 667MTS
check ddr4_pmu_train_imem code
check ddr4_pmu_train_imem code pass
check ddr4_pmu_train_dmem code
check ddr4_pmu_train_dmem code pass
Training PASS
DRAM PHY training for 3200MTS
check ddr4_pmu_train_imem code
check ddr4_pmu_train_imem code pass
check ddr4_pmu_train_dmem code
check ddr4_pmu_train_dmem code pass
Training PASS
DDRINFO:ddrphy calibration done
DDRINFO: ddrmix config done
Normal Boot
Trying to boot from USB SDP
SDP: initialize...
SDP: handle requests...
Downloading file of size 1083276 to 0x40400000... done
Jumping to header at 0x40400000
Header Tag is not an IMX image

U-Boot 2018.03+fslc (Mar 27 2019 - 04:49:52 +0000)

CPU:   Freescale i.MX8MQLite rev2.0 1500 MHz (running at 1000 MHz)
CPU:   Commercial temperature grade (0C to 95C) at 50C
Reset cause: POR
Model: SECO i.MX8MQ C12
DRAM:  992 MiB
MMC:   FSL_SDHC: 0, FSL_SDHC: 1
Loading Environment from MMC... *** Warning - bad CRC, using default environment

Failed (-5)
No panel detected: default to HDMI
dcss is diabled on i.MX8MQL
LCD HDMI cannot be configured: -1
In:    serial
Out:   serial
Err:   serial

 BuildInfo:
  - ATF d6451cc
  - U-Boot 2018.03+fslc

Detect USB boot. Will enter fastboot mode!
Net:   
Error: ethernet@30be0000 address not set.
No ethernet found.
Fastboot: Normal
Boot from USB for uuu
Hit any key to stop autoboot:  0 
request 000000007c934a80 was not queued to ep1in-bulk
Detect USB boot. Will enter fastboot mode!
flash target is MMC:1
================it stops here====================

Host system:
root@Mathan-SECO:/home/seco/mfgtools# sudo ./uuu/uuu -b emmc_all flash.bin core-image-minimal-seco-imx8mq-c12-20190327044741.rootfs.sdcard
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.91-5-g8019217

Success 0    Failure 1                                                                                                                                                                                      
                                                                                                                                                                                                             
3:7    1/ 8   [Bulk(R):LIBUSB_ERROR_TIMEOUT          ] FB: ucmd setenv fastboot_dev mmc

nxpfrankli, 2019-03-27:
look like uboot have not detect USB connection.

Can you try "ctrl-c" in uboot control
then remove usb cable

then run "fastboot 0"
plug usb cable again.

host run 
sudo ./uuu/uuu -b emmc_all flash.bin core-image-minimal-seco-imx8mq-c12-20190327044741.rootfs.sdcard

mathanrajmurugan, 2019-03-27:
Bulk(R):LIBUSB_ERROR_TIMEOUT ===> error released ..
FB: ucmd setenv fastboot_dev mmc ---> After run the fastboot 0 command into the uboot ,
FB: ucmd setenv mmcdev ${emmc_dev} --> current error

nxpfrankli, 2019-03-27:
which board do you use?

mathanrajmurugan, 2019-03-27:
iMX8M with SECO's one

nxpfrankli, 2019-03-29:
Do you use virtual machine?

mathanrajmurugan, 2019-03-31:
No, I am using 64 bit Ubuntu 18.04 Distro

mathanrajmurugan, 2019-04-02:
I feel that mmc is not getting started when i load the bootloader from UUU ,

In uboot log I can see, Loading Environment from MMC... MMC: no card present
*** Warning - No block device, using default environment
Then I set the mmc enviornment device to 0 eg: #define CONFIG_SYS_MMC_ENV_DEV		0
Log goes, But no success in eMMC load.

Can you please suggest what could be the error?

li-ruiqi, 2019-04-02:
look like uboot's usb driver can't get data from host.

nxpfrankli, 2019-04-02:
look like uboot usb driver have received any data sent by uuu.

mathanrajmurugan, 2019-04-05:
okay, So problem with Uboot USB driver if I m correct ?

nxpfrankli, 2019-04-05:
Yes,

closed 2019-05-08 by mathanrajmurugan