nxp-imx/mfgtools #93: U-boot flashing on iMX8MQ not working

batt, 2019-04-05:
Hello, 
we have just started to use UUU for Linux (previously we used mfgtools on windows).
For now we are only trying to flash u-boot on the eMMC.

Here is the command used:
`./uuu -V -b emmc flash.bin
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.91-6-ge742d94

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDPS:	 MX28		 0x15a2	 0x004f
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0x9998]
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x9999..0x9999]
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02

Run built-in script:

uuu_version 1.2.39

# @flash.bin            | bootloader
# @flash.bin   [flash.bin] | image burn to emmc, default is the same as bootloader

# This command will be run when i.MX6/7 i.MX8MM, i.MX8MQ
SDP: boot -f flash.bin

# This command will be run when ROM support stream mode
# i.MX8QXP, i.MX8QM
SDPS: boot -f flash.bin

# These commands will be run when use SPL and will be skipped if no spl
# SDPU will be deprecated. please use SDPV instead of SDPU
# {
SDPU: delay 1000
SDPU: write -f flash.bin -offset 0x57c00
SDPU: jump
# }

# These commands will be run when use SPL and will be skipped if no spl
# if (SPL support SDPV)
# {
SDPV: delay 1000
SDPV: write -f flash.bin -skipspl
SDPV: jump
# }

FB: ucmd setenv fastboot_dev mmc
FB: ucmd setenv mmcdev ${emmc_dev}
FB: ucmd mmc dev ${emmc_dev}
FB: flash bootloader flash.bin
FB: ucmd if env exists emmc_ack; then ; else setenv emmc_ack 0; fi;
FB: ucmd mmc partconf ${emmc_dev} ${emmc_ack} 1 0
FB: Done

Wait for Known USB Device Appear...
New USB Device Attached at 1:16
1:16>Start Cmd:SDP: boot -f flash.bin
6400%1:16>Okay`

The serial log from the iMX8MQ processor is the following:
`U-Boot SPL 2017.03-imx_v2017.03_4.9.88_2.0.0_ga+gb76bb1b (Feb 25 2019 - 13:19:14)
PMIC:  PFUZE100 ID=0x10
check ddr4_pmu_train_imem code
check ddr4_pmu_train_imem code pass
check ddr4_pmu_train_dmem code
check ddr4_pmu_train_dmem code pass
Training PASS
check ddr4_pmu_train_imem code
check ddr4_pmu_train_imem code pass
check ddr4_pmu_train_dmem code
check ddr4_pmu_train_dmem code pass
Training PASS
check ddr4_pmu_train_imem code
check ddr4_pmu_train_imem code pass
check ddr4_pmu_train_dmem code
check ddr4_pmu_train_dmem code pass
Training PASS
Back to ROM, SDP`

After this, the whole process freeze, nothing happens.

Do you have any suggestions on what is happening?

nxpfrankli, 2019-04-05:
"Back to ROM, SDP" means you use old version uboot. 

Please update to last GA one.

closed 2019-04-10 by nxpfrankli

batt, 2019-04-12:
Thank you, where can I find the new u-boot versions?

nxpfrankli, 2019-04-12:
https://www.nxp.com/products/processors-and-microcontrollers/arm-based-processors-and-mcus/i.mx-applications-processors/i.mx-8-processors/i.mx-software-and-development-tool:IMX-SW