nxp-imx/mfgtools #94: uuu tool issue - doesnt detect USB device

Haturi89, 2019-04-08:
Hi,
I have a eval kit from NXP MX8QXPMEK. I can boot it with the pre-configured as SD-Card without problems. When I doesnt insert the SD-Card board switches to SDP. In my device manager the board is detected. The old MFGTools shows "HID device detected".

Now  I am trying to download the pre-compiled u-boot. When I start the uuu (Version 1.2.91) it shows me the message Wait for Known USB Device Appear... and wait. It look like the USB device (board) is not detected.

How can I solve this?

I am using Windows10 as host system.

regards Haturi

[image]

[image]

nxpfrankli, 2019-04-08:
You are using QXP A0 chip, which is not support by UUU. 
A0 chip is not production chip.

Haturi89, 2019-04-09:
Okay but there must be a way how to get it to work. I mean we have purchased those CPUs from NXP so we can´t develop if it´s doesnt work. Furthermore we have bought a development kit (i.MX8QXP-MEK) and it also doesnt work with uuu.
So what should I modify to get it to work?

regards,
Haturi

nxpfrankli, 2019-04-09:
The whole boot flow is changed in QXP B0 totally.  A0 Chip is not production one. If you purchased it, you should contact sale to replace it with B0 chip.

Haturi89, 2019-04-10:
Okay thanks for the explanation.

closed 2019-04-10 by Haturi89