nxp-imx/mfgtools #108: uuu 1.2.135 FBK usblib Bulk(R):LIBUSB_ERROR_PIPE

lskrzypczak, 2019-06-05:
Hi,
When running FBK command:
FBK: ucmd mount -t debugfs debugfs /sys/kernel/debug
I'm getting this:
[Bulk(R):LIBUSB_ERROR_PIPE             ]
uuu is ran on Debian in VirtualBox 6 on Win10 x64 host. SDP is working fine.

Best regards, Lukasz

nxpfrankli, 2019-06-19:
Does it work at windows environment?  what's pid/vid after kernel boot

zephray, 2019-08-14:
I am having similar issue, I am using kernel and initrd from L4.9.88_2.0.0_mfg-tool.tar.gz. After booting up the kernel, the pid/vid are 066f:37ff, which is not supported by uuu by default. I added 066f:377f to the list in config.cpp and recompile. I get the same [Bulk(R):LIBUSB_ERROR_PIPE ] if I try to use FBK command.
OS: Ubuntu 18.04 LTS on physical machine. SDP works fine.

nxpfrankli, 2019-08-14:
uuu is not compatible old mfgtools initramfs. You should take one from at least 4.14 GA.

zephray, 2019-08-14:
Thanks for the clarification. Is 4.14 GA released yet? I couldn’t seem to find it on NXP i.MX Linux BSP page.

nxpfrankli, 2019-08-14:
https://www.nxp.com/design/software/embedded-software/linux-software-and-development-tools/embedded-linux-for-i.mx-applications-processors:IMXLINUX?tab=Design_Tools_Tab

click "more" under BSP, Drivers and Middleware

zephray, 2019-08-14:
Thanks, after switching to 4.14 GA, it works fine now.

closed 2019-10-10 by nxpfrankli