nxp-imx/mfgtools #112: Support of NAND devices ?

myfreescalewebpage, 2019-07-02:
Hi,
I'm starting a new work with an iMX6 module and was looking at uuu for loading u-boot + u-boot-env + kernel + dtb + rootfs on the target.
My system has a NAND and I would like to know if uuu is able to load every thing on the target ? There is no example to flash NAND devices, only eMMC.
Else, is it something scheduled ? Else, do you advice an other tool to achieve that ?
Thanks for the help,
Joel

PS : I generate the binaries with yocto.

nxpfrankli, 2019-07-03:
In release package, there are example NAND script in example directory

nxpfrankli, 2019-07-03:
just update https://github.com/NXPmicro/mfgtools/wiki/Sample-script

closed 2019-07-03 by nxpfrankli

myfreescalewebpage, 2019-07-03:
Hello @nxpfrankli thanks for the answer.
I'm not sure what the example directory is, I have check khttps://github.com/NXPmicro/mfgtools/releases and not found it. I have check the example on the wiki, but this is still unclear how to use that... Are you aware of a tutorial ?
Joel