nxp-imx/mfgtools #114: Error: unknown Option0

keelung-yang, 2019-07-15:
i.MX6D, Win10, uuu v1.2.135
------------------
my script aaa.lst:

> uuu_version 1.2.135
> SDP: dcd -f firmware/u-boot.imx
> SDP: write -f firmware/u-boot.imx -ivt 0
> SDP: write -f firmware/zImage -addr 0x12000000
> SDP: write -f firmware/imx6q-sabreauto-gpmi-weim.dtb -addr 0x18000000
> SDP: jump -f firmware/u-boot.imx -ivt 0

After _uuu.exe aaa.lst_, I got:

> uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.135-0-gacaf035
> 
> Wait for Known USB Device Appear...
> Error: unknown Option0
>

nxpfrankli, 2019-07-15:
SDP: jump -f firmware/u-boot.imx -ivt 0

Remove 0 after -ivt.   That's means jump to IVT address.

keelung-yang, 2019-07-16:
@nxpfrankli Thanks! It works!

closed 2019-07-16 by keelung-yang