nxp-imx/mfgtools #130: command-line arguments in UUU, variables in script

rveens, 2019-09-19:
Dear UUU developers, 

In the old MFG tool we had the option to define variables in the cfg.ini file.

This enabled us to pass some information like product code, MAC address etc to the MFG tool.

Does there exist a similar feature in UUU?  i could not find it.

nxpfrankli, 2019-09-19:
UUU use the another way to do that. 

sh\shell <any shell command> #Run shell command, such as                             wget to file from network
                      <        <any shell command> #use shell command's output                             as uuu command
                               this command generally used for burn some sequence number, such production id, mac address
                                for example:
                                      FB:< echo ucmd print

Use a customized script generate uuu cmd dynamically. 
It is more flexible than ini file.  especially for security key  generate.

rveens, 2019-09-24:
Hi Frank,

In my case i am not able to use the shell option you describe, as the product number i need to provide does not come from the Linux imx device but from a windows / linux PC.

I made a pull request that makes it possible to run scripts with arguments from files:

https://github.com/NXPmicro/mfgtools/pull/131

Maybe you can take a look at it.