nxp-imx/mfgtools #132: Get command output in fastboot

rubenmoral, 2019-09-26:
Is there any way to get the output of a command in fastboot? For example, if I run `uuu FB: ucmd part list mmc 0`, uuu just reports `Okay` if the command is correct or `Error` if it is not , but not the response to that command.

nxpfrankli, 2019-10-08:
It is not easy to capture uboot's output. we have pilot version. but it is not stable enough.

rubenmoral, 2019-11-07:
@nxpfrankli, is there any chance to get that pilot version? How does it work? Would it require changes in u-boot as well? Thanks in advance!

rschrader, 2019-12-03:
I am also very interested in this feature. Any news to this or the pilot version?

nxpfrankli, 2019-12-13:
you try below uboot patch at v2019
``````
From dea93b707adf7c31ce1bf9d3a77df543e5dbf3cd Mon Sep 17 00:00:00 2001
From: Frank Li <Frank.Li@nxp.com>
Date: Fri, 13 Dec 2019 14:36:01 -0600
Subject: [PATCH 1/1] MLK-23124 fastboot: support duplicate uart console output
 to uuu

to enable it, in uuu script add below command
fb: ucmd setenv stdout serial,fastboot
uuu add -v opition.
output will show in uuu console

---
 drivers/fastboot/Kconfig        |  2 ++
 drivers/usb/gadget/f_fastboot.c | 40 +++++++++++++++++++++++++++++++++
 2 files changed, 42 insertions(+)

diff --git a/drivers/fastboot/Kconfig b/drivers/fastboot/Kconfig
index 725135a694..069704da05 100644
--- a/drivers/fastboot/Kconfig
+++ b/drivers/fastboot/Kconfig
@@ -80,6 +80,8 @@ config FASTBOOT_UUU_SUPPORT
        select CMD_GPT
        select RANDOM_UUID
        select CMD_GPT_RENAME
+       select CONSOLE_MUX
+       select SYS_STDIO_DEREGISTER
        help
          The fastboot protocol includes "UCmd" command and "ACmd" command

diff --git a/drivers/usb/gadget/f_fastboot.c b/drivers/usb/gadget/f_fastboot.c
index 11de3bd129..70977d50d8 100644
--- a/drivers/usb/gadget/f_fastboot.c
+++ b/drivers/usb/gadget/f_fastboot.c
@@ -19,6 +19,8 @@
 #include <linux/usb/composite.h>
 #include <linux/compiler.h>
 #include <g_dnl.h>
+#include <serial.h>
+#include <stdio_dev.h>

 #define FASTBOOT_INTERFACE_CLASS       0xff
 #define FASTBOOT_INTERFACE_SUB_CLASS   0x42
@@ -197,6 +199,8 @@ static struct usb_gadget_strings *fastboot_strings[] = {
        NULL,
 };

+static struct stdio_dev g_fastboot_stdio;
+
 static void rx_handler_command(struct usb_ep *ep, struct usb_request *req);

 static void fastboot_fifo_complete(struct usb_ep *ep, struct usb_request *req)
@@ -287,6 +291,7 @@ static int fastboot_bind(struct usb_configuration *c, struct usb_function *f)
        if (s)
                g_dnl_set_serialnumber((char *)s);

+       stdio_register(&g_fastboot_stdio);
        return 0;
 }

@@ -295,6 +300,7 @@ static void fastboot_unbind(struct usb_configuration *c, struct usb_function *f)
        f->os_desc_table = NULL;
        list_del(&fb_os_desc.ext_prop);
        memset(fastboot_func, 0, sizeof(*fastboot_func));
+       stdio_deregister("fastboot", 1);
 }

 static void fastboot_disable(struct usb_function *f)
@@ -629,3 +635,37 @@ static void rx_handler_command(struct usb_ep *ep, struct usb_request *req)
        req->actual = 0;
        usb_ep_queue(ep, req, 0);
 }
+
+void fastboot_putc(struct stdio_dev *dev, const char c)
+{
+       char buff[6] = "INFO";
+       buff[4] = c;
+       buff[5] = 0;
+       fastboot_tx_write_more(buff);
+}
+
+#define FASTBOOT_MAX_LEN 64
+
+void fastboot_puts(struct stdio_dev *dev, const char *s)
+{
+       char buff[FASTBOOT_MAX_LEN + 1] = "INFO";
+       int len = strlen(s);
+       int i, left;
+
+       for (i = 0; i < len; i += FASTBOOT_MAX_LEN - 4) {
+               left = len - i;
+               if (left > FASTBOOT_MAX_LEN - 4)
+                       left = FASTBOOT_MAX_LEN - 4;
+
+               memcpy(buff + 4, s + i, left);
+               buff[left + 4 + 1] = 0;
+               fastboot_tx_write_more(buff);
+       }
+}
+
+static struct stdio_dev g_fastboot_stdio = {
+       .name = "fastboot",
+       .flags = DEV_FLAGS_OUTPUT,
+       .putc = fastboot_putc,
+       .puts = fastboot_puts,
+};
--
2.24.0.rc1
``

rschrader, 2019-12-16:
Thanks for the patch! This will help a lot. I will try it later this week.

rubenmoral, 2019-12-16:
It works like a charm, we really appreciate your help. Thank you very much!

closed 2020-01-16 by nxpfrankli