nxp-imx/mfgtools #134: LIBUSB_ERROR_TIMEOUT with big rootfs.tar.bz2

jessie80, 2019-10-02:
Hi customer met LIBUSB_ERR0R_TIMEOUT during downloading the roofs.tar.bz2 which is compiled with bitbake fsl-image-validation-imx. but smaller rootfs.tar.bz is working well. 

below is the uuu log..  
C:\uuu_cg>uuu -v -V lgecg_emmc_download.uuu
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.135-0-gacaf035

Build in config:
        Pctl     Chip            Vid     Pid     BcdVersion
        ==================================================
        SDPS:    MX8QXP          0x1fc9  0x012f  [0x0002..0xffff]
        SDPS:    MX8QM           0x1fc9  0x0129  [0x0002..0xffff]
        SDPS:    MX28            0x15a2  0x004f
        SDPS:    MX815           0x1fc9  0x013e
        SDP:     MX7D            0x15a2  0x0076
        SDP:     MX6Q            0x15a2  0x0054
        SDP:     MX6D            0x15a2  0x0061
        SDP:     MX6SL           0x15a2  0x0063
        SDP:     MX6SX           0x15a2  0x0071
        SDP:     MX6UL           0x15a2  0x007d
        SDP:     MX6ULL          0x15a2  0x0080
        SDP:     MX6SLL          0x1fc9  0x0128
        SDP:     MX7ULP          0x1fc9  0x0126
        SDP:     MXRT106X        0x1fc9  0x0135
        SDP:     MX8MM           0x1fc9  0x0134
        SDP:     MX8MQ           0x1fc9  0x012b
        SDPU:    SPL             0x0525  0xb4a4  [0x0000..0x04ff]
        SDPV:    SPL1            0x0525  0xb4a4  [0x0500..0x9998]
        SDPU:    SPL             0x0525  0xb4a4  [0x9999..0x9999]
        FBK:                     0x066f  0x9afe
        FBK:                     0x066f  0x9bff
        FB:                      0x0525  0xa4a5
        FB:                      0x18d1  0x0d02
Wait for Known USB Device Appear...
New USB Device Attached at 1:13
1:13>Start Cmd:SDP: boot -f u-boot.imx
6400%libusb: error [windows_transfer_callback] detected I/O error 1167: [1167] 장치가 연결되지 않았습니다.
libusb: error [hid_submit_bulk_transfer] HID transfer failed: [1167] 장치가 연결되지 않았습니다.
1:13>Okay
New USB Device Attached at 1:13
1:13>Start Cmd:FB: ucmd setenv fastboot_dev mmc
1:13>Okay
1:13>Start Cmd:FB: ucmd setenv mmcdev ${emmc_dev}
1:13>Okay
1:13>Start Cmd:FB: flash bootloader u-boot.imx
0x400000000x2001:13>Okay
1:13>Start Cmd:FB: ucmd if env exists emmc_ack; then ; else setenv emmc_ack 0; fi;
1:13>Okay
1:13>Start Cmd:FB: ucmd setenv emmc_cmd mmc partconf ${emmc_dev} ${emmc_ack} 1 0
1:13>Okay
1:13>Start Cmd:FB: ucmd if test "${emmc_skip_fb}" != "yes"; then run emmc_cmd; fi
1:13>Okay
1:13>Start Cmd:FB: ucmd setenv emmc_cmd mmc bootbus ${emmc_dev} 2 2 1;
1:13>Okay
1:13>Start Cmd:FB: ucmd if test "${emmc_skip_fb}" != "yes"; then run emmc_cmd; fi
1:13>Okay
1:13>Start Cmd:FB: ucmd echo '[warkap] step #0 before download -f zImage'
1:13>Okay
1:13>Start Cmd:FB: ucmd setenv fastboot_buffer ${loadaddr}
1:13>Okay
1:13>Start Cmd:FB: download -f zImage
1:13>Okay
1:13>Start Cmd:FB: ucmd echo '[warkap] step #1 after download -f zImage'
1:13>Okay
1:13>Start Cmd:FB: ucmd setenv fastboot_buffer ${fdt_addr}
1:13>Okay
1:13>Start Cmd:FB: download -f imx6sx-sdb-lgecg.dtb
1:13>Okay
1:13>Start Cmd:FB: ucmd setenv fastboot_buffer ${initrd_addr}
1:13>Okay
1:13>Start Cmd:FB: download -f _initramfs.cpio.gz.uboot
1:13>Okay
1:13>Start Cmd:FB: acmd ${kboot} ${loadaddr} ${initrd_addr} ${fdt_addr}
1:13>Okay
New USB Device Attached at 1:13
1:13>Start Cmd:FBK: ucmd while [ ! -e /dev/mmcblk*boot0 ]; do sleep 1; echo "wait for /dev/mmcblk*boot* appear"; done;
1:13>Okay
1:13>Start Cmd:FBK: ucmd dev=`ls /dev/mmcblk*boot*`; dev=($dev); dev=${dev[0]}; dev=${dev#/dev/mmcblk}; dev=${dev%boot*}; echo $dev > /tmp/mmcdev;
1:13>Okay
1:13>Start Cmd:FBK: ucmd mmc=`cat /tmp/mmcdev`; PARTSTR=$'10M,500M,0c\n600M,,83\n'; echo "$PARTSTR" | sfdisk --force /dev/mmcblk${mmc}
Checking that no-one is using this disk right now ... OK

Disk /dev/mmcblk3: 3.7 GiB, 3959422976 bytes, 7733248 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x67e6b631

Old situation:

Device         Boot   Start     End Sectors  Size Id Type
/dev/mmcblk3p1        20480 1044479 1024000  500M  c W95 FAT32 (LBA)
/dev/mmcblk3p2      1228800 7733247 6504448  3.1G 83 Linux

>>> Created a new DOS disklabel with disk identifier 0xe090e6d2.
/dev/mmcblk3p1: Created a new partition 1 of type 'W95 FAT32 (LBA)' and of size 500 MiB.
/dev/mmcblk3p2: Created a new partition 2 of type 'Linux' and of size 3.1 GiB.
/dev/mmcblk3p3: Done.

New situation:
Disklabel type: dos
Disk identifier: 0xe090e6d2

Device         Boot   Start     End Sectors  Size Id Type
/dev/mmcblk3p1        20480 1044479 1024000  500M  c W95 FAT32 (LBA)
/dev/mmcblk3p2      1228800 7733247 6504448  3.1G 83 Linux

The partition table has been altered.
Calling ioctl() to re-read partition table.
Syncing disks.
1:13>Okay
1:13>Start Cmd:FBK: ucmd mmc=`cat /tmp/mmcdev`; dd if=/dev/zero of=/dev/mmcblk${mmc} bs=1k seek=4096 count=1
1:13>Okay
1:13>Start Cmd:FBK: ucmd sync
1:13>Okay
1:13>Start Cmd:FBK: ucmd mmc=`cat /tmp/mmcdev`; while [ ! -e /dev/mmcblk${mmc}p1 ]; do sleep 1; done
1:13>Okay
1:13>Start Cmd:FBK: ucmd mmc=`cat /tmp/mmcdev`; mkfs.vfat /dev/mmcblk${mmc}p1
mkfs.fat 4.1 (2017-01-24)
1:13>Okay
1:13>Start Cmd:FBK: ucmd mmc=`cat /tmp/mmcdev`; mkdir -p /mnt/fat
1:13>Okay
1:13>Start Cmd:FBK: ucmd mmc=`cat /tmp/mmcdev`; mount -t vfat /dev/mmcblk${mmc}p1 /mnt/fat
1:13>Okay
1:13>Start Cmd:FBK: ucp  zImage t:/mnt/fat
99%1:13>Okay
1:13>Start Cmd:FBK: ucp  imx6sx-sdb-lgecg.dtb t:/mnt/fat
152%1:13>Okay
1:13>Start Cmd:FBK: ucmd umount /mnt/fat
1:13>Okay
1:13>Start Cmd:FBK: ucmd mmc=`cat /tmp/mmcdev`; mkfs.ext3 -F -E nodiscard /dev/mmcblk${mmc}p2
Creating filesystem with 813056 4k blocks and 203600 inodes
Filesystem UUID: ea45f09c-b6ed-464c-96e0-7286d9e0c1ba
Superblock backups stored on blocks:
        32768, 98304, 163840, 229376, 294912

Allocating group tables: done
Writing inode tables: done
Creating journal (16384 blocks): done
Writing superblocks and filesystem accounting information: done

1:13>Okay
1:13>Start Cmd:FBK: ucmd mkdir -p /mnt/ext3
1:13>Okay
1:13>Start Cmd:FBK: ucmd mmc=`cat /tmp/mmcdev`; mount /dev/mmcblk${mmc}p2 /mnt/ext3
1:13>Okay
1:13>Start Cmd:FBK: acmd export EXTRACT_UNSAFE_SYMLINKS=1; tar -jx -C /mnt/ext3
1:13>Okay
1:13>Start Cmd:FBK: ucp  imx6sxsabresd.rootfs.tar.bz2 t:-
48%1:13>Fail Bulk(W):LIBUSB_ERROR_TIMEOUT

Could you please check and help me to find the root cause?

nxpfrankli, 2019-10-02:
I may cause slow sd\emmc write speed. 

Can you 

https://github.com/NXPmicro/imx-uuc/blob/master/ufb.c

line 532:
		ret = write_file(g_open_file, p, size);
		if (ret < 0)
			key = FAIL;

you can print out how much time consumed by this function

nxpfrankli, 2019-10-02:
Do you have uart log?

jessie80, 2019-10-04:
the uart log is as below
U-Boot 2018.03-00044-ga3070e6-dirty (Aug 30 2019 - 11:22:41 +0900)

CPU:   Freescale i.MX6SX rev1.2 996 MHz (running at 792 MHz)
CPU:   Extended Commercial temperature grade (-20C to 105C) at 48C
Reset cause: POR
Model: Freescale i.MX6 SoloX SDB RevB Board
Board: MX6SX SABRE SDB rev@
DRAM:  256 MiB
PMIC: PFUZE100! DEV_ID=0x11 REV_ID=0x21
MMC:   FSL_SDHC: 0, FSL_SDHC: 1
Loading Environment from MMC... MMC Device 2 not found
*** Warning - No MMC card found, using default environment

Failed (-5)
Display: Hannstar-XGA (1024x768)
Video: 1024x768x18
In:    serial
Out:   serial
Err:   serial
MMC Device 2 not found
no mmc device at slot 2
flash target is MMC:2
MMC Device 2 not found
invalid mmc device 2
Net:
Error: ethernet@02188000 address not set.
No ethernet found.
Fastboot: Normal
Boot from USB for mfgtools
Use default environment for                              mfgtools
Run bootcmd_mfg: run mfgtool_args;if iminfo ${initrd_addr}; then if test ${tee} = yes; then bootm ${tee_addr} ${initrd_addr} ${fdt_addr}; else bootz ${loadaddr} ${initrd_addr} ${fdt_addr}; fi; else echo "Run fastboot ..."; fastboot 0; fi;
Hit any key to stop autoboot:  0

## Checking Image at 86800000 ...
Unknown image format!
Run fastboot ...
flash target is MMC:1
Starting download of 629760 bytes
....
downloading of 629760 bytes finished
writing to partition 'bootloader'
support sparse flash partition for bootloader
Initializing 'bootloader'
switch to partitions #1, OK
mmc1(part 1) is current device
Writing 'bootloader'

MMC write: dev # 1, block # 2, count 1230 ... 1230 blocks written: OK
Writing 'bootloader' DONE!
[warkap] step #0 before download -f zImage
Starting download of 6870616 bytes
....................................................
downloading of 6870616 bytes finished
[warkap] step #1 after download -f zImage
Starting download of 42895 bytes

downloading of 42895 bytes finished
Starting download of 12348071 bytes
..........................................................................
....................
downloading of 12348071 bytes finished
Kernel image @ 0x80800000 [ 0x000000 - 0x68d658 ]
## Loading init Ramdisk from Legacy Image at 86800000 ...
   Image Name:   fsl-image-mfgtool-initramfs-imx6
   Image Type:   ARM Linux RAMDisk Image (uncompressed)
   Data Size:    12348007 Bytes = 11.8 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 83000000
   Booting using the fdt blob at 0x83000000
   Using Device Tree in place at 83000000, end 8300d78e
switch to ldo_bypass mode!

Starting kernel ...

vddarm: bypassed regulator has no supply!
vddarm: failed to get the current voltage(-517)
vddsoc: bypassed regulator has no supply!
vddsoc: failed to get the current voltage(-517)
imx6sx-pinctrl 20e0000.iomuxc: Invalid fsl,pins or pins property in node /soc/aips-bus@02000000/iomuxc@020e0000/imx6x-sdb/wdoggrp
gpio-9 (cts): _gpiod_direction_output_raw: tried to set a GPIO tied to an IRQ as output
gpio-12 (cts): _gpiod_direction_output_raw: tried to set a GPIO tied to an IRQ as output
fec 2188000.ethernet (unnamed net_device) (uninitialized): Invalid MAC address: 00:00:00:00:00:00
caam-snvs 20cc000.caam-snvs: can't get snvs clock
Found New UDC: ci_hdrc.0
ci_hdrc.0 0
ffs.utp0
run utp at /dev/usb-utp0/ep0
.
uuu fastboot client 1.0.0 [built Apr 14 2019 13:43:48]
Start init usb
write string
Start handle command
uuc /dev/utp
uuc 0.5 [built Apr 14 2019 13:43:48]
UTP: Waiting for /dev/utp to appear
run shell cmd: while [ ! -e /dev/mmcblk*boot0 ]; do sleep 1; echo "wait for /dev/mmcblk*boot* appear"; done;
run shell cmd: dev=`ls /dev/mmcblk*boot*`; dev=($dev); dev=${dev[0]}; dev=${dev#/dev/mmcblk}; dev=${dev%boot*}; echo $dev > /tmp/mmcdev;
run shell cmd: mmc=`cat /tmp/mmcdev`; PARTSTR=$'10M,500M,0c\n600M,,83\n'; echo "$PARTSTR" | sfdisk --force /dev/mmcblk${mmc}
Partition #1 contains a vfat signature.
Partition #2 contains a ext3 signature.
run shell cmd: mmc=`cat /tmp/mmcdev`; dd if=/dev/zero of=/dev/mmcblk${mmc} bs=1k seek=4096 count=1
1+0 records in
1+0 records out
1024 bytes (1.0 kB, 1.0 KiB) copied, 0.00171967 s, 595 kB/s
run shell cmd: sync
run shell cmd: mmc=`cat /tmp/mmcdev`; while [ ! -e /dev/mmcblk${mmc}p1 ]; do sleep 1; done
run shell cmd: mmc=`cat /tmp/mmcdev`; mkfs.vfat /dev/mmcblk${mmc}p1
run shell cmd: mmc=`cat /tmp/mmcdev`; mkdir -p /mnt/fat
run shell cmd: mmc=`cat /tmp/mmcdev`; mount -t vfat /dev/mmcblk${mmc}p1 /mnt/fat
WOpen:/mnt/fat
WOpen:/mnt/fat/zImage
WOpen:/mnt/fat
WOpen:/mnt/fat/imx6sx-sdb-lgecg.dtb
run shell cmd: umount /mnt/fat
run shell cmd: mmc=`cat /tmp/mmcdev`; mkfs.ext3 -F -E nodiscard /dev/mmcblk${mmc}p2
mke2fs 1.43.8 (1-Jan-2018)
run shell cmd: mkdir -p /mnt/ext3
run shell cmd: mmc=`cat /tmp/mmcdev`; mount /dev/mmcblk${mmc}p2 /mnt/ext3
run shell cmd: export EXTRACT_UNSAFE_SYMLINKS=1; tar -jx -C /mnt/ext3
WOpen:-

guwhite, 2019-10-16:
I'm having the same issue and with a similar serial log.  Here are a few levers I've tried (Remove # to invoke).

#Levers for available free memory space 
#FBK: ucmd echo 'vm.min_free_kbytes = 65536' > /etc/sysctl.d/01_minfreekbytes.conf
#FBK: ucmd echo 'vm.swappiness = 80' > /etc/sysctl.d/swappiness.conf
#FBK: ucmd sysctl -p /etc/sysctl.d/01_minfreekbytes.conf
#FBK: ucmd sysctl -p /etc/sysctl.d/swappiness.conf

#Levers for CPU clocking
#FBK: ucmd echo powersave > /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor

#Levers for CPU enable
#FBK: ucmd echo "0" >/sys/devices/system/cpu/cpu1/online
#FBK: ucmd echo "0" >/sys/devices/system/cpu/cpu2/online
#FBK: ucmd echo "0" >/sys/devices/system/cpu/cpu3/online

#Levers for disk caching
#FBK: ucmd echo 100 > /proc/sys/vm/dirty_expire_centisecs
#FBK: ucmd echo 100 > /proc/sys/vm/dirty_writeback_centisecs

The only thing in the above that seems to affect the results is to limit the CPU to only cpu0 (it takes a long time but it passes).  Turning on the other CPUs will eventually fail but at different places.  In my scenario, enabling CPU0 and CPU3 will result in the script failing at the "unmount" command for some reason.  Looking for suggestions.

nxpfrankli, 2019-10-16:
Does https://github.com/NXPmicro/mfgtools/pull/137 can work around this problem?

It's strange problem. Only 40MB used by ufb. I am not sure why usb can't allocate 64k memory. Our USB driver owner is checking it.

guwhite, 2019-10-16:
#137 does not help in my scenario.  Our scripts are different from above by first copying the bz2 to a temporary directory and untaring from there to disk instead of directly host copy / untar to disk.

jessie80, 2019-10-17:
@nxpfrankli 
i'm waiting the feedback from customer. as i know, he is on the business trip. he may be back in next Monday. I'll update soon.

jessie80, 2019-10-28:
@nxpfrankli 
According to the customer , with your temporal binary, TIMEOUT issue is resolved. 
Is there any progress from USBdriver owner?

nxpfrankli, 2019-10-28:
not yet. Not sure why there are not continue 64K memory at that time

nxpfrankli, 2020-04-17:
5.4 kernel's usb support scatter list.  Q2 release will include this change.
this problem can be fixed at 5.4 kernel.

closed 2020-04-17 by nxpfrankli

franklin-woo, 2020-06-25:
I got a same issue.
My rootfs file size 674mb.

I built the yocto(zeus) and kernel version is 5.4.3.
Below log is some part of uuu verbose log.

nxpfrankli, 2020-06-25:
I check your kernel git commit hash value, which not include,  Latest kernel will publish at Jun 30. Please update it and try again. 

commit ff01b0f10d5f5ada52140561b21387092dadf446
Author: Peter Chen <peter.chen@nxp.com>
Date:   Fri Feb 21 21:40:57 2020 +0800

    MLK-23349-2 usb: chipidea: udc: add sg list support

    For low system memory system (eg, 265M), it may met OOM issue if the
    request buffer is large (eg, 64KB). We ran out below OOM issue for f_fs.
    Luckily, the f_fs supports sg list now, the OOM issue is fixed with
    this patch.

    ufb: page allocation failure: order:4, mode:0x40cc0(GFP_KERNEL|__GFP_COMP),
    nodemask=(null),cpuset=/,mems_allowed=0
    CPU: 2 PID: 370 Comm: ufb Not tainted 5.4.3-1.1.0+g54b3750d61fd #1
    Hardware name: NXP i.MX8MNano DDR4 EVK board (DT)

franklin-woo, 2020-06-26:
> udc: add sg list support

Thank you for reply.

I found the patch below link.

https://www.spinics.net/lists/linux-usb/msg191815.html
https://www.spinics.net/lists/linux-usb/msg194332.html

I applied the patch in my kernel, But it's not the work. I get the same error as before.
Can you give me the official patch link?

> I check your kernel git commit hash value, which not include, Latest kernel will publish at Jun 30. Please update it and try again.
> 
> commit ff01b0f10d5f5ada52140561b21387092dadf446
> Author: Peter Chen [peter.chen@nxp.com](mailto:peter.chen@nxp.com)
> Date: Fri Feb 21 21:40:57 2020 +0800
> 
> ```
> MLK-23349-2 usb: chipidea: udc: add sg list support
> 
> For low system memory system (eg, 265M), it may met OOM issue if the
> request buffer is large (eg, 64KB). We ran out below OOM issue for f_fs.
> Luckily, the f_fs supports sg list now, the OOM issue is fixed with
> this patch.
> 
> ufb: page allocation failure: order:4, mode:0x40cc0(GFP_KERNEL|__GFP_COMP),
> nodemask=(null),cpuset=/,mems_allowed=0
> CPU: 2 PID: 370 Comm: ufb Not tainted 5.4.3-1.1.0+g54b3750d61fd #1
> Hardware name: NXP i.MX8MNano DDR4 EVK board (DT)
> ```

> I check your kernel git commit hash value, which not include, Latest kernel will publish at Jun 30. Please update it and try again.
> 
> commit ff01b0f10d5f5ada52140561b21387092dadf446
> Author: Peter Chen [peter.chen@nxp.com](mailto:peter.chen@nxp.com)
> Date: Fri Feb 21 21:40:57 2020 +0800
> 
> ```
> MLK-23349-2 usb: chipidea: udc: add sg list support
> 
> For low system memory system (eg, 265M), it may met OOM issue if the
> request buffer is large (eg, 64KB). We ran out below OOM issue for f_fs.
> Luckily, the f_fs supports sg list now, the OOM issue is fixed with
> this patch.
> 
> ufb: page allocation failure: order:4, mode:0x40cc0(GFP_KERNEL|__GFP_COMP),
> nodemask=(null),cpuset=/,mems_allowed=0
> CPU: 2 PID: 370 Comm: ufb Not tainted 5.4.3-1.1.0+g54b3750d61fd #1
> Hardware name: NXP i.MX8MNano DDR4 EVK board (DT)
> ```

> I check your kernel git commit hash value, which not include, Latest kernel will publish at Jun 30. Please update it and try again.
> 
> commit ff01b0f10d5f5ada52140561b21387092dadf446
> Author: Peter Chen [peter.chen@nxp.com](mailto:peter.chen@nxp.com)
> Date: Fri Feb 21 21:40:57 2020 +0800
> 
> ```
> MLK-23349-2 usb: chipidea: udc: add sg list support
> 
> For low system memory system (eg, 265M), it may met OOM issue if the
> request buffer is large (eg, 64KB). We ran out below OOM issue for f_fs.
> Luckily, the f_fs supports sg list now, the OOM issue is fixed with
> this patch.
> 
> ufb: page allocation failure: order:4, mode:0x40cc0(GFP_KERNEL|__GFP_COMP),
> nodemask=(null),cpuset=/,mems_allowed=0
> CPU: 2 PID: 370 Comm: ufb Not tainted 5.4.3-1.1.0+g54b3750d61fd #1
> Hardware name: NXP i.MX8MNano DDR4 EVK board (DT)
> ```

Thank you for reply.

I found the patch below link.

https://www.spinics.net/lists/linux-usb/msg191815.html
https://www.spinics.net/lists/linux-usb/msg194332.html

I applied the patch in my kernel, But it's not the work. I get the same error as before. :(
Can you give me the official patch link?

scooty334, 2020-06-28:
> > udc: add sg list support
> 
> > I check your kernel git commit hash value, which not include, Latest kernel will publish at Jun 30. Please update it and try again.
> > commit ff01b0f10d5f5ada52140561b21387092dadf446
> > Author: Peter Chen [peter.chen@nxp.com](mailto:peter.chen@nxp.com)
> > Date: Fri Feb 21 21:40:57 2020 +0800
> > ```
> > MLK-23349-2 usb: chipidea: udc: add sg list support
> > 
> > For low system memory system (eg, 265M), it may met OOM issue if the
> > request buffer is large (eg, 64KB). We ran out below OOM issue for f_fs.
> > Luckily, the f_fs supports sg list now, the OOM issue is fixed with
> > this patch.
> > 
> > ufb: page allocation failure: order:4, mode:0x40cc0(GFP_KERNEL|__GFP_COMP),
> > nodemask=(null),cpuset=/,mems_allowed=0
> > CPU: 2 PID: 370 Comm: ufb Not tainted 5.4.3-1.1.0+g54b3750d61fd #1
> > Hardware name: NXP i.MX8MNano DDR4 EVK board (DT)
> > ```
> 
> Thank you for reply.
> 
> I found the patch below link.
> 
> https://www.spinics.net/lists/linux-usb/msg191815.html
> https://www.spinics.net/lists/linux-usb/msg194332.html
> 
> I applied the patch in my kernel, But it's not the work. I get the same error as before. :(
> Can you give me the official patch link?

Hi, have you solved your issue ?  im experiencing the same one under windows/ubuntu 
my rootfs file size only 100MB

===============================================================
C:\Users\---\Downloads\windows>uuu ./example_kernel_nand.uuu
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.3.191-0-g4fe24b9

Success 0    Failure 1

1:10    21/24 [Bulk(R):LIBUSB_ERROR_TIMEOUT          ] FBK: ucp rootfs.tar t:-
==============================================================

and using this script ending:

# burn rootfs
FBK: ucmd source /tmp/mtd.sh; flash_erase /dev/mtd${nandrootfs} 0 0
FBK: ucmd source /tmp/mtd.sh; ubiattach /dev/ubi_ctrl -m ${nandrootfs}
FBK: ucmd source /tmp/mtd.sh; ubimkvol /dev/ubi0 -Nnandrootfs -m
FBK: ucmd source /tmp/mtd.sh; mkdir -p /mnt/mtd
FBK: ucmd source /tmp/mtd.sh; mount -t ubifs ubi0:nandrootfs /mnt/mtd

FBK: acmd export EXTRACT_UNSAFE_SYMLINKS=1; tar -xv -C /mnt/mtd
FBK: ucp rootfs.tar t:-
FBK: sync
FBK: ucmd umount /mnt/mtd

FBK: done

franklin-woo, 2020-07-01:
I tested today with the kernel 5.7.7 and 5.8-rc3. 
But still not the work. It occurs same error. :(

nxpfrankli, 2020-07-01:
I am not sure peter's patch upstreamed.

MLK-23349-2 usb: chipidea: udc: add sg list support
Can you try our release https://source.codeaurora.org/external/imx/linux-imx/commit/?h=rel_imx_5.4.24_2.1.0&id=babac008e5cf168abca1a85bda2e8071ca27a5c0

franklin-woo, 2020-07-02:
imx_5.4.24_2.1.0 branch is the same. doesn't work.
My board is only sent below 128mb. If it exceeds 128mb, the transfer will fail.

---------------------
1:1>Okay (1.568s)
1:1>Start Cmd:FBK: ucmd mmc=`cat /tmp/mmcdev`; dd if=/dev/zero of=/dev/mmcblk${mmc} bs=1k seek=4096 count=1
1:1>Okay (0.049s)
1:1>Start Cmd:FBK: ucmd sync
1:1>Okay (0.025s)
1:1>Start Cmd:FBK: ucmd mmc=`cat /tmp/mmcdev`; echo 0 > /sys/block/mmcblk${mmc}boot0/force_ro
1:1>Okay (0.035s)
1:1>Start Cmd:FBK: ucp firmware/u-boot-dtb.imx t:/tmp
99%1:1>Okay (0.146s)
1:1>Start Cmd:FBK: ucmd mmc=`cat /tmp/mmcdev`; dd if=/tmp/u-boot-dtb.imx of=/dev/mmc${mmc}boot0 bs=1K seek=32
1:1>Okay (0.089s)
1:1>Start Cmd:FBK: ucmd mmc=`cat /tmp/mmcdev`; echo 1 > /sys/block/mmcblk${mmc}boot0/force_ro
1:1>Okay (0.033s)
1:1>Start Cmd:FBK: ucmd mmc=`cat /tmp/mmcdev`; while [ ! -e /dev/mmcblk${mmc}p1 ]; do sleep 1; done
1:1>Okay (0.032s)
1:1>Start Cmd:FBK: ucmd mmc=`cat /tmp/mmcdev`; mkfs.vfat /dev/mmcblk${mmc}p1
mkfs.fat 4.1 (2017-01-24)
1:1>Okay (0.11s)
1:1>Start Cmd:FBK: ucmd mmc=`cat /tmp/mmcdev`; mkdir -p /mnt/fat
1:1>Okay (0.042s)
1:1>Start Cmd:FBK: ucmd mmc=`cat /tmp/mmcdev`; mount -t vfat /dev/mmcblk${mmc}p1 /mnt/fat
1:1>Okay (0.104s)
1:1>Start Cmd:FBK: ucp firmware/zImage t:/mnt/fat
99%1:1>Okay (2.58s)
1:1>Start Cmd:FBK: ucp firmware/imx6dl-sabresd.dtb t:/mnt/fat
102%1:1>Okay (0.014s)
1:1>Start Cmd:FBK: ucmd umount /mnt/fat
1:1>Okay (0.526s)
1:1>Start Cmd:FBK: ucmd mmc=`cat /tmp/mmcdev`; mkfs.ext3 -F -E nodiscard /dev/mmcblk${mmc}p2
Creating filesystem with 262144 4k blocks and 65536 inodes
Filesystem UUID: be6d3d24-4a46-41b3-9eef-b5f71a954457
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (8192 blocks): done
Writing superblocks and filesystem accounting information: done

1:1>Okay (12.51s)
1:1>Start Cmd:FBK: ucmd mkdir -p /mnt/rootfs
1:1>Okay (0.025s)
1:1>Start Cmd:FBK: ucmd mmc=`cat /tmp/mmcdev`; mount -t ext3 /dev/mmcblk${mmc}p2 /mnt/rootfs
1:1>Okay (0.118s)
1:1>Start Cmd:FBK: ucp firmware/fsl-image-qt5-rootfs.tar.bz2 t:/mnt/rootfs
**19%1:1>Fail Bulk(R):LIBUSB_ERROR_TIMEOUT(38.39s)**

nxpfrankli, 2020-07-02:
what's your physical memory size?

franklin-woo, 2020-07-02:
> what's your physical memory size?

Memory size is 512MB.

nxpfrankli, 2020-07-02:
Any output at board uart console?

franklin-woo, 2020-07-02:
> Any output at board uart console?

Refer to below log.
------------------------
U-Boot 2019.04-lf-5.4.y_v2019.04+g228843cdf5 (Jun 24 2020 - 19:00:51 +0900)

CPU:   Freescale i.MX6SOLO rev1.3 996 MHz (running at 792 MHz)
CPU:   Extended Commercial temperature grade (-20C to 105C) at 44C
Reset cause: POR
Model: Freescale i.MX6 DualLite SABRE Smart Device Board
Board: MX6-SabreSD
I2C:   ready
DRAM:  512 MiB

MMC:   FSL_SDHC: 1, FSL_SDHC: 2, FSL_SDHC: 3
Loading Environment from MMC... MMC: no card present
*** Warning - No block device, using default environment

No panel detected: default to HDMI
Display: HDMI (1024x768)
In:    serial
Out:   serial
Err:   serial
MMC: no card present
flash target is MMC:2
MMC: no card present
MMC card init failed!
MMC: no card present
** Block device MMC 2 not supported
Net:   Board Net Initialization Failed
No ethernet found.
Fastboot: Normal
Boot from USB for mfgtools
*** Warning - Use default environment for                                mfgtools
, using default environment

Run bootcmd_mfg: run mfgtool_args;if iminfo ${initrd_addr}; then if test ${tee} = yes; then bootm ${tee_addr} ${initrd_addr} ${fdt_addr}; el
se bootz ${loadaddr} ${initrd_addr} ${fdt_addr}; fi; else echo "Run fastboot ..."; fastboot 0; fi;
Hit any key to stop autoboot:  0 

## Checking Image at 12c00000 ...
Unknown image format!
Run fastboot ...
Starting download of 8559608 bytes
.................................................................
downloading of 8559608 bytes finished
Starting download of 48162 bytes

downloading of 48162 bytes finished
Starting download of 12475878 bytes
..........................................................................
.....................
downloading of 12475878 bytes finished
Kernel image @ 0x12000000 [ 0x000000 - 0x829bf8 ]
## Loading init Ramdisk from Legacy Image at 12c00000 ...
   Image Name:   fsl-image-mfgtool-initramfs-imx6
   Image Type:   ARM Linux RAMDisk Image (uncompressed)
   Data Size:    12475814 Bytes = 11.9 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 18000000
   Booting using the fdt blob at 0x18000000
   Using Device Tree in place at 18000000, end 1800ec21
No PMIC found!

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.24-00060-gbabac008e5cf (frank@frank-desktop) (gcc version 9.2.0 (GCC)) #1 SMP Thu Jul 2 09:29:09 KST 2020
[... 293 lines cut ...]
[   40.198584] pc : [<c01993c8>]    lr : [<c04e1620>]    psr: a0070193
[   40.204859] sp : c1201eb4  ip : 000003ff  fp : 5924d24b
[   40.210092] r10: c1201ed8  r9 : c1264ff4  r8 : 00000000
[   40.215326] r7 : f4a00100  r6 : c1209130  r5 : 000003ff  r4 : f4a0010c
[   40.221861] r3 : c1201ed8  r2 : 00000001  r1 : 00000057  r0 : c6018400
[   40.228399] Flags: NzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment none
[   40.235628] Control: 10c5387d  Table: 16984059  DAC: 00000051
[   40.241384] Process swapper/0 (pid: 0, stack limit = 0x6abaffa9)
[   40.247399] Stack: (0xc1201eb4 to 0xc1202000)
[   40.251768] 1ea0:                                              c04e1620 c07a752c 20070013
[   40.259958] 1ec0: ffffffff c1201f0c c1332ec8 c1200000 00000009 c0101a70 00000001 00000006
[   40.268147] 1ee0: 00000000 c120c580 00000001 589aa498 c120f1c4 cbb16db8 c1332ec8 00000009
[   40.276337] 1f00: 00000009 5924d24b 00000000 c1201f28 c018c3d0 c07a752c 20070013 ffffffff
[   40.284526] 1f20: 00000051 00000000 c120f1c4 c01a7a54 00000001 c1208b08 c1201f83 c01b44a0
[   40.292716] 1f40: 008a8e09 cbb16db8 c120f1c4 00000001 c1208b64 c1197db0 c1208ed4 c120f1c4
[   40.300906] 1f60: cbb16db8 c07a791c c1200000 00000000 c1208b28 c0167eb4 c1208b00 00000001
[   40.309095] 1f80: 00337bc0 c1208b08 c1173a50 000000ce 00000001 c1208b00 00000000 c1337bc0
[   40.317285] 1fa0: 00000001 c1173a50 00000000 c016833c c1337c08 c1100e20 ffffffff ffffffff
[   40.325474] 1fc0: 00000000 c110057c 00000000 c1173a50 824ede4f c1208b08 c1100330 00000051
[   40.333663] 1fe0: 10c0387d 00000f8c 18000000 412fc09a 10c5387d 00000000 00000000 00000000
[   40.341861] [<c01993c8>] (__handle_domain_irq) from [<00000009>] (0x9)
[   40.348403] Code: ea272554 c1216de8 e72de10c e1a0c005 (e7cc700e) 
[   40.354521] ---[ end trace e984bb4bc0ae4bbd ]---
[   40.359151] Kernel panic - not syncing: Attempted to kill the idle task!
[   40.365876] ---[ end Kernel panic - not syncing: Attempted to kill the idle task! ]---

nxpfrankli, 2020-07-02:
It looks like the other problem.  What's your hardware platform? 

[ 40.153402] Unable to handle kernel paging request at virtual address c04e1a1f
[ 40.160644] pgd = e5a40413
[ 40.163368] [c04e1a1f] *pgd=1040840e(bad)
[ 40.167404] Internal error: Oops: 80d [#1] SMP ARM
[ 40.172207] Modules linked in:
[ 40.175280] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.4.24-00060-gbabac008e5cf #1
[ 40.182944] Hardware name: Freescale i.MX6 Quad/DualLite (Device Tree)
[ 40.189494] PC is at __handle_domain_irq+0x8/0xdc
[ 40.194216] LR is at gic_handle_irq+0x48/0x9c

franklin-woo, 2020-07-02:
> It looks like the other problem. What's your hardware platform?
>
> [ 40.153402] Unable to handle kernel paging request at virtual address c04e1a1f
> [ 40.160644] pgd = e5a40413
> [ 40.163368] [c04e1a1f] [...]

Sorry. That log is my test code error log.
Please refer to below log.
And h/w spec is 8G emmc, 512MB ram, i.MX6solo

run shell cmd: mmc=`cat /tmp/mmcdev`; echo 1 > /sys/block/mmcblk${mmc}boot0/force_ro
run shell cmd: mmc=`cat /tmp/mmcdev`; while [ ! -e /dev/mmcblk${mmc}p1 ]; do sleep 1; done
run shell cmd: mmc=`cat /tmp/mmcdev`; mkfs.vfat /dev/mmcblk${mmc}p1
run shell cmd: mmc=`cat /tmp/mmcdev`; mkdir -p /mnt/fat
run shell cmd: mmc=`cat /tmp/mmcdev`; mount -t vfat /dev/mmcblk${mmc}p1 /mnt/fat
WOpen:/mnt/fat
WOpen:/mnt/fat/zImage
[    9.424295] random: crng init done
WOpen:/mnt/fat
WOpen:/mnt/fat/imx6dl-sabresd.dtb
run shell cmd: umount /mnt/fat
run shell cmd: mmc=`cat /tmp/mmcdev`; mkfs.ext4 /dev/mmcblk${mmc}p2
mke2fs 1.45.3 (14-Jul-2019)
run shell cmd: mmc=`cat /tmp/mmcdev`; mkfs.ext4 /dev/mmcblk${mmc}p3
mke2fs 1.45.3 (14-Jul-2019)
run shell cmd: mkdir -p /mnt/rootfs
run shell cmd: mkdir -p /mnt/data
run shell cmd: mmc=`cat /tmp/mmcdev`; mount -t ext4 /dev/mmcblk${mmc}p2 /mnt/rootfs
[   15.256417] EXT4-fs (mmcblk3p2): mounted filesystem with ordered data mode. Opts: (null)
run shell cmd: mmc=`cat /tmp/mmcdev`; mount -t ext4 /dev/mmcblk${mmc}p3 /mnt/data
[   15.339890] EXT4-fs (mmcblk3p3): mounted filesystem with ordered data mode. Opts: (null)
run shell cmd: export EXTRACT_UNSAFE_SYMLINKS=1; tar -jx -C /mnt/rootfs
WOpen:-
[   35.043997] usb_otg_vbus: disabling
[   35.047707] usb_h1_vbus: disabling
[   35.051141] wm8962-supply: disabling
[   35.074458] MPCIE_3V3: disabling
[   35.077733] sensors-supply: disabling
[   35.081430] hdmi-5v-supply: disabling
[   35.108736] mipi_dsi_pwr_on: disabling
[   69.604921] cfg80211: failed to load regulatory.db
**failure write to usb ep**

scooty334, 2020-07-02:
i figure out the solution for this issue. burn the kernel and dtb in the filesystem, will cause ECC error in uboot, so I burn the kernel and dtb in uboot.  this one works successfully every time, im using NAND (check attached config)

[uuu.txt](https://github.com/NXPmicro/mfgtools/files/4862979/uuu.txt)
[image]

franklin-woo, 2020-07-02:
> i figure out the solution for this issue. burn the kernel and dtb in the filesystem, will cause ECC error in uboot, so I burn the kernel and dtb in uboot. this one works successfully every time, im [...]

Your solution doesn't work for me either.
I think, even if you skip downloading the kernel and bootloader, the rootfs will be transferred.

scooty334, 2020-07-02:
maybe add sdcard ?  im assuming you doing prototyping.