nxp-imx/mfgtools #136: ucp copy file in zip fail

115606666, 2019-10-08:
Below okay.
```
2:63>Start Cmd:FBK: ucp  ext4.md5 t:/tmp
204800%2:63>Okay (0.009s)
```

But when we compress all files into one zip file, ucp fail.
```
Prep>Start Cmd:FBK: ucp  ext4.md5 t:/tmp
Decompress file:ext4.md5
Prep>Fail fail open file: >uuu.zip/ext4.md5(1.845e+16s)
```

115606666, 2019-10-09:
The zip file can be opend via win10 File Explorer, and no directory in zip . In fact, the script load u-boot, device tree, kernel, then boot into Linux, and write file system into eMMC via external ext4 image (file name is fsl-image-validation-imx-imx8mmevk.ext4). This ext4 image can be loaded by uuu and write into eMMC successfully. Then we calculate md5 of eMMC content and compare with external pre-calculated md5 (file name is ext4.md5). We use fsl-image-validation-imx-imx8mmevk.ext4.md5 at first, then we change file name to ext4.md5 to make it shorter, but still doesn't work.

nxpfrankli, 2019-10-09:
can you share your zip file?

115606666, 2019-10-10:
send file download link to company mail

commit f1c8b762 (2019-10-10): fix #136: fix decompress md5 file fail in zip

md5 file use uncompress method in zip file
closed 2019-10-10 by nxpfrankli