nxp-imx/mfgtools #138: fatal error: openssl/ssl.h: No such file or directory

jessie80, 2019-10-25:
Hi all.
I met uuu build error on ubuntu .
/mfgtools/libuuu/http.cpp:22:25: fatal error: openssl/ssl.h: No such file or directory

But, openssl is installed in /usr/local/bin. as below
/usr/local/bin/cmake --version
cmake version 3.16.0-rc1

Could you please guide me which point that i need to check more?

BRs
jessie

nxpfrankli, 2019-10-25:
can you try apt-get install libssl-dev?

nxpfrankli, 2019-10-25:
try below patch

diff --git a/libuuu/CMakeLists.txt b/libuuu/CMakeLists.txt
index 64bba71..385b73c 100644
--- a/libuuu/CMakeLists.txt
+++ b/libuuu/CMakeLists.txt
@@ -15,7 +15,7 @@ endif()

 find_package(OpenSSL)

-include_directories(${LIBUSB_INCLUDE_DIRS} include)
+include_directories(${LIBUSB_INCLUDE_DIRS} ${OPENSSL_INCLUDE_DIR} include)

 if(OPENSSL_FOUND)
 set(UUUSSL "-DUUUSSL")

jessie80, 2019-10-28:
Hi Frank.
with your patch for including OPENSSL_INCLUDE_DIR, uuu build succeeded.
Thank you.

closed 2019-10-28 by jessie80