nxp-imx/mfgtools #141: NAND update

AkDmOl, 2019-11-04:
Hi! I'm new in the development under imx boards and have a question. 
I've built NXP images and now would like to port it on my board(iMx6ul in nand mode) using NAND flash  and uuu util, but unfortunately have a problem during the execution. 
`$sudo ./uuu -b nand u-boot.imx core-image-base-imx6ull14x14evk.sdcard.bz2` 

`uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.3.102-3-g64a66c3
Wait for Known USB Device Appear...`
`Error: Unknown Command:FB: nandbcb update ${fastboot_buffer} nandboot ${fastboot_bytes}`

What am I doing wrong and how could I reach my goal?

nxpfrankli, 2019-11-04:
uboot for nand have not released yet. 
-b nand just write uboot. can't support write rootfs yet.

AkDmOl, 2019-11-04:
Okay, thanks

closed 2019-11-04 by AkDmOl