nxp-imx/mfgtools #142: Cannot write to NAND due to missing "FB: nandbcb" command

mikecrowe, 2019-11-06:
I'm attempting to program NAND on an iMX6ULL EVK but the tool doesn't even try because `cmd.cpp` doesn't contain a "FB: nandbcb" command:

```
$ ./uuu -b nand u-boot-imx6ull14x14evk_nand.imx
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.3.102-3-g64a66c3

Wait for Known USB Device Appear...
Error: Unknown Command:FB: nandbcb update ${fastboot_buffer} nandboot ${fastboot_bytes}
```

Is the `nand_burn_loader.lst` file incorrect or is support for that command missing?

Thanks.

Mike.

nxpfrankli, 2019-11-06:
nandbcb supported uboot still not release yet

mikecrowe, 2019-11-21:
Thanks for your reply. I've now upgraded to U-Boot https://source.codeaurora.org/external/imx/uboot-imx/log/?h=imx_v2019.04_4.19.35_1.1.0 which has the nandbcb command. However, the error message above is displayed even before UUU has attempted to communicate with the board so it is independent from the U-Boot version in use. It looks like UUU lacks support for the nandbcb command too.

nxpfrankli, 2019-11-21:
Yes, you are right. 

it should be 
FB: **ucmd** nandbcb init ${fastboot_buffer} nandboot ${fastboot_bytes}

mikecrowe, 2019-11-21:
Thanks. This means that I can UUU in NAND mode now. (It still doesn't work, but for reasons unrelated to this issue.)

closed 2019-11-21 by mikecrowe