nxp-imx/mfgtools #150: uuu hangs on flashing a manually re-packed .sdcard.bz2 image

sposelenov, 2019-12-11:
Hello,
I'm running an uuu script to flash the sdcard image to the board:
```
uuu_version 1.0.1

# Boot the module with the U-Boot image. U-Boot will enter the fastboot mode automatically
SDPS: boot -f imx-boot-imx8mnlego-ddr4-sd.bin
# Configure U-Boot variables
FB: ucmd setenv fastboot_dev mmc
FB: ucmd setenv mmcdev ${emmc_dev}
# Flash the MMC image
FB: flash -raw2sparse all neato-rootfs-debug-imx8mnlego-ddr4.sdcard.bz2/*
FB: Done

```

Everythin is fine, unless I re-pack the .sdcard.bz2 image with bzip2. I chnage nothing in the .sdcard image, just unpack, then pack back.

After that, flashing stops:
```
$ sudo /usr/local/bin/uuu -d uuu.script-debug-imx8mnlego-ddr4 
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.3.82-0-g9c56e46

Success 0    Failure 0                                                                
                                                                                       
                                                                                       
8:1      3/ 4 [                                      ] FB: flash -raw2sparse all neato-

```

No error is returned in reasonable time.
Target serial console shows:
```
Run fastboot ...
Detect USB boot. Will enter fastboot mode!
flash target is MMC:1
Detect USB boot. Will enter fastboot mode!
```

Is this a known issue?
Thanks!

nxpfrankli, 2019-12-11:
https://github.com/NXPmicro/mfgtools/wiki/Release-Notes

Known issue:

Need use pbzip2. 

Actually, not display background uncompress progress. So it wait quite long time to complete uncompress

nxpfrankli, 2019-12-11:
Generally, needn't compress again after uncompressed.  uncompressed sdcard is more efficient than compressed one at memory usage.

closed 2020-04-04 by sposelenov