nxp-imx/mfgtools #152: [Question] How to specify the USB port on Windows UUU?

ajithpv, 2019-12-18:
Hi,

We have to flash 2 devices with different UUU (i.e. contents are different) from the same Host PC (Windows PC). The USB ports should be same (becuase, we are not removing the USB cables from the Windows PC) and only the devices will change for the flashing.

In Linux Host PC (Ubuntu 18.04 LTS), the below commands are working fine and I can able to connect 2 devices with specific port numbers. 
uuu -d -m 1:1 -m 2:1 boardA_uuu.auto           // monitor port 1:1 and 2:1 for boardsA.
uuu -d -m 1:3 -m 4:1 boardB_uuu.auto          // monitor port 1:3 and 4:1 for boardsB.

But, I couldn't find any documentation on how to achieve the same in Windows PC.

Hence, could you please let me know how to flash multiple devices with specific USB ports on Windows PC?

nxpfrankli, 2019-12-18:
yes, it is the same command.  
to get port number
uuu -lsusb

ajithpv, 2019-12-19:
@nxpfrankli , Thank you for your support. I have tried the "uuu -lsusb" command in Windows 10 machine. However, I got the below error:

> unknown option: -lsusb

Is this command specific to any version of UUU or all UUU supports this command?

I'm specifically looking for Windows 10 option to get the USB port number and assign the same for monitoring via UUU.

nxpfrankli, 2019-12-19:
After 1.3.82
yes, it is the same as monitor

ajithpv, 2019-12-20:
Thank you so much @nxpfrankli for your support. I have downloaded the '_1.3.102_' version and it is working on Windows 10 :+1:

closed 2019-12-20 by ajithpv