nxp-imx/mfgtools #153: Unable to boot bootloader on HAB closed i.MX7D rev D processor

scallopedllama, 2019-12-27:
We are working on a custom application using the iMX7D processor and part of our design is to use secure boot. As a result, we will be blowing fuses to program the SRK fuses with our public key hashes and blowing the fuse to put the device into a HAB Closed configuration so that only binaries signed with our private key will be bootable on the processor.

We are able to use uuu with no issues while the processor has not yet had its HAB configuration closed but as soon as we close the configuration, we are unable to boot the processor using any USB serial loader program.

We have followed the instructions described in AN4581 to be able to make a bootloader that can be booted via USB serial loader (since there are some modifications to the process required) but the modified bootloader signed in the fashion described in AN4581 will only boot on a HAB closed i.MX7D rev C processor. We cannot get this binary to boot on an i.MX7D rev D processor.

AN4581 mentions an errata, E11166, for the i.MX7D rev D processor that indicates that the first 4K of OCRAM is not available during boot time and the image start address must be greater or equal to 0x911000. Tests with our JTAG debugger show that the USB loaders manage to send the DCD data to OCRAM location 0x910000 correctly. The debugger shows DDR memory as empty, despite the USB tools reporting  that they successfully sent the data.

Is there a modification required to the U-boot signing process that is different from the process described in AN4581 in order to use this tool to boot it on a HAB closed i.MX7D rev D processor? Or is there a change that must be made to our UUU script to make it properly boot? 

Some help would be very greatly appreciated!!!

nxpfrankli, 2020-01-16:
Can you try change https://github.com/NXPmicro/mfgtools/blob/master/libuuu/rominfo.cpp

0x910000 to 0x911000

scallopedllama, 2020-04-21:
That was part of the issue but the other part was a missing memory configuration line in our U-Boot. NXP will be releasing the fix for everybody in the future

closed 2020-04-21 by scallopedllama

danie007, 2020-11-12:
@nxpfrankli I'm experiencing the same behaviour (`UUU` not working for closed board - it waits indefinitely) on i.MX 6UL EVK. Do I need to change the address as well?

[image]

nxpfrankli, 2020-11-12:
Can you try it by change address?  you can change it and send pull request. CI will generate prebuild image for you

danie007, 2020-11-16:
Hi @nxpfrankli 

I've tried changing as you've asked, still no use

> Can you try change https://github.com/NXPmicro/mfgtools/blob/master/libuuu/rominfo.cpp
> 
> 0x910000 to 0x911000

**I got the same error.**

I've raised it as a issue here: https://github.com/NXPmicro/mfgtools/issues/235 with detailed description. Could you follow that up?

Thanks