nxp-imx/mfgtools #154: No udc avaialble! issue for eMMC flashing i.MX8MQ  by using the UUU tool

venkatesh54, 2020-01-03:
Dear team,

Hardware and Tools Configuration:
Chipset: imx8mq-evk (MX8MQ)
Linux: linux-imx_4.14.78
UUU: libuuu_1.3.102-3-g64a66c3

I am getting below udc issue and i used this below scripts.

:~/ExtHD1/BBWA_KEPLER_PRJ/UUU_MFGTool/Test_Bin

/CUSTOM_DUAL$ sudo ./uuu -v ./eMMC_DUAL_Part_Flash.uuu 
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.3.102-3-g64a66c3

Build in config:
	Pctl	 Chip		 Vid	 Pid	 BcdVersion
	==================================================
	SDPS:	 MX8QXP		 0x1fc9	 0x012f	 [0x0002..0xffff]
	SDPS:	 MX8QM		 0x1fc9	 0x0129	 [0x0002..0xffff]
	SDPS:	 MX28		 0x15a2	 0x004f
	SDPS:	 MX815		 0x1fc9	 0x013e
	SDP:	 MX7D		 0x15a2	 0x0076
	SDP:	 MX6Q		 0x15a2	 0x0054
	SDP:	 MX6D		 0x15a2	 0x0061
	SDP:	 MX6SL		 0x15a2	 0x0063
	SDP:	 MX6SX		 0x15a2	 0x0071
	SDP:	 MX6UL		 0x15a2	 0x007d
	SDP:	 MX6ULL		 0x15a2	 0x0080
	SDP:	 MX6SLL		 0x1fc9	 0x0128
	SDP:	 MX7ULP		 0x1fc9	 0x0126
	SDP:	 MXRT106X	 0x1fc9	 0x0135
	SDP:	 MX8MM		 0x1fc9	 0x0134
	SDP:	 MX8MQ		 0x1fc9	 0x012b
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x0000..0x04ff]
	SDPV:	 SPL1		 0x0525	 0xb4a4	 [0x0500..0x9998]
	SDPU:	 SPL		 0x0525	 0xb4a4	 [0x9999..0x9999]
	FBK:	 		 0x066f	 0x9afe
	FBK:	 		 0x066f	 0x9bff
	FB:	 		 0x0525	 0xa4a5
	FB:	 		 0x18d1	 0x0d02
Wait for Known USB Device Appear...
New USB Device Attached at 3:51
3:51>Start Cmd:SDP: boot -f imx-boot-imx8mqevk-sd.bin-flash_evk
6400%3:51>Okay (0.007s)
New USB Device Attached at 3:51
3:51>Start Cmd:SDPU: delay 1000
3:51>Okay (0s)
3:51>Start Cmd:SDPU: write -f imx-boot-imx8mqevk-sd.bin-flash_evk -offset 0x57c00
100%3:51>Okay (0.051s)
3:51>Start Cmd:SDPU: jump 
6400%3:51>Okay (0s)
New USB Device Attached at 3:51
3:51>Start Cmd:FB: ucmd setenv fastboot_buffer ${loadaddr}
3:51>Okay (0s)
3:51>Start Cmd:FB: download -f Image
3:51>Okay (0.01s)
3:51>Start Cmd:FB: ucmd setenv fastboot_buffer ${fdt_addr}
3:51>Okay (0s)
3:51>Start Cmd:FB: download -f fsl-imx8mq-evk.dtb
3:51>Okay (0s)
3:51>Start Cmd:FB: ucmd setenv fastboot_buffer ${initrd_addr}
3:51>Okay (0s)
3:51>Start Cmd:FB: download -f fsl-image-mfgtool-initramfs-imx_mfgtools.cpio.gz.u-boot
3:51>Okay (0.004s)
3:51>Start Cmd:FB: ucmd setenv mfgtool_args ${mfgtool_args} mfg_mmcdev=${emmc_dev}
3:51>Okay (0s)
3:51>Start Cmd:FB: ucmd run mfgtool_args
3:51>Okay (0s)
3:51>Start Cmd:FB: acmd booti ${loadaddr} ${initrd_addr} ${fdt_addr}
3:51>Okay (0s)

Console Log:
---
[    3.326703] Bluetooth: BNEP socket layer initialized
[    3.331734] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[    3.337669] Bluetooth: HIDP socket layer initialized
[    3.342741] lib80211: common routines for IEEE802.11 drivers
[    3.348448] 9pnet: Installing 9P2000 support
[    3.352881] Key type dns_resolver registered
[    3.359158] registered taskstats version 1
[    3.371013] failed to disable 1.3GHz OPP!
[    3.376019] cpu cpu0: registered imx8mq-cpufreq
[    3.387106] update hantro voltage from 900 mV to 1000 mV
[    3.392722] hantrodec: module inserted. Major = 235
[    3.403682] rtc-rv3029c2 3-0056: setting system clock to 2037-07-03 22:45:03 UTC (2130273903)
[    3.412384] clk: Not disabling unused clocks
[    3.417602] SW1AB: disabling
[    3.421142] VGEN1: disabling
[    3.424691] VGEN6: disabling
[    3.428239] ALSA device list:
[    3.431224]   No soundcards found.
### [    3.435185] Freeing unused kernel memory: 1280K
No udc Available!
[    4.094131] random: crng init done
No udc Available!
[   62.526625] imx-sdma 302c0000.sdma: external firmware not found, using ROM firmware
[   62.526953] imx-sdma 30bd0000.sdma: external firmware not found, using ROM firmware
No udc Available!
No udc Available!
No udc Available!
No udc Available!
No udc Available!
---

In the kernel configuration i have enabled Following Configuration:

+++++++++++++++++++
#
# Gadget/Dual-role mode requires USB Gadget support to be enabled
#
# CONFIG_USB_DWC2_PERIPHERAL is not set
CONFIG_USB_DWC2_DUAL_ROLE=y
# CONFIG_USB_DWC2_PCI is not set
# CONFIG_USB_DWC2_DEBUG is not set
# CONFIG_USB_DWC2_TRACK_MISSED_SOFS is not set
CONFIG_USB_CHIPIDEA=y
CONFIG_USB_CHIPIDEA_OF=y
CONFIG_USB_CHIPIDEA_PCI=y
CONFIG_USB_CHIPIDEA_UDC=y
CONFIG_USB_CHIPIDEA_HOST=y
CONFIG_USB_ISP1760=y
CONFIG_USB_ISP1760_HCD=y
CONFIG_USB_ISP1761_UDC=y
# CONFIG_USB_ISP1760_HOST_ROLE is not set
# CONFIG_USB_ISP1760_GADGET_ROLE is not set
CONFIG_USB_ISP1760_DUAL_ROLE=y
CONFIG_USB_CDNS3=y
CONFIG_USB_CDNS3_GADGET=y
CONFIG_USB_CDNS3_HOST=y
++++++++++++++++

++++++++++++++++
#
# USB Peripheral Controller
#
CONFIG_USB_FOTG210_UDC=y
CONFIG_USB_GR_UDC=y
CONFIG_USB_R8A66597=y
CONFIG_USB_RENESAS_USBHS_UDC=m
CONFIG_USB_RENESAS_USB3=y
# CONFIG_USB_PXA27X is not set
# CONFIG_USB_MV_UDC is not set
# CONFIG_USB_MV_U3D is not set
CONFIG_USB_SNP_CORE=y
CONFIG_USB_SNP_UDC_PLAT=y
CONFIG_USB_M66592=y
CONFIG_USB_BDC_UDC=y

++++++++++++++++++++++++++

Kindly suggest me  if any configuration/steps are missing in this test.

Thanks,
Venkatesh

nxpfrankli, 2020-01-16:
You also need check your dtb, make sure usb driver probed.

nxpfrankli, 2020-03-30:
NO feedback for the long time. Closed it. Please feel free to reopen it

closed 2020-04-17 by nxpfrankli

PREMjITHT, 2023-03-21:
Hi Frankli,
what needs to be enabled in the devicetree?
The usbotg node?

nxpfrankli, 2023-03-21:
All USB OTG related node.

PREMjITHT, 2023-03-21:
I have added the USB OTG node and it's working. Thank you.