nxp-imx/mfgtools #157: [Question] How to pass MAC ID fuse value from a file (on Windows PC) to partition.uuu?

ajithpv, 2020-01-23:
Hi,

I have to fuse MAD ID for the platform from the UUU. The partition.uuu contains the MAC ID fuse similar to the following:

> #To flash BC:90:3A:E8:00:02 MAC address
> FB:ucmd fuse prog -y 0 708 **E83A90BC**
> FB:ucmd fuse prog -y 0 709 **00000200**

For a single board, this will work without any issue. However, this is not the best way, if we considering many boards with unique MAC ID. Changing the MAC ID manually for every board is a tedious and error-prone job.

Hence, I would like to pass this MAC ID from an external file (text file) that UUU can take and update the partition.uuu's corresponding field. Please note that it should update only the highlighted value and other parts of the partition.uuu should not modify.

Whether this kind of support available for UUU in Windows (uuu.exe)?
I have come across  [#131](https://github.com/NXPmicro/mfgtools/pull/131/) however, I don't know whether it integrated or not. Currently _-b_ option is using only the built-in script.

Looking forward to get a positive response.

nxpfrankli, 2020-01-23:
uuu support run a shell script to generate sequence ID, like MAC address.

fb: < a.bat

in a.bat 
echo "ucmd fuse prog -y 0 708 E83A90BC"

you need write small script to generate sequence number according to your rule

#131 is another option method. 

Added below line behind uuu_version 1.2.39

# @_mac            |  mac address

you can run uuu -b yourscript mac_address

uuu will replace _mac with mac_address

ajithpv, 2020-01-24:
Thank you @nxpfrankli for your quick support. I will check the above method and come back with the observations :+1:

ajithpv, 2020-01-30:
It is working like charm :+1:  Thank you very much for your wonderful support.

closed 2020-01-30 by ajithpv