nxp-imx/mfgtools #158: supported u-boot versions

laped83, 2020-01-27:
I have tried to get this tool working with fastboot on a imx7d using a mainline u-boot 2020.01. Looks like u-boot is compiled correctly and working but the fastboot commands is invalid.

command UCmd not recognized.

If I look in the fastboot code it looks like the arguments passed to the fastboot code is not even correct. So am I right to assume that you need a "NXP Vendor" u-boot to get uuu working?

nxpfrankli, 2020-01-27:
some extension of fastboot is not upstream. You need pick up some patch from our repo to enable ucmd.

StehleJochen, 2020-01-27:
try the "uboot-imx" ;) we had the same problem with our imx6ul costum board

laped83, 2020-01-28:
None of the fastboot commands are working between uuu and a mainline u-boot version. 

getvar <--- invalid command
ucmd <any uboot command> <--- invalid command
acmd <any never returned uboot command, like booti, reboot> <--- invalid command
flash [-raw2sparse] <partition> <filename> <--- invalid command
download -f <filename>  <--- invalid command

Maybe it should be mentioned in the README.md that a u-boot-imx version is needed?

nxpfrankli, 2020-01-28:
We suppose user use nxp released uboot.  You can send pull request.

RafNie, 2020-02-14:
@laped83 You need use u-boot with enabled fastboot. If you compile u-boot yourself, double check if fastboot is enable for it.
NXP released u-boot has FB support enabled.

closed 2020-02-18 by laped83