nxp-imx/mfgtools #159: How to save UUU output into a log file from Windows CMD?

ajithpv, 2020-01-30:
Hi,
UUU shows the update progress on the Windows command prompt by default. Is there any way to show these progress in command prompt as well as into a text file/log file? This will help us to see the realtime update on the command prompt and future check in the log file (I mean both is required).

I have tried to redirect the output to file as below from a batch script.

> uuu.exe partition.uuu > <file_name.txt>

However, it didn't work! the file is empty and progress shows on command prompt only.

Is there any way to achieve the same?

nxpfrankli, 2020-01-30:
uuu.exe -b partition.uuu > <file_name.txt>

you can use -v, but not log file yet.

ajithpv, 2020-02-06:
Thank you @nxpfrankli for the help. I can manage to get the log into a file using the above command.

closed 2020-02-06 by ajithpv