nxp-imx/mfgtools #160: Order of script arguments

RafNie, 2020-02-03:
Hi 

I'm working on my own uuu script. I have problems with script argument parsing ( **-b** option ).
Build from master  - 1.3.130.

Script has seven arguments. I passed them in the same order as were defined in script definition header section ( # @ arg | desc ). I expected that the uuu replaces args based on order from this arg definition section. 
But the uuu replace them in order of them use in script body. Please look at script output below.

Now I wonder if this is the behavior expected by the author or a mistake. Is it expected behavior?

I even created fix which use arguments in order defined in script header ( # @ arg | desc ).

Script:
`uuu_version 1.3.130
# arguments for uuu script:
# @_loadaddr	| load addr
# @_initrdaddr	| initrd addr
# @_fdtaddr 	| fdt addr
# @_uboot		| uboot with FB
# @_zImage		| kernel for FBK
# @_initramfs	| initramfs for FBK
# @_dtb			| DTB for FBK

# boot
SDP: boot -f _uboot
FB: ucmd setenv fastboot_buffer _loadaddr
FB: download -f _zImage
FB: ucmd setenv fastboot_buffer _initrdaddr
FB: download -f _initramfs
FB: ucmd setenv fastboot_buffer _fdtaddr
FB: download -f _dtb
FB: acmd bootz _loadaddr _initrdaddr _fdtaddr
FB: done
`

Run command for test:
` ./uuu  -V  -b boot.uuu 1 2 3 4 5 6 7`

Output from script:
` uuu_version 1.3.130
# arguments for uuu script:
# @2    | load addr
# @4    | initrd addr
# @6    | fdt addr
# @1            | uboot with FB
# @3            | kernel for FBK
# @5    | initramfs for FBK
# @7                    | DTB for FBK

# boot
SDP: boot -f 1
FB: ucmd setenv fastboot_buffer 2
FB: download -f 3
FB: ucmd setenv fastboot_buffer 4
FB: download -f 5
FB: ucmd setenv fastboot_buffer 6
FB: download -f 7
FB: acmd bootz 2 4 6
FB: don`

nxpfrankli, 2020-02-03:
Yes, I think it should flow order at beginning.  
-b script file is added in recently. 

All internal script's order is the same. so not found this problem.

your patch partial work, can you improve it and not block no comments case.

RafNie, 2020-02-04:
> your patch partial work, can you improve it and not block no comments case.

Actually, the fix does not block no comment case. I've only add TAB char as another delimiter of 
 argument.  I thought it might be useful. The space and new line works as before.

Look at my test. 
No comment script:
```
uuu_version 1.3.130
# arguments for uuu script:
# @_loadaddr
# @_initrdaddr
# @_fdtaddr
# @_uboot
# @_zImage
# @_initramfs
# @_dtb

# boot
SDP: boot -f _uboot
FB: ucmd setenv fastboot_buffer _loadaddr
FB: download -f _zImage
FB: ucmd setenv fastboot_buffer _initrdaddr
FB: download -f _initramfs
FB: ucmd setenv fastboot_buffer _fdtaddr
FB: download -f _dtb
FB: acmd bootz _loadaddr _initrdaddr _fdtaddr
FB: done
```

Command:
`./uuu  -V -b test1.uuu 1 2 3 4 5 6 7`

And output:
```
Run built-in script:
 uuu_version 1.3.130
# arguments for uuu script:
# @1
# @2
# @3
# @4
# @5
# @6
# @7

# boot
SDP: boot -f 4
FB: ucmd setenv fastboot_buffer 1
FB: download -f 5
FB: ucmd setenv fastboot_buffer 2
FB: download -f 6
FB: ucmd setenv fastboot_buffer 3
FB: download -f 7
FB: acmd bootz 1 2 3
FB: done
```

nxpfrankli, 2020-02-05:
My means no @ at all.
 
# @_loadaddr
# @_initrdaddr
# @_fdtaddr
# @_uboot
# @_zImage
# @_initramfs
# @_dtb

it is optional.  

I don't want to block below case 

uuu_version 1.3.130
SDP: boot -f _uboot
FB: ucmd setenv fastboot_buffer _loadaddr
FB: download -f _zImage
FB: ucmd setenv fastboot_buffer _initrdaddr
FB: download -f _initramfs
FB: ucmd setenv fastboot_buffer _fdtaddr
FB: download -f _dtb
FB: acmd bootz _loadaddr _initrdaddr _fdtaddr
FB: done

RafNie, 2020-02-05:
I understood. I prepared new version of fix [17e646b](https://github.com/RafNie/mfgtools/commit/17e646b9d71f9f26ce93e955f61d7ed536e22b38)

It works with comments:
```
uuu_version 1.3.130
# arguments for uuu script:
# @_loadaddr	| load addr
# @_initrdaddr	| initrd addr
# @_fdtaddr 	| fdt addr
# @_uboot		| uboot with FB
# @_zImage		| kernel for FBK
# @_initramfs	| initramfs for FBK
# @_dtb			| DTB for FBK

# boot
SDP: boot -f _uboot
FB: ucmd setenv fastboot_buffer _loadaddr
FB: download -f _zImage
FB: ucmd setenv fastboot_buffer _initrdaddr
FB: download -f _initramfs
FB: ucmd setenv fastboot_buffer _fdtaddr
FB: download -f _dtb
FB: acmd bootz _loadaddr _initrdaddr _fdtaddr
FB: done
```

output:
```
./uuu -V -b test.uuu 1 2 3 4 5 6 7

uuu_version 1.3.130
# arguments for uuu script:
# @1    | load addr
# @2    | initrd addr
# @3    | fdt addr
# @4            | uboot with FB
# @5            | kernel for FBK
# @6    | initramfs for FBK
# @7                    | DTB for FBK

# boot
SDP: boot -f 4
FB: ucmd setenv fastboot_buffer 1
FB: download -f 5
FB: ucmd setenv fastboot_buffer 2
FB: download -f 6
FB: ucmd setenv fastboot_buffer 3
FB: download -f 7
FB: acmd bootz 1 2 3
FB: done
```

But also for no comment case:
```
uuu_version 1.3.130

# boot
SDP: boot -f _uboot
FB: ucmd setenv fastboot_buffer _loadaddr
FB: download -f _zImage
FB: ucmd setenv fastboot_buffer _initrdaddr
FB: download -f _initramfs
FB: ucmd setenv fastboot_buffer _fdtaddr
FB: download -f _dtb
FB: acmd bootz _loadaddr _initrdaddr _fdtaddr
FB: done
```

output:
```
./uuu -V -b test.uuu 1 2 3 4 5 6 7

 uuu_version 1.3.130

# boot
SDP: boot -f 1
FB: ucmd setenv fastboot_buffer 2
FB: download -f 3
FB: ucmd setenv fastboot_buffer 4
FB: download -f 5
FB: ucmd setenv fastboot_buffer 6
FB: download -f 7
FB: acmd bootz 2 4 6
FB: done

```

RafNie, 2020-02-08:
@nxpfrankli Should I prepare pull request to master?  Or do you prefer to fix it differently?

nxpfrankli, 2020-02-08:
Yes, please send pull request to master

closed 2020-02-10 by RafNie