nxp-imx/mfgtools #161: How to close the UUU command prompt if no device connected to the USB port?

ajithpv, 2020-02-06:
Hi,

We have a requirement where we need to flash using UUU to a particular USB port (this will find using lsusb command). This process is for multiple boards and we are logging the UUU output to a file for later verification. Hence, no one manually closes the Window every time it opens. It is assumed that, the Windows command prompt window will close either the flashing got over/error comes or no USB connection is available to the specified USB port number.

Currently, if no USB connection is available, then UUU flashing open a Windows command prompt and shows the below message and waits there until we close the window:

```
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.3.102-0-gb078bd0

Your console don't support VT mode, fail back to verbose mode
```

1. May I know what "Your console don't support VT mode, fail back to verbose mode" means?

2. Is there any way to close the Windows command prompt if there is no connection is available to the specified USB port number? Or can we set a time out instead of waiting indefinitely until we close the window manually?

ajithpv, 2020-02-06:
> 2. Is there any way to close the Windows command prompt if there is no connection is available to the specified USB port number? Or can we set a time out instead of waiting indefinitely until we close the window manually?

When I use -t command with timeout seconds, I can able to close the Windows command prompt after the specified timeout period. This is a nice feature already available in UUU and thanks to @nxpfrankli for making these useful options :+1: 

Could anyone let me know the answer for the 1st query?

nxpfrankli, 2020-02-06:
Are you use win7?  win7's command console don't support VT mode.  cursor can't return to previous line by vt100 escape sequence

ajithpv, 2020-02-07:
I'm using windows 10. I think this is applicable to windows 10 also. Please correct me if I'm wrong.

nxpfrankli, 2020-02-07:
Do you win10 native cmd or powershell,  or other third part shell

ajithpv, 2020-02-08:
I'm using Windows 10 native CMD and nor using the PowerShell or third part shell.
I must say that I'm redirecting the output using ">" to a file. Whether this is making the above message?

nxpfrankli, 2020-02-08:
Yes. > redirect cause VT failure.

ajithpv, 2020-02-10:
Thank you @nxpfrankli for the confirmation :+1:

closed 2020-02-10 by ajithpv