nxp-imx/mfgtools #165: Possible high memory usage when flashing

denravonska, 2020-03-01:
I'm trying to use a Raspberry Pi 3 (1GB RAM) as a flash device in a pipeline. When flashing an image I can see that the memory usage skyrockets from 140MB to 1GB RAM plus 1GB swap, and the system becomes unresponsive.

Before venturing further down the rabbit hole, does uuu unpack the provided zip to RAM in one swoop, or does it unpack on the fly as needed while streaming from disk?

nxpfrankli, 2020-03-02:
Yes,  uuu design run at 64bit PC, which have enough memory. If run at PI, I suggest you uncompressed zip before run uuu. 

if uuu access regular file,  uuu use mmap, only use bit virtual memory, which is not problem in 64bit system

denravonska, 2020-03-10:
Closing this as it's by design.

When discarding the output of `uuu` in the Azure pipeline the flash procedure completes just fine.

closed 2020-03-10 by denravonska