nxp-imx/mfgtools #167: 'Bulk(R):LIBUSB_ERROR_IO' Failure

robertosartori, 2020-03-04:
Hi,
I'm having an issue flashing a full Android OS using uuu v1.3.136.

OS: Fedora 31 and Ubuntu
uuu: binary from NXP here https://github.com/NXPmicro/mfgtools/releases v1.3.136
board: custom board with iMX7D SOC

I'm running a lst script from uuu that uses the fastboot protocol to flash the images. It usually works fine, but sometimes I get this error:

3:212    7/14 [===============> 43%                  ] FB: flash system system.img      
Success 0    Failure 0                                                                         
3:212    7/14 [===============> 43%                  ] FB: flash system system.img      
Success 0    Failure 0                                                                                 
3:212    7/14 [===============> 43%                  ] FB: flash system system.img      
Success 0    Failure 0                                                                    
3:212    7/14 [================>44%                  ] FB: flash system system.img      
Success 0    Failure 0                                                            
3:212    7/14 [================>44%                  ] FB: flash system system.img      
Success 0    Failure 1                                                           
3:212    7/14 [Bulk(R):LIBUSB_ERROR_IO               ] FB: flash system system.img      
Success 0    Failure 1                                                                 
3:212    7/14 [Bulk(R):LIBUSB_ERROR_IO               ] FB: flash system system.img

The 'Bulk(R):LIBUSB_ERROR_IO ' error can happen flashing the system.img or a userdata.img, and it always seems to happen flashing big files (it never happens flashing boot or recovery).

Any idea about this?

Thank you,

Roberto

nxpfrankli, 2020-03-04:
look like USB disconnected!  Did you touch USB cable or see USB remove and plug again in PC?

robertosartori, 2020-03-06:
Hi,
thanks for your input. I'm able to verify using 'dmesg' from my pc that, when the error happens, I get this:
'usb 3-2.1-port2: disabled by hub (EMI?), re-enabling...
usb 3-2.1.2: USB disconnect, device number 108'

During my tests, I also got the 'Bulk(W):LIBUSB_ERROR_IO' error sometimes.
I did not physically touch the cable or move the board.

Also, note that flashing the same images with fastboot I do not get this error. I've setup a test that continuously flashes the same images using fastboot or uuu, and I get errors only using uuu.

nxpfrankli, 2020-03-06:
Not sure why usb disconnect.  Did you draw current from VBUS?  May over current?

robertosartori, 2020-03-13:
Hi, I'm powering the board using a power bench or using a battery (same issue in both cases).

aicastell, 2020-06-10:
Just the same error here using Ubuntu 18.04 Linux to flash a i.MX8 board.

''
$ sudo ../../uuu-1.3.191/build/uuu/uuu -b emmc ./imx-boot-sd.bin-flash_all fsl-image-20200609144303.rootfs.sdcard
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.3.191-0-g4fe24b9

Success 0    Failure 1                                                                                                                                                                                                                                                         
                                                                                                                                                                                                                                                                                
                                                                                                                                                                                                                                                                                
1:5      4/ 7 [Bulk(W):LIBUSB_ERROR_IO               ] FB: flash bootloader fsl-image-20200609144303.rootfs.sdcard
''                                                                                                                                   

However with an older version the same setup works:

''
$ sudo ../../uuu_1.2.68/uuu -b emmc ./imx-boot-sd.bin-flash_all fsl-image-20200609144303.rootfs.sdcard
uuu (Universal Update Utility) for nxp imx chips -- libuuu_1.2.68-0-gd2000f4

Success 1    Failure 0                                                                                                                                                                                                                                                         
                                                                                                                                                                                                                                                                                
1:5    6/ 6   [Done                        ] FB: Done  
''