nxp-imx/mfgtools #168: Windows driver assignment

StehleJochen, 2020-03-09:
Hello together,

we want to run uuu.exe for device production on windows. We have a custom board with imx6ul. In u-boot we set VID to 0x0525 and PID to 0xa4a5. Now the problem is everytime we plug in a fresh device windows isnt able to autoload the USB Download gadget driver. We have to select it manually via devicemanager. How can we fix that?

Is it correct that every single device would creat a new entry in windows registry? would this occurs on overflow or something if you want to produce lets say thousands of devices?

nxpfrankli, 2020-03-09:
Missed WCID?  did you set up difference serial no for each device?

nxpfrankli, 2021-02-11:
if still have problem, please reopen.

closed 2021-02-11 by nxpfrankli