nxp-imx/mfgtools #173: uuu_1.3.154{.zip,.tar.gz} are missing .tarball-version

OsterlaD, 2020-03-25:
Hi @nxpfrankli ,

is it not possible to add the .tarball-version file to the source tarball, as suggested in the pull request?
I have found [some artifacts](https://github.com/NXPmicro/mfgtools/blob/master/appveyor.yml#L90) but not the source archives. Where is the source tarball generation? Does it appveyor automatically?

See: https://github.com/NXPmicro/mfgtools/pull/171
See: #77

nxpfrankli, 2020-03-25:
It is generated by github when publish release.

OsterlaD, 2020-03-25:
oh. So the Problem is not jet fixed. Some may now be able to build from the source tarball with the extra step to write .tarball-version but this is not yet perfect. Do you know if it is possible to suppress automatically generated source archives and add custom?

nxpfrankli, 2020-03-25:
I don't know yet. I sent a request to github customer support. let wait for what github said?

OsterlaD, 2020-03-25:
Sounds good. Let's wait for the answer from support.

OsterlaD, 2020-04-11:
Hi,

https://github.com/NXPmicro/imx-uuc/issues/12#issuecomment-604176968

I guess this comment was for this issue.

It seems to be no problem. Look at https://github.com/MicrochipTech/cryptoauthlib/releases/tag/v3.1.0 the generated archive has archive in the URL but the uploaded archive has release in the URL. So it should be no problem. It might be a good idea to add a hint to the readme that only the uploaded archive builds.

Regards

nxpfrankli, 2020-04-12:
looks like script generate new source code package.  it looks strange about there are two sets of source package

OsterlaD, 2020-04-12:
Hi, I see. But to build the program from tarball the info can not come from git, because the history is not there. In my opinion the added generated file has that advantage that it is easy to automate.
The other option would be a "prepare release" commit to store version number of release before release. This has option has the advantage that it works with all tarballs generated by GitHub and even works with git shallow clone. The decision is up to you. Regards

nxpfrankli, 2020-04-14:
can you try https://ci.appveyor.com/project/nxpfrankli/mfgtools-kvqcg/builds/32176686/job/1rqh8jrjr9suppv5/artifacts

OsterlaD, 2020-04-15:
Hi Frank, great job! It works!
Thanks you very much.

Looking forward for your official 163 release ;-)

closed 2020-04-15 by OsterlaD

OsterlaD, 2020-04-20:
Hi @nxpfrankli ,
one question about the uuu_source-1.3.167.tar.gz.
In compare to uuu-1.3.163.tar.gz (your link above) the new tarball has two top-level directories.
The 163 contains only uuu-1.3.163 directory, but 167 contains uuu-1.3.167 and uuu_source-1.3.167.
Was this an accident or has it some special meaning?
Regards

nxpfrankli, 2020-04-20:
Sorry, it is caused by I rename it. try 169 pre-release

OsterlaD, 2020-04-20:
There it is only one toplevel dir.
Thanks