nxp-imx/mfgtools #175: "Config Config::find(const string &protocal)" not doing what it should?

markuspg, 2020-03-26:
Hi,

looking at the code of "Config Config::find(const string &protocal)" I guess the author intended to have a Config instance returned that only contains ConfigItem of the given protocol. But because of Config's default constructor I'd guess that instead a Config instance is returned which contains all ConfigItem instances of the given protocol twice.

nxpfrankli, 2020-03-30:
Script support "CFG" dynamic protocol with customized PID VID.  It copy to new object configs in case user change scripts file in daemon mode.  A thread load new scripts, but still a thread running with old scripts.

nxpfrankli, 2020-04-21:
No update for more than 3 weeks.  feel free to reopen it if still problem

closed 2020-04-21 by nxpfrankli