nxp-imx/mfgtools #176: [Configuring incomplete] cmake .

nkevy, 2020-03-29:
HI all,

After running the following cmds on macos 10.14.6:
#####
git clone https://github.com/NXPmicro/mfgtools.git
rm -r libusb
git clone https://github.com/nxpfrankli/libusb.git
sudo cmake . 
#####
I receive the following 
#####
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at libuuu/CMakeLists.txt:56:
  Syntax Warning in cmake code at column 159

  Argument not separated from preceding token by whitespace.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Checking for module 'libusb-1.0>=1.0.16'
--   No package 'libusb-1.0' found
CMake Error at /Applications/CMake.app/Contents/share/cmake-3.17/Modules/FindPkgConfig.cmake:467 (message):
  A required package was not found
Call Stack (most recent call first):
  /Applications/CMake.app/Contents/share/cmake-3.17/Modules/FindPkgConfig.cmake:647 (_pkg_check_modules_internal)
  libuuu/CMakeLists.txt:8 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "/Users/#####myuser#####/work/flash/build_uuu/mfgtools/CMakeFiles/CMakeOutput.log".

####### 
I have used brew to install and link libusb 
The .tar file exists in /Users/#####myuser#####/Library/Caches/Homebrew as
######
libusb--1.0.23.mojave.bottle.tar.gz
######
If yall could give me a hint or alter the pkg-config directive in the CMakeLists.txt that would be amazing. 
Thanks!

nxpfrankli, 2020-03-30:
mac os please refer https://github.com/NXPmicro/mfgtools/blob/master/.travis.yml

nxpfrankli, 2020-03-30:
Mac OS main problem is can't unload i.MX's HID driver, need signed key to do that

nxpfrankli, 2020-04-21:
No response for more than 3 weeks. feel free to reopen it.

closed 2020-04-21 by nxpfrankli

michhar, 2021-03-18:
If it helps someone I did the following on Catalina (10.15.7) to compile `uuu` (these steps before `cmake` command).  I have Xcode 11.6.

1) In `libuuu/CMakeLists.txt` and `uuu/CMakeLists.txt` add the following to allow `cmake` to find your packages.

```
list(APPEND CMAKE_PREFIX_PATH /usr/local)
```

2) Update the following environment variables.

```
export PKG_CONFIG_PATH="/usr/local/opt/zlib/lib/pkgconfig:$PKG_CONFIG_PATH"
export PATH="/usr/local/sbin:$PATH"
```